bb-qq/r8152 #180: DS916+ confirmed with Club 3D USB 3.2 Gen1 Adapter, but ...

ThoDieX, 2022-01-08:
### Description of the problem

After a shutdown and restart the negotiated speed at a TL-SG108-M2 is only 100 or 1000. It starts with 2.5Gb and after STATUS is stopping blinking (start fulfilled) the adapter is restarted (network off and on). After this I have only 100Mb or 1000Mb.

Diconnecting the network cable and reinserting it helps to negotiate at 2.5Gb until the system is restarted.

Problem with the swicth? Other network devices are working fine with this switch (Intel l255-v and a Fritz Repeater 6000).

### Description of your products

> - DS 916+
>    Linux SchlumpfNAS 3.10.108 #42218 SMP Fri Sep 24 02:39:06 CST 2021 x86_64 GNU/Linux synology_braswell_916+
>     - DSM version 7.0.142218
> - Club 3D USB 3.2 Gen1 Adapter Typ-A zu 2, 5 Gigabit Ethernet 

### Description of your environment

> - Connected directly to a TL-SG108-M2 switch from TP-LINK
>     > - CAT 5e

### Output of ``dmesg`` command
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.108 (root@build17) (gcc version 7.5.0 (GCC) ) #42218 SMP Fri Sep 24 02:39:06 CST 2021
[    0.000000] Command line: root=/dev/md0 ihd_num=4 netif_num=2 HddHotplug=1 SataPortMap=23 sata_remap=0>2:1>3:2>0:3>1                                                                       acpi_enforce_resources=lax libata.force=5:3.0G syno_hw_version=DS916+ vender_format_version=2 syno_hdd_detect=56,59,63,5                                                                      7 syno_hdd_enable=61,60,71,58 syno_usb_vbus_gpio=93@0000:00:14.0@2 KPTI_on sn=1640NZN335101 macs=0011325bbf09,0011325bbf                                                                      0a
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000006efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000006f000-0x000000000006ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000070000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007a4befff] usable
[    0.000000] BIOS-e820: [mem 0x000000007a4bf000-0x000000007aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aebf000-0x000000007bfbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007bfbf000-0x000000007bffefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007bfff000-0x000000007bffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] e820: update [mem 0x01d87330-0x01d8733f] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000006efff] usable
[    0.000000] reserve setup_data: [mem 0x000000000006f000-0x000000000006ffff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000000070000-0x0000000000085fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000086000-0x000000000009ffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000001d8732f] usable
[    0.000000] reserve setup_data: [mem 0x0000000001d87330-0x0000000001d8733f] usable
[    0.000000] reserve setup_data: [mem 0x0000000001d87340-0x000000007a4befff] usable
[    0.000000] reserve setup_data: [mem 0x000000007a4bf000-0x000000007aebefff] reserved
[    0.000000] reserve setup_data: [mem 0x000000007aebf000-0x000000007bfbefff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x000000007bfbf000-0x000000007bffefff] ACPI data
[    0.000000] reserve setup_data: [mem 0x000000007bfff000-0x000000007bffffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000100000000-0x000000027fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by INSYDE Corp.
[    0.000000] efi:  ACPI 2.0=0x7bffe014  SMBIOS=0x7a850000
[    0.000000] efi: mem00: type=3, attr=0xf, range=[0x0000000000000000-0x0000000000001000) (0MB)
[    0.000000] efi: mem01: type=7, attr=0xf, range=[0x0000000000001000-0x000000000006f000) (0MB)
[    0.000000] efi: mem02: type=10, attr=0xf, range=[0x000000000006f000-0x0000000000070000) (0MB)
[    0.000000] efi: mem03: type=7, attr=0xf, range=[0x0000000000070000-0x0000000000086000) (0MB)
[    0.000000] efi: mem04: type=0, attr=0xf, range=[0x0000000000086000-0x0000000000088000) (0MB)
[    0.000000] efi: mem05: type=6, attr=0x800000000000000f, range=[0x0000000000088000-0x00000000000a0000) (0MB)
[    0.000000] efi: mem06: type=2, attr=0xf, range=[0x0000000000100000-0x0000000000ea2000) (13MB)
[    0.000000] efi: mem07: type=7, attr=0xf, range=[0x0000000000ea2000-0x000000003765e000) (871MB)
[... 1200 lines cut ...]
          inet addr:192.168.44.81  Bcast:192.168.44.255  Mask:255.255.255.0
          inet6 addr: fd00::a2ce:c8ff:fef6:4d17/64 Scope:Global
          inet6 addr: fe80::a2ce:c8ff:fef6:4d17/64 Scope:Link
          inet6 addr: 2a02:8070:29a9:7b00:a2ce:c8ff:fef6:4d17/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6168 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10307 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1562516 (1.4 MiB)  TX bytes:10527824 (10.0 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1409 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1409 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:204856 (200.0 KiB)  TX bytes:204856 (200.0 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

bb-qq, 2022-01-09:
Could you try disabling auto-negotiation by `ethtool -s eth2 speed 2500 duplex full autoneg off`?

ThoDieX, 2022-01-09:
Yes, had tried this several times but no success. Also not in the configuration file itself:

root@SchlumpfNAS:~# sudo ethtool -s eth2 speed 2500 duplex full autoneg off
Cannot set new settings: Invalid argument
  not setting speed
  not setting duplex
  not setting autoneg

Seems that this is blocked even the command is available.

I have now tried it on an older DS415+. The adapter runs on this one as well. The same symptoms appear:

- Adapter is recognized and switch shows 2.5Gb at startup.
- Once startup is complete, STATUS stops flashing, then shortly thereafter the network disables (both LEDs on the switch's port are off) and after 5 seconds or so, activates at a random rate. Sometimes it is 100 Mbit (at the switch left off, right orange), sometimes it is 1000Mbit (at the switch left off, right green). 

- If I now want to have 2500Mbit, then I have to disconnect the LAN cable briefly and reconnect it again. Then the port LED left is green and right is off, a sign for 2500Mbit and in DSM itself is also.

However, the performance is not great on the DS415+, because of the weak CPU the device only manages 1200Mbit for simple SMB2/3 tasks (e.g. copying a 32GB Raspi image). But this is another thing and not about the driver ...

bb-qq, 2022-01-09:
One user reported that `ethtool -s eth2 speed 1000 duplex full autoneg off` disables auto-negotiation with 2.5 Gbps link, I'm not sure why, but it does seem to work for him. Try that command if you like.

ThoDieX, 2022-01-09:
With speed 1000 it is working. But speed 2500 is ignored ... 

Also it is shown in adapter features:

MyAdmin@SchlumpfNAS:~$ sudo ethtool eth2
Settings for eth2:
        Supported ports: [ MII ]
        Supported link modes:   10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseX/Full
        Supported pause frame use: No
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  10baseT/Half 10baseT/Full
                                100baseT/Half 100baseT/Full
                                1000baseT/Full
                                2500baseX/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: Yes
        Advertised FEC modes: Not reported
        Link partner advertised link modes:  100baseT/Half 100baseT/Full
                                             1000baseT/Full
        Link partner advertised pause frame use: No
        Link partner advertised auto-negotiation: Yes
        Link partner advertised FEC modes: Not reported
        Speed: 2500Mb/s
        Duplex: Full
        Port: MII
        PHYAD: 32
        Transceiver: internal
        Auto-negotiation: on
        Supports Wake-on: pumbg
        Wake-on: g
        Current message level: 0x00007fff (32767)
                               drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
        Link detected: yes

bb-qq, 2022-01-11:
So, is your issue completely resolved?

ThoDieX, 2022-01-11:
Eh ... no!?

a) Why should reducing speed to 1000Mb solve this issue?
b) I cannot set to 2500Mb because the system refuses it
c) The behavior has not changed. The adapter is reset after starting and gets a speed randomly between 100 and 1000 on two different NAS (916+ & 415+) 

Does anyone has this issue after a cold start, too?

ThoDieX, 2022-01-11:
I've checked with 2.15.0-5 and tested another adapter (_ASUS USB-C2500_) but it's the same. Starting up fine unless status is "booted" and the LED not not blinking anymore. The network LED are complete off for 10 seconds and the wrong speed is negotated afterwards even it was correct before. Plugging out and in of the ethernet cable fixes this until next reboot.

Sjekke, 2022-01-11:
I also have the 916+ but use the Plugable 2,5gbit adapter. When I look at my Asus router it's connected at 2,5gbit. Can I check something on my NAS to check if it's really the case?

ThoDieX, 2022-01-12:
Hej Sjekke,

if the NAS tells you you have 2500Mb it's fine ... go to network in DSM and expand the network settings for your card., There you will see the speed.

ThoDieX, 2022-01-12:
@bb-qq : I have connected the NAS with 2,5Gb to the AVM Repeater port and it works fine. So it must be a problem of the driver to negotiate the connection with the switch. I have also replaced the cables, too - no change. On a Windows machine and on a Raspberry Pi, there are no such problems with the switch and from boot I have negotiated a fixed 2500Mb.

Sjekke, 2022-01-12:
> Hej Sjekke,
> 
> 
> 
> if the NAS tells you you have 2500Mb it's fine ... go to network in DSM and expand the network settings for your card., There you will see the speed.

Ok, checked it and this is the case. 😀

bb-qq, 2022-01-12:
I was expecting to disable auto-negotiation by running that command, keeping the link speed at 2500mbps. Even if we specified 1000mbps to the parameter of the command, this may be ignored and the speed remains 2500mbps in some environments.

The fact that it's working well on the PC suggests that it might be a power supply issue. In that case, using a Y-cable or a self-powered USB hub with an AC adapter may help.

ThoDieX, 2022-01-12:
@bb-qq : I disagree with you. If this would be true, it has a problem during using it. But it is stable. If this is true it would have problems with other connections. But at my AVM Repeater 6000 it also runs well and has no problem at start up. It runs well on a low powered Raspberry Pi. 

So at the and it is a problem of the driver in combination with the TP Link TL-SG108-M2 switch that after start-up (!) something went wrong on the handling out the speed (a second time!?). I don't know what DSM does to the USB port or how it handles the driver itself but fact is, it's not a power issue because running it on the same devices connected to other 2.5Gb connections/ports there are no problems.

--> Driver issue

Sjekke, 2022-01-12:
If it works well with your AVM repeater and a low powered RPI, why can't the problem be situated on your TP?

bb-qq, 2022-01-13:
Thanks for the summary. I can understand from this situation that you think this is a driver issue. However, the Synology platform has various USB issues, especially with older models, so it's not surprising that this is happening.

In fact, the source code for my driver is straight from RealTek, and its contents are almost identical to the RaspberryPi's kernel driver. 
And the auto-negotiation is controlled by a few register values these do not vary at startup or not, and negotiation is done by the PHY. We can not configure details such as timeout. Because of this, I can not think of anything about modifying the driver. The only workaround I can think of is to disable auto-negotiation, sadly.

ThoDieX, 2022-01-13:
All fine :-) ... it's just an "Sherlock Holmes" effect to figure it out ...

I've ordered an different switch from Zyxel with 8x 2,5Gb and I will tell if this is running better or not. If yes, maybe you can add a small warning in your FAQ abut TP-Link switche :-) ...

Thanks for helping!

Sjekke, 2022-01-13:
If I read all the post, I really can recommend the Plugable 2,5gbit adapter. Works like a charm ...

ThoDieX, 2022-01-14:
So, Zyxel MG-108-ZZ0101F arrived and working like charm now ... no problems anymore ... full 2.5Gb at all boot processes ...

ThoDieX, 2022-01-15:
Not the switch allone. The combination of Realtek USB adapter + your driver + Synology NAS + TP Link switch at start up.

One last, rather cosmetic thing now is that when the NAS is shut down or booted, a message is sent that the NAS is no longer connected to Synology (quickconnect). But that is to get over and I have that even without the adapter in one of a hundred boot processes and can be turned off or ignored. Here the network connection is terminated faster than the rest of the system notices it and the system can tell Synology that there is only a reboot and no network failure. So this is not a matter for you.

jensche, 2022-03-23:
Hello, I got the same issue. But in my case the adapter worked like a charm for 1 month. after 1 restart it just jumped down from 2.5gbit down to 100mbit. 

`Supported ports: [ MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseX/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	                                     1000baseT/Half 1000baseT/Full 
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 100Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: pumbg
	Wake-on: g
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes
`

jensche, 2022-03-23:
which is quite funny. it has worked 1 months without problems. After a reboot has no longer worked.

bb-qq, 2022-04-10:
My reading of the driver code indicates that this device does not have the ability to lock the link speed to 2.5Gbps. So sadly `sudo ethtool -s eth2 speed 2500 duplex full autoneg off` will always fail in any environment.

I also searched Google and found many reports of Realtek chips giving link speeds of 100mbps. Some people said that replacing the cable solved the problem, while others got around it by fixing the link speed with an opposing switch. I can't think of anything else to try.

bb-qq, 2024-06-24:
We now know how to disable autonegotiation on the driver side and use the following command:
```
sudo ethtool -s eth2 autoneg on advertise 0x802f
```
https://github.com/bb-qq/r8152/wiki/Troubleshooting#fix-to-25gbps-r8152-specific

I would like to close this issue as it has not been commented on for a long time. If you have any questions, please create a new issue.

closed 2024-06-24 by bb-qq