bb-qq/r8152 #181: Weird case with DS918+ and CableMatters USB 2.5Gbe adapter

nubbie61, 2022-01-09:
### Description of the problem

> DS918+ direct connected to a PC via two CableMatters USB 2.5Gbe adapters (without a switch). Both are recognized and do have a link.; manual IP static addresses, the same LAN; 255.255.255.0 masks, no gateways nor DNS. PC is using driver from Realtek's site - currently version 10.49.201102, while NAS has bb-qq version 13.0.1 installed and active. Using iperf3 to measure the real world speed (when PC is the client reading from the NAS server) it gives 2,37 Gbps which is OK and expected. So far, so good. When turn it the other way around - NAS to become a client reading from the PC acting as server with iperf3 - it hangs dramatically. The LAN3 (2.5Gbe) could be revived only by "ifconfig eht2 down" and then "ifconfig eth2 up" commands or by stopping and restating the driver from Package Center. Headscratcher for me big time. Tried playing with the firewall, Kaspersky etc - no luck.

### Description of your products

> DS918+
> Linux HomeNAS 4.4.59+ #25556 SMP PREEMPT Sat Aug 28 02:17:26 CST 2021 x86_64 GNU/Linux synology_apollolake_918+
> DSM 6.2.4-25556 Update 2
> CableMatters USB 2.5Gbe adapter

### Description of your environment

> connect directly to a PC
> CableMatters USB 2.5Gbe adapter
> NA
> Cat.6

### Output of ``dmesg`` command

````
[331186.074244] init: scsi_plugin_server main process (1266) killed by TERM signal
[331188.932401] init: synowsdiscoveryd main process ended, respawning
[331199.402020] init: synowsdiscoveryd main process ended, respawning
[331209.775754] init: synowsdiscoveryd main process ended, respawning
[331216.321144] r8152 2-2:1.0 eth2: carrier off
[331217.827018] init: scsi_plugin_server main process (2281) killed by TERM signal
[331220.152919] init: synowsdiscoveryd main process ended, respawning
[331230.523890] init: synowsdiscoveryd main process ended, respawning
[331240.898537] init: synowsdiscoveryd main process ended, respawning
[331251.272668] init: synowsdiscoveryd main process ended, respawning
[331261.656659] init: synowsdiscoveryd main process ended, respawning
[331270.228614] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[331272.031419] init: synowsdiscoveryd main process ended, respawning
[331282.404690] init: synowsdiscoveryd main process ended, respawning
[331292.776443] init: synowsdiscoveryd main process ended, respawning
[331293.369489] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[331293.376858] r8152 2-2:1.0 eth2: carrier on
[331294.897619] init: scsi_plugin_server main process (3090) killed by TERM signal
[331295.126048] init: nmbd main process (5185) killed by TERM signal
[331295.417213] r8152 2-2:1.0 eth2: carrier off
[331296.842074] init: iscsi_pluginserverd main process (5103) killed by TERM signal
[331296.870560] init: iscsi_pluginengined main process (5088) killed by TERM signal
[331297.011824] init: scsi_plugin_server main process (5112) killed by TERM signal
[331297.517827] init: iscsi_pluginserverd main process (6295) killed by TERM signal
[331297.534954] init: iscsi_pluginengined main process (6269) killed by TERM signal
[331297.621577] init: scsi_plugin_server main process (6325) killed by TERM signal
[331298.009628] init: nmbd main process (6543) killed by TERM signal
[331298.233642] r8152 2-2:1.0 eth2: carrier on
[331299.981455] init: iscsi_pluginserverd main process (6508) killed by TERM signal
[331299.999087] init: iscsi_pluginengined main process (6492) killed by TERM signal
[331300.058713] init: scsi_plugin_server main process (6533) killed by TERM signal
[331300.364742] init: nmbd main process (7444) killed by TERM signal
[331301.380650] init: nmbd main process (7792) killed by TERM signal
[331301.888583] init: iscsi_pluginserverd main process (7379) killed by TERM signal
[331301.930181] init: iscsi_pluginengined main process (7362) killed by TERM signal
[331302.090035] init: scsi_plugin_server main process (7397) killed by TERM signal
[331302.171871] init: nmbd main process (8537) killed by TERM signal
[331302.384498] init: iscsi_pluginserverd main process (8492) killed by TERM signal
[331302.396767] init: iscsi_pluginengined main process (8454) killed by TERM signal
[331302.456069] init: scsi_plugin_server main process (8535) killed by TERM signal
[331302.928313] init: iscsi_pluginserverd main process (8684) killed by TERM signal
[331302.940520] init: iscsi_pluginengined main process (8673) killed by TERM signal
[331302.988339] init: scsi_plugin_server main process (8700) killed by TERM signal
[331303.170699] init: synowsdiscoveryd main process ended, respawning
[331306.099437] init: scsi_plugin_server main process (8904) killed by TERM signal
[331313.696390] init: synowsdiscoveryd main process ended, respawning
[331315.384498] r8152 2-2:1.0 eth2: carrier off
[331316.877880] init: scsi_plugin_server main process (9555) killed by TERM signal
[331317.050543] init: nmbd main process (10015) killed by TERM signal
[... 1634 lines cut ...]
ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:B4:BC:FE
          inet addr:169.254.41.218  Bcast:169.254.255.255  Mask:255.255.0.0
          inet6 addr: fe80::211:32ff:feb4:bcfe/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:7987 errors:0 dropped:0 overruns:0 frame:0
          TX packets:21978 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:2653726 (2.5 MiB)  TX bytes:32332964 (30.8 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00
          inet addr:10.10.112.19  P-t-P:10.10.112.19  Mask:255.255.255.0
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:13299 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19208 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100
          RX bytes:9992116 (9.5 MiB)  TX bytes:2448587 (2.3 MiB)

````

bb-qq, 2022-01-10:
It seems to run out of skb buffer in the kernel. So could you try to increase `vm.min_free_kbytes`?

* read current value: `sysctl -n vm.min_free_kbytes`
* write new value: `sysctl -w vm.min_free_kbytes=XXXX`

nubbie61, 2022-01-10:
You’re STAR! Thanks a lot for your prompt response. Will check your advice out later on today and will be back to you.

Best 

Nick

nubbie61, 2022-01-10:
Hi, again. Sadly, that's not working. Originally it was 65MB, tried it at 128MB and then even at 256MB - no luck, even after reboot.

Perhaps you could point me to some other direction?

Thank you

bb-qq, 2022-01-11:
Did you still see warnings such as `warn_alloc_failed: 22439 callbacks suppressed` when you increased `min_free_kbytes`?
if so, could you paste outputs of `sudo cat /proc/slabinfo` and `sudo cat /proc/meminfo`?

nubbie61, 2022-01-13:
Hi. Thank you for your continuing support. Very much appreciated, indeed.

Sadly, with ver. 2.15.0.1 it still gives plenty of warn_alloc_failed errors even with 256MB.

Here are the outputs as requested:
- slabinfo:
- ======
slabinfo - version: 2.1
# name            <active_objs> <num_objs> <objsize> <objperslab> <pagesperslab> : tunables <limit> <batchcount> <sharedfactor> : slabdata <active_slabs> <num_slabs> <sharedavail>
nf_conntrack_4         0      0    280   14    1 : tunables   54   27    8 : slabdata      0      0      0
nf_conntrack_3         0      0    280   14    1 : tunables   54   27    8 : slabdata      0      0      0
nf_conntrack_2         0      0    280   14    1 : tunables   54   27    8 : slabdata      0      0      0
au_finfo               0      0    128   32    1 : tunables  120   60    8 : slabdata      0      0      0
au_icntnr              0      0    768    5    1 : tunables   54   27    8 : slabdata      0      0      0
au_dinfo               0      0    128   32    1 : tunables  120   60    8 : slabdata      0      0      0
cifs_request           5      5  16512    1    8 : tunables    8    4    0 : slabdata      5      5      0
cifs_inode_cache     361    365    808    5    1 : tunables   54   27    8 : slabdata     73     73      0
udf_inode_cache        0      0    800    5    1 : tunables   54   27    8 : slabdata      0      0      0
isofs_inode_cache      0      0    696    5    1 : tunables   54   27    8 : slabdata      0      0      0
tcm_loop_cmd_cache      0      0    952    4    1 : tunables   54   27    8 : slabdata      0      0      0
t10_alua_tg_pt_gp_cache      0      0    288   14    1 : tunables   54   27    8 : slabdata      0      0      0
kcached-jobs        1024   1050    136   30    1 : tunables  120   60    8 : slabdata     35     35      0
md2-raid5-self-heal-sh-v0    256    258   1224    3    1 : tunables   24   12    8 : slabdata     86     86      0
raid5-md2           4096   4096   1704    4    2 : tunables   24   12    8 : slabdata   1024   1024      0
sw_flow               48     60    600    6    1 : tunables   54   27    8 : slabdata     10     10      0
nf_conntrack_1       558    616    280   14    1 : tunables   54   27    8 : slabdata     44     44      0
nf_conntrack_expect      0      0    224   18    1 : tunables  120   60    8 : slabdata      0      0      0
exfat_inode_cache      0      0    832    4    1 : tunables   54   27    8 : slabdata      0      0      0
btrfs_delayed_data_ref      3     30    136   30    1 : tunables  120   60    8 : slabdata      1      1      0
btrfs_delayed_tree_ref     96    264    120   33    1 : tunables  120   60    8 : slabdata      8      8     84
btrfs_delayed_ref_head    126    306    240   17    1 : tunables  120   60    8 : slabdata     18     18     84
btrfs_inode_defrag      0      0     96   42    1 : tunables  120   60    8 : slabdata      0      0      0
btrfs_delayed_node    223    637    304   13    1 : tunables   54   27    8 : slabdata     49     49      0
btrfs_ordered_extent      2     18    424    9    1 : tunables   54   27    8 : slabdata      2      2      0
bio-1                140    216    320   12    1 : tunables   54   27    8 : slabdata     18     18      0
btrfs_extent_buffer   6648  12559    208   19    1 : tunables  120   60    8 : slabdata    661    661      0
btrfs_extent_state   4691  20750     80   50    1 : tunables  120   60    8 : slabdata    415    415     60
btrfs_path            27     72    112   36    1 : tunables  120   60    8 : slabdata      2      2      0
btrfs_transaction      1      8    496    8    1 : tunables   54   27    8 : slabdata      1      1      0
btrfs_trans_handle   3586   7608    168   24    1 : tunables  120   60    8 : slabdata    317    317      0
btrfs_inode        11223  19542   1232    3    1 : tunables   24   12    8 : slabdata   6514   6514      0
i915_dependency        0      0     64   63    1 : tunables  120   60    8 : slabdata      0      0      0
drm_i915_gem_request      0      0    576    7    1 : tunables   54   27    8 : slabdata      0      0      0
drm_i915_gem_object     33     55    768    5    1 : tunables   54   27    8 : slabdata     11     11      0
hfsplus_icache         0      0    960    4    1 : tunables   54   27    8 : slabdata      0      0      0
fuse_request           0      0    400   10    1 : tunables   54   27    8 : slabdata      0      0      0
fuse_inode             0      0    832    9    2 : tunables   54   27    8 : slabdata      0      0      0
fat_inode_cache        0      0    784    5    1 : tunables   54   27    8 : slabdata      0      0      0
fat_cache              0      0     32  124    1 : tunables  120   60    8 : slabdata      0      0      0
ecryptfs_key_record_cache      0      0    576    7    1 : tunables   54   27    8 : slabdata      0      0      0
ecryptfs_sb_cache      0      0   1152    7    2 : tunables   24   12    8 : slabdata      0      0      0
ecryptfs_inode_cache      0      0   1088    7    2 : tunables   24   12    8 : slabdata      0      0      0
ecryptfs_file_cache      0      0     16  240    1 : tunables  120   60    8 : slabdata      0      0      0
ecryptfs_auth_tok_list_item      0      0    832    9    2 : tunables   54   27    8 : slabdata      0      0      0
ip6-frags              0      0    200   20    1 : tunables  120   60    8 : slabdata      0      0      0
[... cut ...]
Shmem:             31360 kB
Slab:             400996 kB
SReclaimable:     209960 kB
SUnreclaim:       191036 kB
KernelStack:       10320 kB
PageTables:        52884 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:     6091956 kB
Committed_AS:    4386408 kB
VmallocTotal:   34359738367 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
DirectMap4k:       66256 kB
DirectMap2M:     8142848 kB
DirectMap1G:     2097152 kB

I do look forward for your valuable support

chmooreck, 2022-01-21:
got same issue
DS1019+
DSM 7.0.1-42218 Update 2
$ uname -r
4.4.180+

UNITEK USB-A to 2.5G

$ cat /proc/meminfo 
MemTotal:        7989460 kB
MemFree:          857700 kB
MemAvailable:    5488336 kB
Buffers:           37076 kB
Cached:          6137544 kB
SwapCached:          156 kB
Active:          3182980 kB
Inactive:        3454332 kB
Active(anon):     257560 kB
Inactive(anon):   380496 kB
Active(file):    2925420 kB
Inactive(file):  3073836 kB
Unevictable:        1312 kB
Mlocked:            1312 kB
SwapTotal:       6893484 kB
SwapFree:        6890456 kB
Dirty:             84340 kB
Writeback:             0 kB
AnonPages:        463908 kB
Mapped:           255256 kB
Shmem:            174024 kB
Slab:             289652 kB
SReclaimable:     222780 kB
SUnreclaim:        66872 kB
KernelStack:       10960 kB
PageTables:        52212 kB
NFS_Unstable:          0 kB
Bounce:                0 kB
WritebackTmp:          0 kB
CommitLimit:    10888212 kB
Committed_AS:    3073020 kB
VmallocTotal:   34359738367 kB
VmallocUsed:           0 kB
VmallocChunk:          0 kB
DirectMap4k:       23248 kB
DirectMap2M:     1894400 kB
DirectMap1G:     8388608 kB

[278308.393783] warn_alloc_failed: 1 callbacks suppressed
[278308.399533] swapper/0: page allocation failure: order:4, mode:0x2084020
[278308.407023] CPU: 0 PID: 0 Comm: swapper/0 Tainted: P           OE   4.4.180+ #42218
[278308.415659] Hardware name: Synology DS1019+/DS1019+, BIOS M.412 04/01/2020
[278308.423432]  0000000000000000 ffff88027fc03c20 ffffffff812ef89d 0000000002084020
[278308.431801]  0000000000000004 ffff88027fc03ca8 ffffffff81137b67 ffffffff818c35f0
[278308.440170]  0208402000000000 0000000000000010 0000000000000002 00000000000003c0
[278308.448540] Call Trace:
[278308.451360]  <IRQ>  [<ffffffff812ef89d>] dump_stack+0x4d/0x70
[278308.457886]  [<ffffffff81137b67>] warn_alloc_failed+0xd7/0x130
[278308.464497]  [<ffffffff8113ad4a>] __alloc_pages_nodemask+0x92a/0xaf0
[278308.471694]  [<ffffffffa0c9fd15>] alloc_rx_agg+0x65/0x130 [r8152]
[278308.478594]  [<ffffffffa0cca954>] r8152_poll+0x304/0x860 [r8152]
[278308.485390]  [<ffffffff814ce4bb>] net_rx_action+0xeb/0x320
[278308.491612]  [<ffffffff81056d5c>] __do_softirq+0xdc/0x300
[278308.497728]  [<ffffffff810570dc>] irq_exit+0x9c/0xb0
[278308.503366]  [<ffffffff81007137>] do_IRQ+0x87/0xe0
[278308.508812]  [<ffffffff8157a75b>] common_interrupt+0x9b/0x9b
[278308.515224]  <EOI>  [<ffffffff8100e1ad>] ? default_idle+0x2d/0x100
[278308.522229]  [<ffffffff8100efb0>] arch_cpu_idle+0x10/0x20
[278308.528351]  [<ffffffff8108eb68>] default_idle_call+0x28/0x40
[278308.534853]  [<ffffffff8108ed25>] cpu_startup_entry+0x1a5/0x340
[278308.541549]  [<ffffffff81574442>] rest_init+0x72/0x80
[278308.547284]  [<ffffffff818f1e92>] start_kernel+0x3e8/0x3f5
[278308.553503]  [<ffffffff818f12f3>] x86_64_start_reservations+0x2a/0x2c
[278308.560789]  [<ffffffff818f13fb>] x86_64_start_kernel+0x106/0x113
[278308.567744] Mem-Info:
[278308.570404] active_anon:64026 inactive_anon:95129 isolated_anon:0
                 active_file:729718 inactive_file:751514 isolated_file:0
                 unevictable:328 dirty:22909 writeback:0 unstable:0
                 slab_reclaimable:58355 slab_unreclaimable:16759
                 mapped:63793 shmem:43506 pagetables:13160 bounce:0
                 free:230352 free_pcp:907 free_cma:0
[278308.609551] DMA free:15796kB min:1036kB low:1292kB high:1552kB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:15888kB managed:15796kB mlocked:0kB dirty:0kB writeback:0kB mapped:0kB shmem:0kB slab_reclaimable:0kB slab_unreclaimable:0kB kernel_stack:0kB pagetables:0kB unstable:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? yes
[278308.655792] lowmem_reserve[]: 0 1777 7780 7780
[278308.660889] DMA32 free:300108kB min:119872kB low:149840kB high:179808kB active_anon:25040kB inactive_anon:81056kB active_file:702180kB inactive_file:611148kB unevictable:440kB isolated(anon):0kB isolated(file):0kB present:1901264kB managed:1826748kB mlocked:440kB dirty:2196kB writeback:0kB mapped:63592kB shmem:29932kB slab_reclaimable:43992kB slab_unreclaimable:15428kB kernel_stack:1952kB pagetables:11212kB unstable:0kB bounce:0kB free_pcp:1812kB local_pcp:312kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[278308.713434] lowmem_reserve[]: 0 0 6002 6002
[278308.718241] Normal free:604868kB min:403372kB low:504212kB high:605056kB active_anon:231116kB inactive_anon:299460kB active_file:2216704kB inactive_file:2394896kB unevictable:872kB isolated(anon):0kB isolated(file):0kB present:6291456kB managed:6146916kB mlocked:872kB dirty:89440kB writeback:0kB mapped:191584kB shmem:144092kB slab_reclaimable:189428kB slab_unreclaimable:51608kB kernel_stack:8944kB pagetables:41428kB unstable:0kB bounce:0kB free_pcp:2428kB local_pcp:620kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[278308.771666] lowmem_reserve[]: 0 0 0 0
[278308.775871] DMA: 3*4kB (U) 3*8kB (U) 1*16kB (U) 2*32kB (U) 3*64kB (U) 1*128kB (U) 0*256kB 0*512kB 1*1024kB (U) 1*2048kB (U) 3*4096kB (M) = 15796kB
[278308.790755] DMA32: 32549*4kB (UME) 14939*8kB (UME) 3060*16kB (UM) 44*32kB (M) 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 300076kB
[278308.805614] Normal: 87692*4kB (UME) 27984*8kB (UME) 1866*16kB (UM) 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 604496kB
[278308.820078] 1525362 total pagecache pages
[278308.824656] 39 pages in swap cache
[278308.828552] Swap cache stats: add 2289, delete 2250, find 106/233
[278308.835458] Free swap  = 6890456kB
[278308.839353] Total swap = 6893484kB
[278308.843247] 2052152 pages RAM
[278308.846659] 0 pages HighMem/MovableOnly
[278308.851038] 54787 pages reserved

$ sysctl -n vm.min_free_kbytes                  
524288

bb-qq, 2022-01-23:
What is happening is a memory allocation error, but it also looks like this error is caused by the Ethernet device hanging up. So can you try a different USB port to connect to, or try a device from a different vendor? It may be unlikely that doing so will improve the situation, though.

chmooreck, 2022-01-23:
Connecting it to front USB is no-go: it tries to initialize and then restarts - I think it can be power issue.

I've got same issue with Club 3D adapter (it's same revision chip on both adapters - v14).

I'll try to use some powered USB HUB (got to buy it) and let you know.

chmooreck, 2022-01-25:
After connect thru (cheap) powered USB hub:

[181273.119257] r8152 2-1:1.0 eth2: v2.15.0 (2021/04/15)
[181273.124910] r8152 2-1:1.0 eth2: This product is covered by one or more of the following patents:
                		US6,570,884, US6,115,776, and US6,327,625.

[181273.142602] r8152 2-1:1.0 eth2: chip rev 14
[181291.061615] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[181291.313944] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[181291.321360] r8152 2-1:1.0 eth2: carrier on
[181291.485024] warn_alloc_failed: 1 callbacks suppressed
[181291.490800] synow3tool: page allocation failure: order:4, mode:0x2084020
[181291.498401] CPU: 0 PID: 20469 Comm: synow3tool Tainted: P           OE   4.4.180+ #42218
[181291.507537] Hardware name: Synology DS1019+/DS1019+, BIOS M.412 04/01/2020
[181291.515310]  0000000000000000 ffff88027fc03c20 ffffffff812ef89d 0000000002084020
[181291.523680]  0000000000000004 ffff88027fc03ca8 ffffffff81137b67 ffffffff818c35f0
[181291.532051]  0208402000000000 0000000000000010 ffffffff818b4f00 00000000000003c0
[181291.540425] Call Trace:
[181291.543248]  <IRQ>  [<ffffffff812ef89d>] dump_stack+0x4d/0x70
[181291.549768]  [<ffffffff81137b67>] warn_alloc_failed+0xd7/0x130
[181291.556380]  [<ffffffff8113ad4a>] __alloc_pages_nodemask+0x92a/0xaf0
[181291.563576]  [<ffffffff81507260>] ? inet_del_offload+0x40/0x40
[181291.570189]  [<ffffffff814cdb2b>] ? __netif_receive_skb_core+0x39b/0x930
[181291.577777]  [<ffffffffa0c1fd15>] alloc_rx_agg+0x65/0x130 [r8152]
[181291.584682]  [<ffffffffa0c4a954>] r8152_poll+0x304/0x860 [r8152]
[181291.591490]  [<ffffffff814ce4bb>] net_rx_action+0xeb/0x320
[181291.597715]  [<ffffffff81056d5c>] __do_softirq+0xdc/0x300
[181291.603842]  [<ffffffff810570dc>] irq_exit+0x9c/0xb0
[181291.609481]  [<ffffffff81007137>] do_IRQ+0x87/0xe0
[181291.614927]  [<ffffffff8157a75b>] common_interrupt+0x9b/0x9b
[181291.621343]  <EOI> Mem-Info:
[181291.624728] active_anon:61050 inactive_anon:92472 isolated_anon:0
                 active_file:830735 inactive_file:842842 isolated_file:0
                 unevictable:288 dirty:18069 writeback:0 unstable:0
                 slab_reclaimable:46277 slab_unreclaimable:16609
                 mapped:57608 shmem:42019 pagetables:13286 bounce:0
                 free:55816 free_pcp:893 free_cma:0
[181291.663812] DMA free:15796kB min:128kB low:160kB high:192kB active_anon:0kB inactive_anon:0kB active_file:0kB inactive_file:0kB unevictable:0kB isolated(anon):0kB isolated(file):0kB present:15888kB managed:15796kB mlocked:0kB dirty:0kB writeback:0kB mapped:0kB shmem:0kB slab_reclaimable:0kB slab_unreclaimable:0kB kernel_stack:0kB pagetables:0kB unstable:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? yes
[181291.709778] lowmem_reserve[]: 0 1777 7780 7780
[181291.714921] DMA32 free:67568kB min:14984kB low:18728kB high:22476kB active_anon:32628kB inactive_anon:78280kB active_file:768888kB inactive_file:772160kB unevictable:644kB isolated(anon):0kB isolated(file):0kB present:1901264kB managed:1826748kB mlocked:644kB dirty:4132kB writeback:0kB mapped:51132kB shmem:32620kB slab_reclaimable:37004kB slab_unreclaimable:14840kB kernel_stack:3968kB pagetables:10940kB unstable:0kB bounce:0kB free_pcp:2384kB local_pcp:712kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[181291.767110] lowmem_reserve[]: 0 0 6002 6002
[181291.771946] Normal free:138252kB min:50420kB low:63024kB high:75628kB active_anon:210524kB inactive_anon:291608kB active_file:2554060kB inactive_file:2600232kB unevictable:508kB isolated(anon):0kB isolated(file):0kB present:6291456kB managed:6146916kB mlocked:508kB dirty:69188kB writeback:0kB mapped:179252kB shmem:135456kB slab_reclaimable:148260kB slab_unreclaimable:51892kB kernel_stack:8816kB pagetables:41808kB unstable:0kB bounce:0kB free_pcp:2256kB local_pcp:208kB free_cma:0kB writeback_tmp:0kB pages_scanned:0 all_unreclaimable? no
[181291.825111] lowmem_reserve[]: 0 0 0 0
[181291.829383] DMA: 3*4kB (U) 3*8kB (U) 1*16kB (U) 2*32kB (U) 3*64kB (U) 1*128kB (U) 0*256kB 0*512kB 1*1024kB (U) 1*2048kB (U) 3*4096kB (M) = 15796kB
[181291.844273] DMA32: 1363*4kB (UME) 4229*8kB (UME) 1743*16kB (UM) 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 67172kB
[181291.858357] Normal: 22196*4kB (UME) 6119*8kB (M) 0*16kB 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 137736kB
[181291.871746] 1716145 total pagecache pages
[181291.876325] 0 pages in swap cache
[181291.880126] Swap cache stats: add 4, delete 4, find 0/0
[181291.886062] Free swap  = 6893468kB
[181291.889959] Total swap = 6893484kB
[181291.893858] 2052152 pages RAM
[181291.897269] 0 pages HighMem/MovableOnly
[181291.901649] 54787 pages reserved

It throws it constantly.

$ dmesg | grep warn_alloc_failed
[181291.485024] warn_alloc_failed: 1 callbacks suppressed
[181291.549768]  [<ffffffff81137b67>] warn_alloc_failed+0xd7/0x130
[181297.608486] warn_alloc_failed: 80 callbacks suppressed
[181297.672730]  [<ffffffff81137b67>] warn_alloc_failed+0xd7/0x130
[181303.283996] warn_alloc_failed: 31 callbacks suppressed
[181303.348212]  [<ffffffff81137b67>] warn_alloc_failed+0xd7/0x130
[181313.460549] warn_alloc_failed: 6 callbacks suppressed
[181313.524671]  [<ffffffff81137b67>] warn_alloc_failed+0xd7/0x130
[181323.627611] warn_alloc_failed: 4 callbacks suppressed
[181323.691737]  [<ffffffff81137b67>] warn_alloc_failed+0xd7/0x130
[... cut ...]
eth2      Link encap:Ethernet  HWaddr 00:E0:4C:3F:B5:50  
          inet addr:192.168.1.117  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe3f:b550/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:332 errors:0 dropped:0 overruns:0 frame:0
          TX packets:57 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:77452 (75.6 KiB)  TX bytes:11980 (11.6 KiB)

Its connected at 2500Mb/s.
Didn't do any transfer test yet.

chmooreck, 2022-01-26:
r8169 in 4.4 uses kmalloc and mapping to dma. I feel I'm too short to make such changes.

In contrary r8152 4.4 allocates all memory on device open, tp->rx_buf_sz = 16k instead of 32k needed for v14 chip...

It all looks more messy than I thought.

bb-qq, 2022-01-26:
Isn't `rx_buf_sz` 48kb for `chip rev=14`?
Indeed the amount of memory usage will vary depending on the device by the chip revision.

I've prepared a special version that reduces this `rx_buf_sz` to 16kb, so do you try it? This change should make the page allocator less fragmented.

However, I don't have a RealTek adapter on hand at the moment, so I can only confirm that it is loaded successfully. Please try if you like.
https://github.com/bb-qq/r8152/releases/download/2.15.0-5/r8152-apollolake-2.15.0-6_rxbuf-test.spk

chmooreck, 2022-01-26:
> Isn't `rx_buf_sz` 48kb for `chip rev=14`?
https://github.com/bb-qq/r8152/blob/18b4d79e2ea822dfe10d30b99e8925ed0fec7f9b/r8152.c#L19088

Isn't it that amount of space required ?

bb-qq, 2022-01-26:
Confusingly, chip rev=14 is `RTL_VER_13`.
https://github.com/bb-qq/r8152/blob/master/r8152.c#L981-L1002

chmooreck, 2022-01-26:
Thanks for clarification.

16kB version seems working without issues (warnings).
Fast test (copy few GB file under mc using nfs) shows performance increase (from ~180Mb/s yesterday to ~220MB).

chmooreck, 2022-01-26:
If there there is any potential improvement we could try with 32kB too - as I understand 32kB page should be enough for it ?

bb-qq, 2022-01-26:
I remember that 220Mb/s throughput is almost maximum at MTU=1500 configuration. So if you want further performance, try to increase MTU to 9k.

nubbie61, 2022-01-28:
Is there a chance to have this test driver for testing on DSM6.2?

 

https://github.com/bb-qq/r8152/releases/download/2.15.0-5/r8152-apollolake-2.15.0-6_rxbuf-test.spk

bb-qq, 2022-02-05:
At this time, I am not considering releasing new drivers for DSM6.2. However, if the change is effective in other cases, I may release drivers for older models for which DSM7 is not provided.

Since DSM7 is already stable, I would prioritize reducing the number of variations in the environments it supports.

nubbie61, 2022-02-05:
Thank you very much for your time and efforts.

 

I’ll keep looking here as all my equipment I’ve just bought for that is now sitting still. So, hopefully, one day I’ll get working solution from here for my adapters.

 

Keep up the good work

 

Nick

bb-qq, 2022-02-05:
Thank you for understanding. By the way, I'm curious why you don't upgrade to DSM7, could you explain that?

nubbie61, 2022-02-05:
Plex is my answer. Still tons of troubles out there. Even on 7.0.1. Will wait for a while before upgrading.

closed 2025-01-02 by bb-qq