bb-qq/r8152 #183: ver 2.15.0-5 and 2.15.0-4 not work for DS220j

jcs130, 2022-01-12:
[image]
it works when not copy data but once I copy data from NAS, the network will break.
But if I switch the cable to the orignal port  or plug in the cable again it will work

bb-qq, 2022-01-13:
Some users reported issues with less memory(<1GB) models. It might be avoided by stopping services, but I'm not sure.

jcs130, 2022-01-15:
Thanks. It's possable working on a 512MB ram devide? The DS220J cannot add ram

Long0529-on9, 2022-01-21:
my ds416play upgraded to 8gb of ram but happened to have the same issue. Is there any chance it's related to the lack of power of usb port/ bad usb port?

bb-qq, 2022-01-21:
@jcs130 , In a couple of cases, the driver didn't seem to be working because it failed to allocate memory in each case. I can't say for sure without looking at your logs, but it is possible that you are experiencing the same problem.

@Long0529-on9 , it is possible that the USB port lacks power. If you have a Y-Cable or a self-powered USB hub with an AC adapter, try that.

However, Synology's hardware seems to have individual differences, so I don't know whether these methods will solve the problem.

bb-qq, 2022-02-05:
I created an experimental driver with low memory consumption. Could you try this?
https://github.com/bb-qq/r8152/releases/download/2.15.0-5/r8152-rtd1296-2.15.0-6_rxbuf-test.spk

grsdmz, 2022-02-18:
Thank you for your update.I meet the same problem as jcs130. I installed 2.15.0-6 on my DS220J but the network broke down like ver 0-5 at every time I copy large file.

bb-qq, 2022-03-13:
I created another driver. Could you try this?
https://github.com/bb-qq/r8152/releases/download/2.15.0-6/r8152-rtd1296-2.15.0-6nolpm.spk

grsdmz, 2022-03-13:
It seems something wrong, It can't be installed and report  " Invalid file format. Please contact the package developer. "

jcs130, 2022-03-14:
> 

Thanks I installed the new version success. I will try it after I go home

bb-qq, 2022-04-10:
Sorry, the previous file is invalid. I uploaded the fixed one.
https://github.com/bb-qq/r8152/releases/download/2.15.0-6/r8152-rtd1296-2.15.0-7nolpm2.spk

robbel, 2022-05-21:
@bb-qq is there also a version of this low-memory one for 6.2.x? Thanks!

Sjekke, 2022-05-21:
Why installing a 2,5gbit adapter on slow old Nas'ses? You can't make a Porsche from an Trabant.

bb-qq, 2022-05-28:
Sorry, I have no plans to release a new driver for 6.2 at this time. This is due to resource constraints on my development machine and to reduce the number of environments to troubleshoot.

Jachub123, 2022-06-05:
thanks for your work. I have a D420j running on "DSM 7.1-42661 Update 2" also with a rtd1296 cpu. i cant find anything about it in the driver compatibility list. What driver would you recommend for me to install on my 420j.
Thanks in advance

NikitaOsotsky, 2022-07-28:
Hey! No one has written whether the last proposed driver works correctly. I have the same problem with my DS218
Guys respond. Was it possible to solve this problem)

> https://github.com/bb-qq/r8152/releases/download/2.15.0-6/r8152-rtd1296-2.15.0-7nolpm2.spk

bb-qq, 2022-08-11:
At the moment I do not have any ideas to improve the situation....

NikitaOsotsky, 2022-08-11:
> At the moment I do not have any ideas to improve the situation....

Is there anything I can do to help? Make some tests on my system, and collect some logs?

bb-qq, 2022-08-25:
The latest release [2.16.3-1](https://github.com/bb-qq/r8152/releases/tag/2.16.3-1) includes a few stability-enhancing changes by Realtek.

It is unlikely to be related to this issue, but it may be worth a try.

cqwangding, 2022-11-22:
> The latest release [2.16.3-1](https://github.com/bb-qq/r8152/releases/tag/2.16.3-1) includes a few stability-enhancing changes by Realtek.
> 
> It is unlikely to be related to this issue, but it may be worth a try.

Tried the new release on my ds220J, the similar issue was noticed. One thing I should mention is that I can copy to the SMB with speed 200+MB/s without any issue (I copied more than 10G of data). However, when I tried to copy from the SMB, the network is down immediately. I feel like it is not the power issue (as I can copy the files for a long time and no heat issue was noticed). I reboot and reinstall it several times. The same issue remains.

bb-qq, 2022-12-04:
Thank you for trying the test driver.

I decided to merge topics on rtd1296 into issue #275.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq

YohoYue, 2025-01-10:
I'm a Mac user. but I used an old PC to helped. I have DS220j NAS. before I install 2.5G driver, I updated DSM 7 to DSM7.2,(maybe I shouldn't update DSM);  r8152-rtd1296-2.19.2-1_7.2.spk driver installed. Show up both 1G and 2.5G connections with 2 IPs. but 2.5G having hard time to connect, some time works I got double speed of writing but 0 MB reading then stuck and connecting drops. I tried Ugreen 2.5G adapter and Plugable 2.5G adapter. Same problem. The 2.5G doesn't show up on my network. I have to find the 2.5G connection IP address to manual connect it. I have to keep both 1G and 2.5G connections. 2.5G is not really working yet. Some time my router don't see 2.5G IP, some time sees it (both 1G & 2.5G) .. Some time I can mount the 2.5G and able to drop files to my NAS, but when you try to READ the NAS files, connection will drop after few minutes.

YohoYue, 2025-01-10:
> Tried the new release on my ds220J, the similar issue was noticed. One thing I should mention is that I can copy to the SMB with speed 200+MB/s without any issue (I copied more than 10G of data). [...]

[image]
When I did Blackmagic Disk Speed Test: I got 185MB/s WRITE, but 0MB/s READ.  WRITE is great, but it will crash every-time It try to READ.