bb-qq/r8152 #185: Openvswitch 6.2

DJGIG83, 2022-01-15:
Could you please also make a spk. for DSM 6.2 like this : r8152-broadwell-2.15.0-5.spk ?
becouse i need the support of openvswitch for my VMM.
thx.

bb-qq, 2022-01-21:
Type this command to update startup script to latest version which supports openvswitch:
`sudo curl -o /var/packages/r8152/scripts/start-stop-status https://raw.githubusercontent.com/bb-qq/r8152/master/scripts/start-stop-status`

DJGIG83, 2022-01-25:
I did it, as you told me, but it doesn`t function :( The "LAN 2" (which the usb - ethernet port is ) is visible under "Network" but not under the VMM. Just the LAN 1 which is the Onboard lan card.
It is also not possibly to make a link Aggregation with the 2 Lan Ports. I think because the LAN 2 doesn't work already with openvswitch and the Lan 1 is already used with openvswich in my VMM. so i can`t bound them together :(

bb-qq, 2022-01-26:
Did you reboot your NAS? If so, please fill out the bug report template. Especially I want to see the output of `dmesg`.

DJGIG83, 2022-01-26:
Yes i was rebooting my nas.
B.R
dmesg ----
[dmesg.txt](https://github.com/bb-qq/r8152/files/7942952/dmesg.txt)
>

bb-qq, 2022-02-05:
It looks like `eth1` is the realtek interface, and the `ovs_eth1` interface for the Open vSwitch is created without any problems.
What will happen once I disable and re-enable the Open vSwitch feature?

DJGIG83, 2022-02-05:
Thank you very much :) I`ve deleted the VMM and re-installed it. Now everythink works fine i`ve bounded the two (eth0 & eth1) Thx for your work !!!

bb-qq, 2022-02-05:
Just in case, it may be better if you confirm ovs settings persist over reboot.

DJGIG83, 2022-02-05:
Damed :( After reboot my DS shows me 1 Bond Network and !!!! LAN 2

DJGIG83, 2022-02-08:
After a long time, i have bought a new usb interface ( Club 3D USB 3.2 Gen1 Adapter Typ-A zu 2, 5 Gigabit ) and replaced it to my old no-name product. Now everything works fine . Even the bound ( LAG ). Also with OpenvSwitch ! I also shutdown and rebooted my NAS several times ! -----> it works :) 
Thanks for your great work.

closed 2022-02-08 by DJGIG83

bb-qq, 2022-02-12:
Could you paste the outputs of both adapters again? I'm curious about the difference between these. Logically these should not have any differences.