bb-qq/r8152 #187: RTD1296 platform supported

malfunctioningEddy, 2022-01-20:
Hi,

First of all : reat job expanding these nasses to 2.5 GB!
I saw from issue 136 that there are problems with the RTD1296 platform.
You're still releasing drivers for that platform however. Did someone ever get the driver running on a RTD1296 platform?

To summarize the reports in this forum, it seems that the USB controller of the RTD1296 platform has a flaw that prevents this driver from working stably.

It's odd that it doesn't work with the RealTek hardware, but there doesn't seem to be anything I can do about it at the moment. Sorry.

_Originally posted by @bb-qq in https://github.com/bb-qq/r8152/issues/136#issuecomment-903207112_

bb-qq, 2022-01-22:
Some users reported the driver for rtd1296 is working in this thread. So I decided to continue uploading the driver although rtd1296 seems to have many issues.
https://github.com/bb-qq/r8152/issues/152

Also, rtd1296 is a current platform, so stability may be improved in future hardware.

malfunctioningEddy, 2022-01-24:
Hi,

Thanks for getting back.
Was planning to get a 418 but now I'm not so sure. 

Think I now will just get a 920+ to be sure it works.

Again, thanks for the great work!

Regards,

Erik

closed 2022-01-24 by malfunctioningEddy