bb-qq/r8152 #188: DS918+ Issue with RTL8153 USB NIC (UNI brand) data flow stops after appox 1 hour

jterranella, 2022-01-21:
### Description of the problem

Driver installs fine after running terminal command. The additional NIC shows up and allows configuration within the DSM. Everything works for about an hour and then traffic stops flowing through the USB NIC. It still shows in the network settings as LAN 3 and connected, but cannot ping or transfer information. Have to stop the RTL Service and restart it to restore connectivity. The NIC is plugged in to the front USB3 port on the NAS.

### Description of your products

>     - DS918+ 
>     - Linux TerraSyno 4.4.180+ #42218 SMP Mon Oct 18 19:17:56 CST 2021 x86_64 GNU/Linux synology_apollolake_918+
>     - DSM 7.0.1-42218
>     - uni Driver Free USB 3.0 to 100/1000 Gigabit Ethernet LAN Network Adapter (RTL8153 chipset)

### Description of your environment

>     - NAS to PC (SAN connection)
>     - PC: Going in to the same type of USB3.0 NIC. , Micro Dell with Intel i7 8700T, 16Gb Ram, Win 10 Enterprise
>     - Hub: N/A
>     - cable category CAT6 - 3ft

### Output of ``dmesg`` command

````
[270683.502781] r8152 2-1:1.0 eth2: Tx status -71
[270683.511274] r8152 2-1:1.0 eth2: Tx status -71
[270685.506882] r8152 2-1:1.0 eth2: Tx status -71
[270685.515365] r8152 2-1:1.0 eth2: Tx status -71
[270705.531790] r8152 2-1:1.0 eth2: Tx status -71
[270985.835766] r8152 2-1:1.0 eth2: Tx status -71
[270987.839360] r8152 2-1:1.0 eth2: Tx status -71
[270987.847855] r8152 2-1:1.0 eth2: Tx status -71
[270989.843450] r8152 2-1:1.0 eth2: Tx status -71
[270989.851947] r8152 2-1:1.0 eth2: Tx status -71
[271291.495950] r8152 2-1:1.0 eth2: Tx status -71
[271293.499982] r8152 2-1:1.0 eth2: Tx status -71
[271293.508484] r8152 2-1:1.0 eth2: Tx status -71
[271295.504073] r8152 2-1:1.0 eth2: Tx status -71
[271295.512578] r8152 2-1:1.0 eth2: Tx status -71
[271425.653989] r8152 2-1:1.0 eth2: Tx status -71
[271480.231964] r8152 2-1:1.0 eth2: Tx status -71
[271541.230223] r8152 2-1:1.0 eth2: Tx status -71
[271595.818463] r8152 2-1:1.0 eth2: Tx status -71
[271597.699877] usb 1-1: usbfs: USBDEVFS_CONTROL failed cmd usbhid-ups rqt 128 rq 6 len 255 ret -71
[271597.828552] r8152 2-1:1.0 eth2: Tx status -71
[271597.837055] r8152 2-1:1.0 eth2: Tx status -71
[271599.832640] r8152 2-1:1.0 eth2: Tx status -71
[271599.841145] r8152 2-1:1.0 eth2: Tx status -71
[271895.892334] r8152 2-1:1.0 eth2: Tx status -71
[271897.896676] r8152 2-1:1.0 eth2: Tx status -71
[271897.905181] r8152 2-1:1.0 eth2: Tx status -71
[271899.901263] r8152 2-1:1.0 eth2: Tx status -71
[271899.909771] r8152 2-1:1.0 eth2: Tx status -71
[271935.399385] r8152 2-1:1.0 eth2: Tx status -71
[271936.399419] r8152 2-1:1.0 eth2: Tx status -71
[271937.399470] r8152 2-1:1.0 eth2: Tx status -71
[271938.399770] r8152 2-1:1.0 eth2: Tx status -71
[272080.245706] r8152 2-1:1.0 eth2: Tx status -71
[272141.253223] r8152 2-1:1.0 eth2: Tx status -71
[272152.686495] r8152 2-1:1.0 eth2: Tx status -71
[272202.746770] r8152 2-1:1.0 eth2: Tx status -71
[272204.757612] r8152 2-1:1.0 eth2: Tx status -71
[272204.766110] r8152 2-1:1.0 eth2: Tx status -71
[272206.762207] r8152 2-1:1.0 eth2: Tx status -71
[272206.770698] r8152 2-1:1.0 eth2: Tx status -71
[272380.263834] r8152 2-1:1.0 eth2: Tx status -71
[272441.269605] r8152 2-1:1.0 eth2: Tx status -71
[272505.383016] r8152 2-1:1.0 eth2: Tx status -71
[272507.387605] r8152 2-1:1.0 eth2: Tx status -71
[272507.396105] r8152 2-1:1.0 eth2: Tx status -71
[272509.391188] r8152 2-1:1.0 eth2: Tx status -71
[272509.399691] r8152 2-1:1.0 eth2: Tx status -71
[272805.946652] r8152 2-1:1.0 eth2: Tx status -71
[... 1553 lines cut ...]
eth2      Link encap:Ethernet  HWaddr 00:E0:4C:08:07:C2
          inet addr:10.0.5.10  Bcast:10.0.5.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe08:7c2/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:498 errors:0 dropped:0 overruns:0 frame:0
          TX packets:211 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:49164 (48.0 KiB)  TX bytes:76960 (75.1 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:1257746 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1257746 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:861791423 (821.8 MiB)  TX bytes:861791423 (821.8 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2022-01-22:
Are you using the front port?

jterranella, 2022-01-22:
> Are you using the front port?

I am.

bb-qq, 2022-01-23:
Oh sorry, you already wrote about connecting port.

I suspect a problem on the ethernet adapter side, as there are many reports of it working with the DS918+. (Eg. design of power line, overheat, etc.)
https://github.com/bb-qq/r8152/wiki/Compatibility

Could you try another vendor? Also using a USB Hub with an external power supply might improve the situation.

bb-qq, 2024-12-31:
I would like to close this Issue once because it is old and the version of the driver we are talking about is also old. If you have any other questions, feel free to re-open.

closed 2024-12-31 by bb-qq