bb-qq/r8152 #189: [Success] (potentially slow) DS416 Alpine4k, Cable Matters 2.5GB USB 3, DSM 7.0.1-42218

ugzzugzz, 2022-01-23:
Synology DS416, 1GB Alpine4k model. Success, but speed is low.
Cable Matters 2.5GB USB 3. Model Number: 202095
DSM: 7.0.1-42218
Driver version tested: 2.15.0-4 DSM7.0 

Notes for functionality (it appears -5 may solve some of this, but I tested -4): 
Must install and let fail, run the command via SSH as instructions show, but then I had to reboot and STOP and START the service. Some testing has been done and in my case I need to stop and start the service after each reboot for the USB stick to be detected.
Plus, the USB drive ONLY works in the Front USB port for me, it is not detected in either of the two rear ports no matter how many times I reboot and start / stop. Plugged into the front, it works first try.

Speed tests are only showing a max of 160MBps, running 3x IronWolf ST4000VN008-2DR166 drives in SHR. I have not tested with an SSD to see if it's the raid, or if the 1GB of ram is just slowing me down.  Tests will continue.

bb-qq, 2022-01-23:
Increase MTU to 9000 and set SMB protocol version to v3 may help to improve performance.

ugzzugzz, 2022-01-23:
> 

Thank you for posting. This increased my read speed from 120MB up to 165-170 in windows, and up to 195 from linux. A good improvement.