bb-qq/r8152 #190: DS918+ Issue with RTL8156 USB NIC (biaze) can not transfer data

bornkilled, 2022-01-27:
### Description of the problem
Installed correct and eth2 can work  in Setup。
But it can not transfer file when I copy from SMB 。
The RTL8156 USB NIC (biaze) connect on  back usb port 

### Description of your products

- DS918+ 
- Linux HomeStation 4.4.59+ #23824 SMP PREEMPT Tue Feb 12 16:52:55 CST 2019 x86_64 GNU/Linux synology_apollolake_918+
- DSM version 6.2.1-23824 Update 6
- biaze brand , 8156b chip

### Description of your environment

- NAS to PC with single LAN Cable
- PC: 1000M Realtek GbE  port on MSI B450M MotherBoard 
- Hub: N/A
- cable category (Cat.5e)

### Output of ``dmesg`` command

[ 1188.901421] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[ 1188.901873] r8152 2-1.1:1.0 eth1: carrier on
[ 1191.459255] init: nmbd main process (3213) killed by TERM signal
[ 1192.153100] init: iscsi_pluginserverd main process (3020) killed by TERM signal
[ 1192.160276] init: iscsi_pluginengined main process (3009) killed by TERM signal
[ 1192.202032] init: nmbd main process (3497) killed by TERM signal
[ 1192.890741] init: iscsi_pluginserverd main process (3613) killed by TERM signal
[ 1192.917323] init: iscsi_pluginengined main process (3601) killed by TERM signal
[ 1198.372498] r8152 2-1.1:1.0 eth1: carrier off
[ 1206.053544] r8152 2-1.1:1.0 eth1: carrier on
[ 1234.285636] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1234.289180] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1234.292673] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1234.299924] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1234.499098] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1234.920045] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1235.761030] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1237.445186] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1239.736946] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1240.809537] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1242.791736] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1242.795228] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1243.092678] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1243.701023] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1244.642495] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1244.645915] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1244.914919] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1245.406876] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1246.125289] net_ratelimit: 1 callbacks suppressed
[ 1246.125300] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1246.157721] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1246.161236] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1247.340436] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1247.545196] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1248.152972] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1248.265018] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1249.269921] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1249.750670] r8152 2-1.1:1.0 eth1: Tx status -71
[ 1250.263270] r8152 2-1.1:1.0 eth1: Tx status -71

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__1-1         174c:2074:0100 09  2.10  480MBit/s 100mA 1IF  (ASRock ASM107x ffffffccfffffffbffffffcbfffffffa) hub
  |__1-5         05e3:0608:8831 09  2.00  480MBit/s 100mA 1IF  ( ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-5.1     f400:f400:0100 00  2.00  480MBit/s 80mA 1IF  (Kingmax USB2.0 FlashDisk 020000000001C1)
    |__1-5.2     0665:5161:0002 00  1.10  1.5MBit/s 100mA 1IF  (INNO TECH USB to Serial ffffffd8ffffffb9ffffffd0ffffffa2)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__2-1         174c:3074:0100 09  3.00 5000MBit/s 8mA 1IF  (ASRock ASM107x ffffffccfffffffbffffffcbfffffff0) hub
    |__2-1.1     0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)

### Output of ``ifconfig -a`` command

docker0   Link encap:Ethernet  HWaddr 02:42:A4:FE:A4:03
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:a4ff:fefe:a403/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:496 errors:0 dropped:0 overruns:0 frame:0
          TX packets:524 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:52122 (50.9 KiB)  TX bytes:69643 (68.0 KiB)

docker2a9 Link encap:Ethernet  HWaddr A6:D1:0C:C9:78:8E
          inet6 addr: fe80::a4d1:cff:fec9:788e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:30 errors:0 dropped:0 overruns:0 frame:0
          TX packets:119 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:24452 (23.8 KiB)  TX bytes:13650 (13.3 KiB)

dockerd97 Link encap:Ethernet  HWaddr DA:50:FD:86:EB:46
          inet6 addr: fe80::d850:fdff:fe86:eb46/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:466 errors:0 dropped:0 overruns:0 frame:0
          TX packets:497 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:34614 (33.8 KiB)  TX bytes:67126 (65.5 KiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:95:0E:7F
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:594841 errors:0 dropped:0 overruns:0 frame:0
          TX packets:728818 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:532371006 (507.7 MiB)  TX bytes:675628186 (644.3 MiB)
          Interrupt:119 base 0xa000

eth1      Link encap:Ethernet  HWaddr 00:E0:4C:68:00:53
          inet addr:192.168.100.2  Bcast:192.168.100.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:53/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:461 errors:0 dropped:0 overruns:0 frame:0
          TX packets:377 errors:149 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:80189 (78.3 KiB)  TX bytes:110571 (107.9 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:120832 errors:0 dropped:0 overruns:0 frame:0
          TX packets:120832 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:16414343 (15.6 MiB)  TX bytes:16414343 (15.6 MiB)

ovs-syste Link encap:Ethernet  HWaddr BA:EA:9E:3F:12:0A
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:95:0E:7F
          inet addr:10.0.0.2  Bcast:10.0.0.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe95:e7f/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:543912 errors:0 dropped:0 overruns:0 frame:0
          TX packets:464193 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:521931015 (497.7 MiB)  TX bytes:675588522 (644.2 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap021132 Link encap:Ethernet  HWaddr E6:50:3A:FD:17:5E
          inet6 addr: fe80::e450:3aff:fefd:175e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:273 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5711 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:40448 (39.5 KiB)  TX bytes:1477263 (1.4 MiB)

bb-qq, 2022-02-05:
Are the symptoms the same when using the front port?

bornkilled, 2022-02-18:
yes ,it is has  the same problem when i plug it to the front port

bb-qq, 2022-03-06:
Could you try to detach all other USB devices and connect the ethernet adaptor only?

bornkilled, 2022-03-06:
I don't connect any usb device except the boot usb stick and the ethernet adaptor

Could you try to detach all other USB devices and connect the ethernet adaptor only?

bb-qq, 2022-03-13:
Your output of lsusb contains a device that I have never seen before. Perhaps you are using Xpenology?

bb-qq, 2024-06-24:
I would like to close this issue as I have not heard back from you in a long time.
If you have any questions, feel free to re-open.

closed 2024-06-24 by bb-qq