bb-qq/r8152 #191: ELECOM EDC-QUA3C-B (RTL8156B) does not work with DS720+

zshoei, 2022-01-31:
Thank you for your great work.

### Description of the problem

The driver installation seems to have completed successfully (no error messages shown),
but eth2 doesn't show up.

### Description of your products

- DS720+
- Linux xxxxxxxxxxxxx 4.4.180+ #42218 SMP Mon Oct 18 19:16:55 CST 2021 x86_64 GNU/Linux synology_geminilake_720+
- DSM 7.0.1-42218
- ELECOM EDC-QUA3C-B (RTL8156B chip) inserted to the front USB-A port

### Description of your environment

- NAS connected to a PC via a hub
- Hub: Planex FX2G-05EM
- cable category: Cat.5e

### Output of ``dmesg`` command

[   60.231200] loop: module loaded
[   60.334919] Synotify use 16384 event queue size
[   60.340120] Synotify use 16384 event queue size
[   61.388248] capability: warning: `nginx' uses 32-bit capabilities (legacy support in use)
[   61.446627] ata1.00 (slot 1): configured for UDMA/133
[   61.452300] ata1 (slot 1): EH complete
[   61.584443] ata2.00 (slot 2): configured for UDMA/133
[   61.590113] ata2 (slot 2): EH complete
[   62.128352] ext2: synoboot2 mounted, process=synopkicompatsy
[   62.140559] synoboot2 unmounted, process=synopkicompatsy
[   63.624311] iSCSI:target_core_rodsp_server.c:1027:rodsp_server_init RODSP server started, login_key(9009d010ac76).
[   63.644437] iSCSI:extent_pool.c:766:ep_init syno_extent_pool successfully initialized
[   63.665147] iSCSI:target_core_device.c:613:se_dev_align_max_sectors Rounding down aligned max_sectors from 4294967295 to 4294967288
[   63.678412] iSCSI:target_core_configfs.c:5446:target_init_dbroot db_root: cannot open: /etc/target
[   63.688575] iSCSI:target_core_lunbackup.c:361:init_io_buffer_head 512 buffers allocated, total 2097152 bytes successfully
[   63.749486] iSCSI:target_core_file.c:149:fd_attach_hba RODSP plugin for fileio is enabled.
[   63.758731] iSCSI:target_core_file.c:156:fd_attach_hba ODX Token Manager is enabled.
[   63.767405] iSCSI:target_core_multi_file.c:91:fd_attach_hba RODSP plugin for multifile is enabled.
[   63.777436] iSCSI:target_core_ep.c:795:ep_attach_hba RODSP plugin for epio is enabled.
[   63.786287] iSCSI:target_core_ep.c:802:ep_attach_hba ODX Token Manager is enabled.
[   63.923142] workqueue: max_active 1024 requested for vhost_scsi is out of range, clamping between 1 and 512
[   64.576893] ext2: synoboot1 mounted, process=syno_bios_log_c
[   64.664661] vfat: synoboot1 mounted, process=syno_bios_log_c
[   64.727119] synoboot1 unmounted, process=syno_bios_log_c
[   64.933987] r8152: module verification failed: signature and/or required key missing - tainting kernel
[   64.945074] usbcore: registered new interface driver r8152
[   65.099040] 8021q: 802.1Q VLAN Support v1.8
[   68.405233] fuse init (API version 7.23)

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65001C189E345A92)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         056e:4017:3100 00  3.20 5000MBit/s 304mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 001000001)

### Output of ``ifconfig -a`` command

eth0      Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx  
          inet addr:192.168.xx.xx  Bcast:192.168.xx.xx  Mask:255.255.252.0
          inet6 addr: fe80::xxxx:xxxx:xxxx:xxxx/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6694 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3189 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1527239 (1.4 MiB)  TX bytes:3013256 (2.8 MiB)
          Interrupt:98 base 0xc000  

eth1      Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx  
          inet addr:169.254.47.32  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:97 base 0x8000  

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:854 errors:0 dropped:0 overruns:0 frame:0
          TX packets:854 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:115176 (112.4 KiB)  TX bytes:115176 (112.4 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

chmooreck, 2022-01-31:
# ifconfig eth2 up
?

zshoei, 2022-01-31:
$ ifconfig eth2 up
eth2: ERROR while getting interface flags: No such device

chmooreck, 2022-01-31:
**#**

zshoei, 2022-01-31:
Thanks for the replies.
Yeah, of course I have performed the command by root.

$ sudo ifconfig eth2 up
Password: 
eth2: ERROR while getting interface flags: No such device

chmooreck, 2022-01-31:
Ok, now I see your device (056e:4017) is not listed in driver. I'm pretty sure @bb-qq will be able to add it. You just need some patience.

zshoei, 2022-01-31:
@chmooreck 
Thank you so much for the information!
Ok, I'll wait for some response from @bb-qq .

bb-qq, 2022-02-05:
I will release a new driver when I have completed the verification of the URB modifications, and I will add the VIDs together with it at that time. Please wait for a while.

zshoei, 2022-02-06:
@bb-qq 
Yes, I'm looking forward to the next release. Thank you!

bb-qq, 2022-03-06:
I uploaded new drivers. It would be appreciated if you could let me know whether this driver works.
https://github.com/bb-qq/r8152/releases/tag/2.15.0-6

zshoei, 2022-03-06:
Using the new driver (r8152-geminilake-2.15.0-6.spk), network connection via ELECOM EDC-QUA3C-B seems fine so far.
I need some time to check the stability and throughput. I will post more info here and also in the compatibility thread.
Thanks again for your great work.

bb-qq, 2022-03-06:
Thank you for your report. I close this issue.

closed 2022-03-06 by bb-qq