bb-qq/r8152 #192: Ugreen USB CARD always link up link down when PC shotdown

fox0933, 2022-01-31:
### Description of the problem

> Ds918 and [UGREEN 2.5G ](https://item.jd.com/68118817284.html), [PCI-E CARD](https://item.jd.com/67379933665.html).
Ugreen USB CARD insert back of DS918.   when i reboot PC,  DS918 log is so many tips about LANS link up | LAN3 link down.
after 1-2hours, the UGreen 2.5 Card is disconnect. it still in network interface UI, but can't work, even you retry pull out Ethernet cable. it still doesn't work.

### Description of your environment

> DS918 and DSM 7.0-41890

chenlei630, 2022-02-01:
The same problem, r8152-apollolake-2.15.0-4, DS918+, DSM7.0.1-42218 UGREEN USB A to 2.5G Ethernet Adapter Connected to Front USB port, DS918 log so many tips about LAN3 link up and LAN3 link down, after 1-2 hours power off PC, the Green 2.5 Card link down, and the connection between NAS and Ugreen card does not work. even pull out and pull in the net card, stop and start the driver, it doesn't work any more!

bb-qq, 2022-02-05:
Your symptoms seem similar to #145, do you see any similarities with your environment?

chenlei630, 2022-02-05:
Yes, it is similar to #145, but as I said in the #145, I couldn't  use the command to set the autoneg off at the speed 2500. The default setting with 2500 is autoneg on. Do you know how to set autoneg off at speed 2500 with MTU 9000? @bb-qq

Tezouro, 2022-02-06:
Did you tried to connect to the USB in the front?

chenlei630, 2022-02-07:
Yes I had tried front and back USB but the same problem.@Tezouro

> Did you tried to connect to the USB in the front?

fox0933, 2022-02-09:
#145, it can fixed and works fine;  The USB in the back can aslo works fine; My System is DSM7.0 and Ds918+

fox0933, 2022-02-09:
> Yes, it is similar to #145, but as I said in the #145, I couldn't use the command to set the autoneg off at the speed 2500. The default setting with 2500 is autoneg on. Do you know how to set autoneg off at speed 2500 with MTU 9000? @bb-qq

switch user root,  `sudo -i`, now you can vi edit privelleage

chenlei630, 2022-02-15:
> switch user root, `sudo -i`, now you can vi edit privelleage

all commands that I used are in the user root mode, but could not set autoneg off at the speed 2500.

chenlei630, 2022-02-15:
> all commands that I used are in the user root mode, but could not set autoneg off at the speed 2500.

as the situation in #145 if I change the MTU to 1500, the link up and link down will be reduced.

Payn, 2022-02-28:
Like i said in #145 try 
`ethtool -s eth2 speed 1000 duplex full autoneg off` 
while connected through that interface. In my case it turned autoneg off but didn't change the speed to 1000.

chenlei630, 2022-03-06:
> 

as I said, I can set the autoneg off with the speed 1000 duplex, but the speed changed to 1000 in my case. If I want to set the speed back to 2500, then autoneg will be set on automatically.

bb-qq, 2025-01-02:
Since there has been no response for a long time, this issue is closed. See also #145, a similar problem, which has been resolved.
Please feel free to re-open it if you have any questions.

closed 2025-01-02 by bb-qq