bb-qq/r8152 #193: DSM 6.2.4 Not installed.

SergeySergeevitch, 2022-02-02:
r8152-cedarview-2.15.0-5.spk DS1513+ DSM 6.2.4 Not installed. The package is intended for version 7.

fox0933, 2022-02-02:
yes， The package is intended for version 7。
please，pay attention to look out.

DSM 6.2.4  need use to old version

SergeySergeevitch, 2022-02-02:
Installed. My adapter is not defined.

https://aliexpress.ru/item/1005002637149626.html?gatewayAdapt=glo2rus&sku_id=12000021520723520

USB\VID_0BDA&PID_8156&REV_3100
USB\VID_0BDA&PID_8156

bb-qq, 2022-02-05:
Could you fill out [this template](https://github.com/bb-qq/r8152/blob/master/.github/ISSUE_TEMPLATE/custom.md)?

bb-qq, 2022-08-11:
I close this issue due to no response. Please feel free to reopen the issue when you post additional information.

closed 2022-08-11 by bb-qq