bb-qq/r8152 #194: Stuck on last step - "install: cannot start '/var/packages/r8152/target/r8152/spk_su': No such file or directory

zyzzma, 2022-02-02:
### Description of the problem

> I've done everything right but stuck on the last step when I input the command via ssh: sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su **/opt/sbin/spk_su**

### Description of your products

> DS920+
>     - and output of ``uname -a`` command ??
>     - DSM version 7.0.1-42218 
> - Cablematters 2.5gbe dongle

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: Realtek 2.5gbe
>     - Hub: model name
>  Cat 6

### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

fox0933, 2022-02-04:
Have you ever used "sudo -i"?

bb-qq, 2022-02-05:
Could you paste the output from this command?
`sudo cat /var/log/synopkg.log`

coldheater1, 2022-04-26:
I am having the same issue.  Here is my oupt from the command above.

2022/04/26 00:18:37 Acquire web-config for SynologyDrive when 0x0000 (done)
2022/04/26 00:18:37 Acquire synocrond for SynologyDrive when 0x0000 (ready)
2022/04/26 00:18:37 Acquire synocrond for SynologyDrive when 0x0000 (done)
2022/04/26 00:18:37 Acquire csp-rule for SynologyDrive when 0x0000 (ready)
2022/04/26 00:18:37 Acquire csp-rule for SynologyDrive when 0x0000 (done)
2022/04/26 00:18:37 Acquire content-extractor for SynologyDrive when 0x0000 (ready)
2022/04/26 00:18:37 Acquire content-extractor for SynologyDrive when 0x0000 (done)
2022/04/26 00:18:37 (system) trigger SynologyDrive 3.1.0-22920 Begin start-stop-status start
2022/04/26 00:18:40 (system) trigger SynologyPhotos 1.2.0-0263 End start-stop-status start ret=[0]
2022/04/26 00:18:45 (system) trigger SynologyDrive 3.1.0-22920 End start-stop-status start ret=[0]

closed 2022-04-30 by zyzzma

donwrightdesigns, 2025-09-14:
I figured it out - the text he has for us to copy is MISSING underscores that are visible in the screenshot!