bb-qq/r8152 #195: rtd1296 cdc-ncm driver for rtl8156

yuhuan417, 2022-02-04:
### Description of the problem

> - After 3 days struggling with realtek driver(r8152.c) on rtd1296 (works, but can't run any heavy payload, something like [https://bugzilla.kernel.org/show_bug.cgi?id=198931#c51](https://bugzilla.kernel.org/show_bug.cgi?id=198931#c51)), I switched to cdc-ncm driver. At lease it runs over 1Gbps in one direction(NAS->PC). Attachment zip file includes cdc_ncm.c (with little backport patch to preventing syslog spam)  and usbnet.c from kernel 4.4.180, and corresponding cdc_ncm.ko and usbnet.ko.
> - So it has nothing to do with the power supply, the cpu is not powerful enough, and the memory is not enough, it is a bug of the driver software level.

btw, the interface name is usb0 and not visable in DSM control panel, I'm still looking for how to rename it to eth1.

### Description of your products

> - the product name of your NAS model: DS420j
>     - Linux DiskStation 4.4.180+ #42218 SMP Fri Sep 24 02:39:50 CST 2021 aarch64 GNU/Linux synology_rtd1296_ds420j
>     - DSM version 7.01
> - ULT-unite 2.5G usb with rtl8156b chip rev 14

```-----------------------------------------------------------
Server listening on 5201 (test #1)
-----------------------------------------------------------
Accepted connection from 192.168.3.6, port 10164
[  5] local 192.168.3.9 port 5201 connected to 192.168.3.6 port 10166
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  58.1 MBytes   487 Mbits/sec
[  5]   1.00-2.00   sec  51.4 MBytes   431 Mbits/sec
[  5]   2.00-3.00   sec  55.9 MBytes   469 Mbits/sec
[  5]   3.00-4.00   sec  54.8 MBytes   459 Mbits/sec
[  5]   4.00-5.00   sec  49.7 MBytes   417 Mbits/sec
[  5]   5.00-6.00   sec  52.7 MBytes   442 Mbits/sec
[  5]   6.00-7.00   sec  59.1 MBytes   496 Mbits/sec
[  5]   7.00-8.00   sec  53.8 MBytes   451 Mbits/sec
[  5]   8.00-9.00   sec  53.4 MBytes   448 Mbits/sec
[  5]   9.00-10.00  sec  48.7 MBytes   408 Mbits/sec
[  5]  10.00-10.00  sec   120 KBytes   480 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   538 MBytes   451 Mbits/sec                  receiver
-----------------------------------------------------------
Server listening on 5201 (test #2)
-----------------------------------------------------------
Accepted connection from 192.168.3.6, port 10168
[  5] local 192.168.3.9 port 5201 connected to 192.168.3.6 port 10170
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.01   sec   170 MBytes  1.42 Gbits/sec   27    249 KBytes
[  5]   1.01-2.00   sec   174 MBytes  1.47 Gbits/sec   33   70.7 KBytes
[  5]   2.00-3.01   sec   172 MBytes  1.43 Gbits/sec   28    322 KBytes
[  5]   3.01-4.01   sec   176 MBytes  1.48 Gbits/sec    6    197 KBytes
[  5]   4.01-5.01   sec   158 MBytes  1.32 Gbits/sec   58    358 KBytes
[  5]   5.01-6.01   sec   167 MBytes  1.41 Gbits/sec   52    417 KBytes
[  5]   6.01-7.00   sec   147 MBytes  1.24 Gbits/sec   52    242 KBytes
[  5]   7.00-8.01   sec   140 MBytes  1.16 Gbits/sec   80    471 KBytes
[  5]   8.01-9.01   sec   172 MBytes  1.44 Gbits/sec   26    267 KBytes
[  5]   9.01-10.00  sec   158 MBytes  1.33 Gbits/sec   96   66.5 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.60 GBytes  1.37 Gbits/sec  458             sender
-----------------------------------------------------------
```

[rtd1296.zip](https://github.com/bb-qq/r8152/files/8001501/rtd1296.zip)

yuhuan417, 2022-02-05:
notice dsm already has usbnet.ko, upload again for missing cdc_ncm.c

achieved 2.3G and 1.2G with more iperf3 parallel clients.

[rtd1296.zip](https://github.com/bb-qq/r8152/files/8007188/rtd1296.zip)

bb-qq, 2022-02-05:
I didn't know that the CDC driver can also make the device work. Interesting.

I am experimenting with reducing the memory consumption of the Realtek driver on my end. You can try the test version of the driver if you like.
https://github.com/bb-qq/r8152/releases/download/2.15.0-5/r8152-rtd1296-2.15.0-6_rx_urb_queue_test.spk

yuhuan417, 2022-02-05:
by r8152.c , change the speed to 1G works.
```ethtool -s eth1 speed 1000 duplex full```

```
$ iperf3 --client 192.168.3.9
Connecting to host 192.168.3.9, port 5201
[  5] local 172.19.2.46 port 46614 connected to 192.168.3.9 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   114 MBytes   957 Mbits/sec    0   3.00 MBytes
[  5]   1.00-2.00   sec   112 MBytes   944 Mbits/sec    0   3.00 MBytes
[  5]   2.00-3.00   sec   112 MBytes   944 Mbits/sec    0   3.00 MBytes
[  5]   3.00-4.00   sec   111 MBytes   933 Mbits/sec    0   3.00 MBytes
[  5]   4.00-5.00   sec   112 MBytes   944 Mbits/sec    0   3.00 MBytes
[  5]   5.00-6.00   sec   112 MBytes   944 Mbits/sec    0   3.00 MBytes
[  5]   6.00-7.00   sec   112 MBytes   944 Mbits/sec    0   3.00 MBytes
[  5]   7.00-8.00   sec   111 MBytes   933 Mbits/sec    0   3.00 MBytes
[  5]   8.00-9.00   sec   112 MBytes   944 Mbits/sec    0   3.00 MBytes
[  5]   9.00-10.00  sec   112 MBytes   944 Mbits/sec    0   3.00 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.10 GBytes   943 Mbits/sec    0             sender
[  5]   0.00-10.03  sec  1.10 GBytes   940 Mbits/sec                  receiver

iperf Done.
yuhuan@DESKTOP-NUC:~$ iperf3 --client 192.168.3.9 -R
Connecting to host 192.168.3.9, port 5201
Reverse mode, remote host 192.168.3.9 is sending
[  5] local 172.19.2.46 port 46618 connected to 192.168.3.9 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   112 MBytes   938 Mbits/sec
[  5]   1.00-2.00   sec   112 MBytes   942 Mbits/sec
[  5]   2.00-3.00   sec   112 MBytes   942 Mbits/sec
[  5]   3.00-4.00   sec   112 MBytes   942 Mbits/sec
[  5]   4.00-5.00   sec   112 MBytes   942 Mbits/sec
[  5]   5.00-6.00   sec   112 MBytes   941 Mbits/sec
[  5]   6.00-7.00   sec   112 MBytes   942 Mbits/sec
[  5]   7.00-8.00   sec   112 MBytes   942 Mbits/sec
[  5]   8.00-9.00   sec   112 MBytes   942 Mbits/sec
[  5]   9.00-10.00  sec   112 MBytes   941 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.10 GBytes   943 Mbits/sec    0             sender
[  5]   0.00-10.00  sec  1.10 GBytes   941 Mbits/sec                  receiver

iperf Done.
```

yuhuan417, 2022-02-05:
Tried r8152.ko in r8152-rtd1296-2.15.0-6_rxbuf-test.spk, no luck. In fact, RX direction is fine (around 2.3G) even without buf size changing. TX direction lost connect immediately after launching iperf3.

My guess is something wrong with TX interrupts, hardware or firmware level. If the device runs over 1G, somehow it stops trigger interrupt and then timeout and reset by watchdog. Maybe usb power supply is still the root cause, I have ordered a usb Y cable and will test again.

USB cdc(communications device class) driver works because it is based on USB layer and not depends on hardware interrupts.

yuhuan417, 2022-02-08:
Self-powered USB hub doesn't help.

Ngakt, 2022-02-11:
Only r8156 supports cdc ncm. https://www.realtek.com/en/products/communications-network-ics/item/rtl8156b-s-cg

yifei101152044, 2022-02-12:
I have tried r8152-rtd1296-2.15.0-6_rxbuf-test.spk, but the kmsg told me that 
r8152: module verification failed: signature and/or required key missing - tainting kernel
And there are memory problems when using it. 
The features it shows is that the speed will decrease dramatically once the transmitted starts and then the connection is lost and reset. Any idea about it?

bb-qq, 2022-03-06:
Could you try another test driver?
https://github.com/bb-qq/r8152/releases/download/2.15.0-6/r8152-rtd1296-2.15.0-6nolpm.spk

yuhuan417, 2022-03-10:
I have sold my box :)

Before this I disabled lpm via usb quirks in the usbcore module with no improvement.

udosan174, 2022-03-15:
I am using DS418J.
I tried 2.15.0-6 but unfortunately it is unstable.
(I can save files from PC to NAS at high speed but cannot load them)

I want to try r8152-rtd1296-2.15.0-6nolpm.spk but I get a "file format is incorrect" error.
Please upload a new test driver.

bb-qq, 2022-03-19:
Thanks for trying to the driver. I revised the package.
https://github.com/bb-qq/r8152/releases/download/2.15.0-6/r8152-rtd1296-2.15.0-7nolpm2.spk

udosan174, 2022-03-19:
I tested r8152-rtd1296-2.15.0-7nolpm2.spk.
Unfortunately, the network interface hangs when sending data from the NAS.
Communication is recovered by linking eth1 down & up.
$ sudo ip link set eth1 down
$ sudo ip link set eth1 up
(ssh connected to onboard eth0)

PC to NAS
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   267 MBytes  2236 Mbits/sec
[  4]   1.00-2.00   sec   284 MBytes  2386 Mbits/sec
[  4]   2.00-3.00   sec   288 MBytes  2416 Mbits/sec
[  4]   3.00-4.00   sec   284 MBytes  2387 Mbits/sec
[  4]   4.00-5.00   sec   291 MBytes  2441 Mbits/sec
[  4]   5.00-6.00   sec   279 MBytes  2342 Mbits/sec
[  4]   6.00-7.00   sec   291 MBytes  2444 Mbits/sec
[  4]   7.00-8.00   sec   293 MBytes  2457 Mbits/sec
[  4]   8.00-9.00   sec   292 MBytes  2449 Mbits/sec
[  4]   9.00-10.00  sec   292 MBytes  2451 Mbits/sec
[  4]  10.00-11.00  sec   292 MBytes  2449 Mbits/sec
[  4]  11.00-12.00  sec   269 MBytes  2259 Mbits/sec
[  4]  12.00-13.00  sec   278 MBytes  2335 Mbits/sec
[  4]  13.00-14.00  sec   290 MBytes  2436 Mbits/sec
[  4]  14.00-15.00  sec   293 MBytes  2457 Mbits/sec
[  4]  15.00-16.00  sec   292 MBytes  2448 Mbits/sec
[  4]  16.00-17.00  sec   291 MBytes  2438 Mbits/sec
[  4]  17.00-18.00  sec   292 MBytes  2452 Mbits/sec
[  4]  18.00-19.00  sec   292 MBytes  2446 Mbits/sec
[  4]  19.00-20.00  sec   282 MBytes  2370 Mbits/sec
[  4]  20.00-21.00  sec   292 MBytes  2450 Mbits/sec
[  4]  21.00-22.00  sec   291 MBytes  2441 Mbits/sec
[  4]  22.00-22.38  sec   106 MBytes  2340 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-22.38  sec  6.27 GBytes  2407 Mbits/sec                  sender
[  4]   0.00-22.38  sec  0.00 Bytes  0.00 Mbits/sec

NAS to PC
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   743 KBytes  6.08 Mbits/sec    2   7.74 KBytes
[  5]   1.00-2.00   sec  0.00 Bytes  0.00 Mbits/sec    1   7.74 KBytes
[  5]   2.00-3.00   sec  0.00 Bytes  0.00 Mbits/sec    0   7.74 KBytes
[  5]   3.00-4.00   sec  0.00 Bytes  0.00 Mbits/sec    1   7.74 KBytes
[  5]   4.00-5.00   sec  0.00 Bytes  0.00 Mbits/sec    0   7.74 KBytes
[  5]   5.00-6.00   sec  0.00 Bytes  0.00 Mbits/sec    0   7.74 KBytes
[  5]   6.00-7.00   sec  0.00 Bytes  0.00 Mbits/sec    0   7.74 KBytes
^C[  5]   7.00-7.76   sec  0.00 Bytes  0.00 Mbits/sec    1   7.74 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-7.76   sec   743 KBytes  0.78 Mbits/sec    5             sender
[  5]   0.00-7.76   sec  0.00 Bytes  0.00 Mbits/sec                  receiver
iperf3: interrupt - the client has terminated

GorgiGR, 2022-04-21:
It is very interesting to hear that there are instances of people that managed to get it working in either direction with speeds greater than 1GBps.
This means not all hope is lost for this platform :)
In case there are newer versions of this driver I will keep an eye on this thread in order to try them too 👍

bb-qq, 2022-08-20:
> The relevant comment is this:
> https://bugzilla.kernel.org/show_bug.cgi?id=198931#c96
> where also the relevant commit is mentioned and one can see the code changes.

This patch has been integrated into the release [2.16.3-1](https://github.com/bb-qq/r8152/releases/tag/2.16.3-1).

bb-qq, 2022-12-04:
I decided to merge topics on rtd1296 into issue #275.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq