bb-qq/r8152 #197: USB NIC isnt avaiable to SMB multichannel from boot

Tezouro, 2022-02-06:
### Description of the problem

After a ton of tests i think USB NIC isnt avaiable to SMB multichannel from boot.
Probably the driver is loaded after SMB config/service is loaded.

If i restarted the SMB service with "synosystemctl restart pkg-synosamba-smbd.service" the multichannel use the 3 nics. It is possible to fix this? 

No other problems so far : )

Thanks
 

### Description of your products

> Synology 918+
> Linux N2 4.4.180+ #42218 SMP Fri Sep 24 02:41:40 CST 2021 x86_64 GNU/Linux synology_apollolake_918+
> DSM 7.0.1-42218 Update 2
> UGREEN USB C to 2.5G Ethernet Adapter

### Description of your environment

> NAS and PC connected with 3 cables each to the switch
> PC: NIC Broadcom NetXtreme Gigabit Ethernet 2x Dual NICs
> Switch Tp-link TL-SG1016DE
> Cat.6a cables 

### Output of ``dmesg`` command

````
 dmesg
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.180+ (root@build15) (gcc version 7.5.0 (GCC) )                                                                                                              #42218 SMP Fri Sep 24 02:41:40 CST 2021
[    0.000000] Command line: root=/dev/md0 earlyprintk=apl console=ttyS2,115200n                                                                                                             8 ihd_num=4 netif_num=2 HddHotplug=1 SataPortMap=23 sata_remap=0>2:1>3:2>0:3>1 s                                                                                                             yno_hw_version=DS918+ vender_format_version=2 syno_hdd_detect=18,179,176,175 syn                                                                                                             o_hdd_enable=21,20,19,9 syno_usb_vbus_gpio=13@0000:00:15.0@1,11@0000:00:15.0@2 s                                                                                                             n=1980PDN193208 macs=001132b7aa1d,001132b7aa1e
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point regis                                                                                                             ters'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x08: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x10: 'MPX CSR'
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 1088 bytes                                                                                                             , using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000067c17fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067c18000-0x0000000067c18fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000067c19000-0x0000000067c42fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000067c43000-0x0000000067c7dfff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067c7e000-0x0000000068c7dfff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000068c7e000-0x000000007822efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007822f000-0x000000007aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aebf000-0x000000007afbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007afbf000-0x000000007affefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007afff000-0x000000007affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3709000-0x00000000d370afff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000037fffffff] usable
[    0.000000] bootconsole [earlyser0] enabled
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x01dd9a88-0x01dd9a97] usable ==> usable
[... 1232 lines cut ...]

eth2      Link encap:Ethernet  HWaddr 2C:16:DB:A7:02:14
          inet addr:192.168.0.52  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:1690781 errors:0 dropped:0 overruns:0 frame:0
          TX packets:583340 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:14870484557 (13.8 GiB)  TX bytes:59302970 (56.5 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:5098 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5098 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:777261 (759.0 KiB)  TX bytes:777261 (759.0 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

bb-qq, 2022-02-12:
Unfortunately, a package installed by the package manager is not able to load before built-in services as far as I know.

I can change the driver to restart the SMB service, but it is not needed for most users. So I think a possible workaround is that you add a "boot trigger task" to the Synology's Task Scheduler.

mrjohnpoz, 2022-02-19:
So your saying your using smb3 multichannel with 3x1 gig on your 918?  You just have to restart the smb after boot?  I wouldn't mind playing with that.. Not sure how much a performance increase would see at 3x1 vs just 1 at 2.5..  I max out file transfers at 280MBps... Which was a nice bump from the 220ish seeing with smb3 multichannel and 2x1 interfaces..

Tezouro, 2022-02-19:
I reach 354MBps and with a Gigabit Switch  ; )
Its is very good since 918+ doesnt have 10Gb nic.

mrjohnpoz, 2022-02-19:
Hmmm - I think I might have to pull the nic card out of my old pc and give this a go.. I had tried it when I first got the 2.5 card.. But I don't think I had ever actually restarted smb on the nas and was wondering why it wasn't working ;) thanks!!

This would allow me to get rid of the sep network that is between nas and pc via the 2.5 nics..  Might be a fun little project for tmrw ;)

bb-qq, 2022-03-06:
I am going to close this Issue as there does not seem to be anything more I can do. If there is anything else I can do, feel free to re-open it.

closed 2022-03-06 by bb-qq