bb-qq/r8152 #202: Club 3D USB 3.2 Gen1 with DSM 7 – Problem only 100Mbit

jensche, 2022-03-23:
### Description of the problem

Hello, I got the same issue. But in my case the adapter worked like a charm for 1 month. after 1 restart it just jumped down from 2.5gbit down to 100mbit. 
______________________
Supported ports: [ MII ]
	Supported link modes:   10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	                        1000baseT/Full 
	                        2500baseX/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  10baseT/Half 10baseT/Full 
	                        100baseT/Half 100baseT/Full 
	Advertised pause frame use: Symmetric Receive-only
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Link partner advertised link modes:  10baseT/Half 10baseT/Full 
	                                     100baseT/Half 100baseT/Full 
	                                     1000baseT/Half 1000baseT/Full 
	Link partner advertised pause frame use: No
	Link partner advertised auto-negotiation: Yes
	Link partner advertised FEC modes: Not reported
	Speed: 100Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 32
	Transceiver: internal
	Auto-negotiation: on
	Supports Wake-on: pumbg
	Wake-on: g
	Current message level: 0x00007fff (32767)
			       drv probe link timer ifdown ifup rx_err tx_err tx_queued intr tx_done rx_status pktdata hw wol
	Link detected: yes
______________________
`sudo ethtool -s eth2 speed 2500 duplex full autoneg off`
is not working.

______________________
I've deinstalled the driver, restarted, then installed the driver again, and restarted. still not working. 
______________________
I've tried all USB-Ports... Still only 100Mbit. 

### Description of your products
>     - synology_braswell_916+
>     - DSM version DSM 7.0.1-42218 Update 3
> - Club 3D USB 3.2 Gen1

### Description of your environment

> - connect directly to a PC or using a hub
>     - Mac M1 via SMB

> - cable category Cat. 7

### Output of ``dmesg`` command

````
[ 1983.685944] usbcore: deregistering interface driver r8152
[ 2205.800530] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 2 with no TDs queued?
[ 2205.810193] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 3 with no TDs queued?
[ 2205.819902] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 2 ep 6 with no TDs queued?
[ 2206.931883] usb 4-2: reset SuperSpeed USB device number 6 using etxhci_hcd-170202
[ 2207.490572] r8152 4-2:1.0 eth2: v2.15.0 (2021/04/15)
[ 2207.496125] r8152 4-2:1.0 eth2: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 2207.513540] r8152 4-2:1.0 eth2: chip rev 12
[ 2207.518276] usbcore: registered new interface driver r8152
[ 2222.591529] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[ 2232.620949] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[ 2232.630431] r8152 4-2:1.0 eth2: carrier on
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650071C92CE41604)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__2-2         0bc2:aa15:0712 00  3.00 5000MBit/s 896mA 1IF  (Seagate Basic NABCXW4A)
|__usb3          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__4-1         0bc2:ab21:0143 00  3.00 5000MBit/s 144mA 1IF  (Seagate  Backup+  BL NA76DL0L)
  |__4-2         0bda:8156:3000 00  3.20 5000MBit/s 512mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 000000001)
````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 00:11:32:73:46:5B  
          inet addr:192.168.106.118  Bcast:192.168.106.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe73:465b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:47129 errors:0 dropped:0 overruns:0 frame:0
          TX packets:44122 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:12000647 (11.4 MiB)  TX bytes:25131155 (23.9 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:73:46:5C  
          inet addr:169.254.144.136  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr A0:CE:C8:E8:8F:9E  
          inet addr:192.168.106.20  Bcast:192.168.106.255  Mask:255.255.255.0
          inet6 addr: fe80::a2ce:c8ff:fee8:8f9e/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:1953 errors:0 dropped:0 overruns:0 frame:0
          TX packets:848 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:607143 (592.9 KiB)  TX bytes:159662 (155.9 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:15977 errors:0 dropped:0 overruns:0 frame:0
          TX packets:15977 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:3655803 (3.4 MiB)  TX bytes:3655803 (3.4 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

MKVRG99, 2022-03-31:
have you tried mapping a drive to PC and transfer files via SMB 3

jensche, 2022-04-04:
> have you tried mapping a drive to PC and transfer files via SMB 3

yeah. I really only got 11.7MB/s

bb-qq, 2022-04-10:
When you check the link speed on the Mac (if connected directly) or on the switch (if through a switch), did you confirm that the link is actually at 100mbps? If so, is it possible to fix the link speed in the oncoming configuration?

Realtek devices do not seem to have the ability to fix the link speed to 2.5g.

jensche, 2022-04-10:
The strange thing is that it worked perfectly for 1 month. Then I’ve restarted the synology… now its broke.

bb-qq, 2022-04-10:
I agree it's a strange situation. It could be heat or some other hardware problem that has caused a change in timing or performance. We don't really know, but at least the possible workarounds are mentioned above.

jensche, 2022-08-19:
After many tests, new cables etc. I've found out cable length is a big issue. Over 5m does not support higher speeds...

bb-qq, 2022-08-20:
Are you talking about USB cables?
Indeed, by standard, USB 3 does not support cables longer than 3 meters.

jensche, 2022-08-20:
Ethernet cable. If I’m using a more than 5m long ethermet cable its not working. Ive tried 5e and cat 7 cables…

bb-qq, 2022-08-20:
Oh, you were talking about Ethernet cables.

Cat7 is a scam and many cables are of low quality, so it is still not surprising to see such a result.
By standards, MultiGiga should support Cat5e and work well, but Cat5e was not originally designed for MultiGiga, so that may be the case. I updated [Troubleshooting](https://github.com/bb-qq/r8152/wiki/Troubleshooting) about this topic.

I would expect a little better results with Cat6A cables.

bb-qq, 2022-08-20:
I will leave this Issue Open as is, but please close it if your problem is completely resolved and you won't do any additional experimentation.

jensche, 2022-08-20:
In my case, the strange thing is: It worked for over 2 Month with those Cat7 calbes...
Now I'm using a 3m Cat7 Cable, same brand like the longer ones... and it works. 

Synology 916+ (DSM 7.1-42661 Update 4)
Driver: 2.15.0-10
Hot Swap not working.

 I'll try with another Cat6a cable.

bb-qq, 2023-08-13:
I would like to close this issue since there has been no response. If you have any questions, feel free to re-open it.

closed 2023-08-13 by bb-qq