bb-qq/r8152 #205: DS916+ DSM7.1 2.5GB Realtek RTL8156b

Sandor-dev, 2022-04-01:
### Description of the problem

> RTL 8156B usb3.0 2.5GBE card is not working

### Description of your products

DS916+ 8Gb
Linux Dragon 3.10.108 #42621 SMP Mon Mar 21 21:24:06 CST 2022 x86_64 GNU/Linux synology_braswell_916+
7.1-42621
Fenvi RTL8156b

### Description of your environment

Directly
same connector at PC site
Cat.6e

### Output of ``dmesg`` command

**[553785.802513] usb 2-2: new SuperSpeed USB device number 38 using xhci_hcd
[553785.820605] usb 2-2: Parent hub missing LPM exit latency info.  Power management will be impacted.**
[553785.831332] ppid:2(kthreadd), pid:27(kdevtmpfs), dirtied inode 110368065 (038) on devtmpfs
[553785.831540] ppid:1(systemd), pid:5386(systemd-udevd), dirtied inode 110367170 (queue) on tmpfs
**[553785.955666] usb 2-2: reset SuperSpeed USB device number 38 using xhci_hcd**
[553785.963576] ppid:1(systemd), pid:5022(systemd-journal), dirtied inode 15044 (subsystem) on sysfs
**[553785.973611] usb 2-2: Parent hub missing LPM exit latency info.  Power management will be impacted.**
[553785.983909] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88023c9d35c0
[553785.994132] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88023c9d3600
[553785.994187] ppid:1(systemd), pid:5022(systemd-journal), dirtied inode 15012 (189:165) on sysfs
[553786.004359] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff88023c9d3640
**[553786.067000] r8152 2-2:1.0 eth2: v2.15.0 (2021/04/15)**
[553786.067016] ppid:1(systemd), pid:32199(syslog-ng), dirtied inode 19994 (messages) on md0
[553786.067029] ppid:1(systemd), pid:32199(syslog-ng), dirtied inode 19994 (messages) on md0
[553786.067032] ppid:1(systemd), pid:32199(syslog-ng), dirtied inode 19994 (messages) on md0
[553786.067158] ppid:1(systemd), pid:32199(syslog-ng), dirtied inode 26783 (kern.log) on md0
[553786.067164] ppid:1(systemd), pid:32199(syslog-ng), dirtied inode 26783 (kern.log) on md0
[553786.067167] ppid:1(systemd), pid:32199(syslog-ng), dirtied inode 26783 (kern.log) on md0
**[553786.072673] r8152 2-2:1.0 eth2: This product is covered by one or more of the following patents:
                                US6,570,884, US6,115,776, and US6,327,625.**

**[553786.090379] r8152 2-2:1.0 eth2: chip rev 14
[553786.095163] r8152 2-2:1.0 eth2: rx_buf_sz will be overridden to 16384**
[553786.102490] ppid:1(systemd), pid:5022(systemd-journal), dirtied inode 19525 (2-2:1.0) on sysfs
[553786.102573] ppid:1(systemd), pid:5022(systemd-journal), dirtied inode 19526 (subsystem) on sysfs
[553786.264809] ppid:4894(ups-check.sh), pid:4925(cp), dirtied inode 37289 (ups.conf) on md0
[553786.264824] ppid:4894(ups-check.sh), pid:4925(cp), dirtied inode 37289 (ups.conf) on md0
[553786.264957] ppid:4894(ups-check.sh), pid:4925(cp), WRITE block 3526888 on md0 (8 sectors)
[553786.265024] ppid:2(kthreadd), pid:3899(md0_raid1), WRITE block 4980352 on sda1 (8 sectors)
[553786.265056] ppid:2(kthreadd), pid:3899(md0_raid1), WRITE block 4980352 on sdd1 (8 sectors)
[553786.265070] ppid:2(kthreadd), pid:3899(md0_raid1), WRITE block 4980352 on sdc1 (8 sectors)
[553786.265081] ppid:2(kthreadd), pid:3899(md0_raid1), WRITE block 4980352 on sdb1 (8 sectors)
[553786.318601] ppid:5386(systemd-udevd), pid:4827(systemd-udevd), dirtied inode 110367200 (189:165) on devtmpfs
[553786.318656] ppid:5386(systemd-udevd), pid:4827(systemd-udevd), dirtied inode 110367201 (c189:165.tmp) on tmpfs
[553786.319493] ppid:5386(systemd-udevd), pid:4945(systemd-udevd), dirtied inode 110368118 (oom_score_adj) on proc
[553786.337523] ppid:5386(systemd-udevd), pid:4945(systemd-udevd), dirtied inode 110368123 (+usb:2-2:1.0.tmp) on tmpfs
[553786.337770] ppid:5386(systemd-udevd), pid:4828(systemd-udevd), dirtied inode 110368848 (+usb:2-2:2.1.tmp) on tmpfs
[553786.338034] ppid:5386(systemd-udevd), pid:4945(systemd-udevd), dirtied inode 110368124 (n73.tmp) on tmpfs
[553786.338082] ppid:5386(systemd-udevd), pid:4945(systemd-udevd), dirtied inode 110368125 (n73) on tmpfs
[553786.339932] ppid:5386(systemd-udevd), pid:4827(systemd-udevd), dirtied inode 110366618 (+usb:2-2:2.0.tmp) on tmpfs
[553786.342019] ppid:5386(systemd-udevd), pid:4827(systemd-udevd), dirtied inode 110367206 (+queues:rx-0.tmp) on tmpfs
[553786.342058] ppid:5386(systemd-udevd), pid:4828(systemd-udevd), dirtied inode 110368849 (+queues:tx-0.tmp) on tmpfs

### Output of ``lsusb`` command

lsusb
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__1-5         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500631DA6E41699)
|__usb2          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (xhci_hcd 0000:00:14.0) hub
  |__2-2         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 0013000000)

[... 111 lines cut ...]
          RX bytes:8505656 (8.1 MiB)  TX bytes:47769253 (45.5 MiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:5B:3B:C4
          inet addr:192.168.5.55  Bcast:192.168.5.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe5b:3bc4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1002 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5105 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:254803 (248.8 KiB)  TX bytes:3223164 (3.0 MiB)

ppp201    Link encap:Point-to-Point Protocol
          inet addr:10.0.0.0  P-t-P:10.0.0.1  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1396  Metric:1
          RX packets:19112 errors:0 dropped:0 overruns:0 frame:0
          TX packets:38155 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3
          RX bytes:2357543 (2.2 MiB)  TX bytes:44388820 (42.3 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

chmooreck, 2022-04-05:
> eth2 Link encap:Ethernet HWaddr 1C:BF:CE:D4:00:77
> BROADCAST MULTICAST MTU:1500 Metric:1
> RX packets:0 errors:0 dropped:0 overruns:0 frame:0
> TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
> collisions:0 txqueuelen:1000
> RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)

interface is up. Isn't it visible in network settings in DSM ?

Sandor-dev, 2022-04-05:
Thank you for reply.

In DSM  control panel it isn't visible. I used the front panel.

chmooreck, 2022-04-05:
$ dmesg | grep eth2

You've got openvswitch configured on eth0 and eth1 ?
Maybe try to furn it off for eth2 test  ?

Sandor-dev, 2022-04-05:
I uninstalled the Virtual Machine Manager and switched off the open virtual switch.
I removed the bond from eth0 and eth1
Restarted the NAS.
I cannot see any error in dmesg and the dmesg | grep eth2 does not show anything. 

I stopped the docker too -> nothing happen.
I tried many linux commands what i found on internet, most of it not working.
I tried ifconfig eth2 up --> THEN IT WORKS!!!!

thank you guys!

Sandor-dev, 2022-04-05:
When i plug off- plug in and do ifconfig eth2 up command it shows in control panel for a minutes, then disappear.

dmesg: 
[ 1603.559312] ppid:26169(ssdp.sh), pid:26184(reg_ssdp_servic), dirtied inode 33504 (desc-DSM-eth0.xml) on md0
[ 1603.559323] ppid:26169(ssdp.sh), pid:26184(reg_ssdp_servic), dirtied inode 33504 (desc-DSM-eth0.xml) on md0
[ 1603.560071] ppid:26169(ssdp.sh), pid:26184(reg_ssdp_servic), dirtied inode 33681 (desc-DSM-eth1.xml) on md0
[ 1603.560078] ppid:26169(ssdp.sh), pid:26184(reg_ssdp_servic), dirtied inode 33681 (desc-DSM-eth1.xml) on md0
[ 1603.566343] ppid:26169(ssdp.sh), pid:26185(cp), dirtied inode 441162 (desc-DSM-eth0.xml) on tmpfs
[ 1603.566464] ppid:26169(ssdp.sh), pid:26185(cp), dirtied inode 441163 (desc-DSM-eth1.xml) on tmpfs
[ 1606.180000] ppid:26717(ssdp.sh), pid:26736(reg_ssdp_servic), dirtied inode 33887 (desc-DSM-eth0.xml) on md0
[ 1606.180010] ppid:26717(ssdp.sh), pid:26736(reg_ssdp_servic), dirtied inode 33887 (desc-DSM-eth0.xml) on md0
[ 1606.180711] ppid:26717(ssdp.sh), pid:26736(reg_ssdp_servic), dirtied inode 35572 (desc-DSM-eth1.xml) on md0
[ 1606.180733] ppid:26717(ssdp.sh), pid:26736(reg_ssdp_servic), dirtied inode 35572 (desc-DSM-eth1.xml) on md0
[ 1606.180738] ppid:26717(ssdp.sh), pid:26736(reg_ssdp_servic), dirtied inode 35572 (desc-DSM-eth1.xml) on md0
[ 1606.187107] ppid:26717(ssdp.sh), pid:26740(cp), dirtied inode 446883 (desc-DSM-eth0.xml) on tmpfs
[ 1606.187251] ppid:26717(ssdp.sh), pid:26740(cp), dirtied inode 446884 (desc-DSM-eth1.xml) on tmpfs
[ 1607.668902] ppid:26975(python3), pid:26992(cat), dirtied inode 18027 (eth2) on sysfs
[ 1609.399569] r8152 2-2:1.0 eth2: get_registers -110
then:

[ 1700.081097] ppid:32173(ssdp.sh), pid:32184(reg_ssdp_servic), dirtied inode 33679 (desc-DSM-eth0.xml) on md0
[ 1700.081104] ppid:32173(ssdp.sh), pid:32184(reg_ssdp_servic), dirtied inode 33679 (desc-DSM-eth0.xml) on md0
[ 1700.081847] ppid:32173(ssdp.sh), pid:32184(reg_ssdp_servic), dirtied inode 33887 (desc-DSM-eth1.xml) on md0
[ 1700.081856] ppid:32173(ssdp.sh), pid:32184(reg_ssdp_servic), dirtied inode 33887 (desc-DSM-eth1.xml) on md0
[ 1700.089933] ppid:32173(ssdp.sh), pid:32188(cp), dirtied inode 482832 (desc-DSM-eth0.xml) on tmpfs
[ 1700.090076] ppid:32173(ssdp.sh), pid:32188(cp), dirtied inode 482833 (desc-DSM-eth1.xml) on tmpfs
[ 1727.727697] ppid:473(python3), pid:480(cat), dirtied inode 18031 (eth2) on sysfs
plugged eth1 cable off
plug eth cable off-then on:
[ 1783.459493] ppid:2178(ssdp.sh), pid:2188(reg_ssdp_servic), dirtied inode 33671 (desc-DSM-eth0.xml) on md0
[ 1783.459501] ppid:2178(ssdp.sh), pid:2188(reg_ssdp_servic), dirtied inode 33671 (desc-DSM-eth0.xml) on md0
[ 1783.468145] ppid:2178(ssdp.sh), pid:2189(cp), dirtied inode 497530 (desc-DSM-eth0.xml) on tmpfs
[ 1787.721758] ppid:2663(python3), pid:2674(cat), dirtied inode 18038 (eth2) on sysfs
**[ 1800.312318] r8152 2-2:1.0 eth2: v2.15.0 (2021/04/15)
[ 1800.317882] r8152 2-2:1.0 eth2: This product is covered by one or more of the following patents:
[ 1800.335295] r8152 2-2:1.0 eth2: chip rev 14
[ 1800.339979] r8152 2-2:1.0 eth2: rx_buf_sz will be overridden to 16384**
[ 1866.080332] r8152 2-2:1.0 eth2: get_registers -71
[ 1866.089070] r8152 2-2:1.0 eth2: get_registers -71
[ 1866.097815] r8152 2-2:1.0 eth2: get_registers -71
[ 1866.111989] r8152 2-2:1.0 eth2: get_registers -71
[ 1867.629130] r8152 2-2:1.0 eth2: v2.15.0 (2021/04/15)
[ 1867.634688] r8152 2-2:1.0 eth2: This product is covered by one or more of the following patents:
[ 1867.652123] r8152 2-2:1.0 eth2: chip rev 14
[ 1867.656833] r8152 2-2:1.0 eth2: rx_buf_sz will be overridden to 16384

chmooreck, 2022-04-06:
Ok, this is something. But unfortunately beyond my knowledge.

bb-qq, 2022-04-10:
I have never heard any reports of Fenvi RTL8156b before so it could be a problem on the sides of this device. Is it possible to return it and exchange it for another device that has been reported to work with braswell models?
https://github.com/bb-qq/r8152/wiki/Compatibility

Sandor-dev, 2022-04-10:
Nope, I would try with this model (SHould be same isn't it? I already asked some friends IRL to help me. I will reply the progression of work.

Sandor-dev, 2022-12-04:
Update!

The original Fenvi usb adapters started to work on DS916+

1., installed the latest driver from this repo: [2.16.3-3 DSM7.x (reuploaded)](https://github.com/bb-qq/r8152/releases/tag/2.16.3-3)
during the installation got warning: we must login via ssh and paste the command and retry the installation.
2., start the package
3.: via SSH check the usb hardwares: `lsusb` 
4.: if you found your network card switch on: `ifconfig eth2 up`
5.: now you can find it on GUI, but you cannot set up an IP for it. `ip a s eth2` 
6.: give IP address: `ip addr add 192.168.yoursubnet.3rdcardIP/24 dev eth2`
you are done!
And my Fenvi cards started to working. But never buy FENVI 2.5G to USB ethernet, because it can link up as 2.5G but the total speed is maximum 1 gigabit. ( tested)
Now I have DLINK cards.

closed 2023-08-13 by bb-qq