bb-qq/r8152 #207: SDM 7.1 support?

mrjohnpoz, 2022-04-09:
So it seems 7.1 is official released now.

Wondering if anyone validated driver works?  I would move to 7.1 from 7.01, but do not want to loose my 2.5ge connection.

I see one issue where user having issues with 7.1... I am running DS918+ with club 3d CAC-1420 

Can anyone validate that driver is working, or should wait to move to 7.1 until specific new release for 7.1?

Sjekke, 2022-04-09:
I'm running DSM7.1 RC on a 916+ and my 2,5gbit adapter works as a charm.

bb-qq, 2022-04-10:
Thank you for your information.

I have not been able to verify actual functioning, but I expect that the current drivers will work as-is on DSM7.1, as it does not require any new toolchains.

Sjekke, 2022-04-10:
I'm using https://plugable.com/products/usbc-e2500)

mrjohnpoz, 2022-04-10:
Well updated to 7.1-42661, after a bit of scare.. It didn't update, had to reboot it after 40 minutes.  It was saying on just the http://nasip that it was recoverable, but it would fail trying to do the recover..   After the manual reboot, the recovery worked, and everything is working that I can see, and still have my 2.5 connection..

fengziX, 2022-04-20:
DS920+  DSM7.1 ,  The driver does not work, try to reinstall , the system told me can not be installed.

mrjohnpoz, 2022-08-10:
Since this actually hasn't been closed, figured I would mention updated to 7.1.1-42951 (release candidate) this morning.  Update went fine, usb nic was showing.. but wasn't actually showing IP on nas via ifconfig.. Noticed a new 2.15.0-10 version, so updated to that, did the sudo command.. It was just the nas was not actually putting the IP on the interface even though in the gui it was showing.

What cleared it was setting the interface on the nas to dhcp, this failed as I expected since no dhcp on that network between my nas and pc and showed a 169.254 address - changed it back to my static 192.168.10.10 address and back in business.

All seems to be good so far

closed 2023-08-13 by bb-qq