bb-qq/r8152 #208: DS220+ Asustor 2.5G upload speed slow

ChristophePhan, 2022-04-10:
### Description of the problem

Upload speed capped to 30mbps with speedtest cli and transfert file from nas to pc.
But on integrated ethernet of DS220+ upload is ok.
Download from/to nas is ok.

### Description of your products

> - DS220+
>     - Linux DS220plus 4.4.180+ #42218 SMP Mon Oct 18 19:16:55 CST 2021 x86_64 GNU/Linux synology_geminilake_220+
>     - DSM 7.0.1-42218 Update 3
> - AS-U2.5G2

### Description of your environment

> - PC connected to wifi 6
>     - PC: surface laptop 4
>     - Hub: isp box
> - cable Cat.5e

### Output of ``dmesg`` command

[dmesg.txt](https://github.com/bb-qq/r8152/files/8459264/dmesg.txt)

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F000AD6CE3D5174)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         0bda:8156:3104 00  3.20 5000MBit/s 304mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 401000001)
````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:84 base 0xe000

eth1      Link encap:Ethernet  HWaddr
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:2951 errors:0 dropped:0 overruns:0 frame:0
          TX packets:632 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:356283 (347.9 KiB)  TX bytes:90591 (88.4 KiB)
          Interrupt:85 base 0x2000

eth2      Link encap:Ethernet  HWaddr 
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:9000  Metric:1
          RX packets:10842 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7930 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3780640 (3.6 MiB)  TX bytes:3356326 (3.2 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3995 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3995 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:945893 (923.7 KiB)  TX bytes:945893 (923.7 KiB)
````

bb-qq, 2022-04-23:
Could you try one previous release and let me know if there are any changes? There may have been a problem with a recent change.
https://github.com/bb-qq/r8152/releases/download/2.15.0-5/r8152-geminilake-2.15.0-5.spk

cph001, 2022-04-23:
Hello, I tried with the version 2.15.0-5 and I have the same issue.

bb-qq, 2022-05-01:
Can you try what happens when the PC and NAS are directly connected?
There have been some reported compatibility issues with ethernet switches.

cph001, 2022-05-03:
I bought the club 3d adapter and after some tests on both with direct connection on pc, I found out that the asustor is the issue.

bb-qq, 2022-05-04:
So you can transfer files at full speed with the CLUB 3D adapter?

The reason I can think of for why the ASUSTOR adapter doesn't work well is the orientation of the Type A-C conversion adapter. from the output of the lsusb, it seems you are using it in the correct orientation, but what happens if you try it and reverse the Type-C connector? Also if you have another Type A-C conversion adapter you could try that.

bb-qq, 2022-05-28:
Thank you.
Can we close this Issue now?

closed 2022-05-28 by cph001