bb-qq/r8152 #209: Is DS3622xs+ supported?

PeterSuh-Q3, 2022-04-11:
### Description of the problem

> Replace this line with your answer.

I am a DS3622xs+ XPENOLOGY user.
I used the r8152-broadwell-2.15.0-5.spk file, but it does not install because it is an unsupported platform.

### Description of your products

> - DS3622xs+
>     - Linux DSM701_PURE 4.4.180+ #42661 SMP Fri Apr 1 15:33:00 CST 2022 x86_64 GNU/Linux synology_broadwellnk_3622xs+
>     - DSM version 7.1
> - IPTIME U1G

### Description of your environment

> - using a hub
>     - PC: Realtek RTL8111H
>     - Hub: TP-LINK TL-SG108E
> - Cat.7

### Output of ``dmesg`` command

````
[dmesg.txt](https://github.com/bb-qq/r8152/files/8464337/dmesg.txt)
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1a.0) hub
  |__1-1         8087:8009:0000 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1d.0) hub
  |__2-1         8087:8001:0000 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffa0) hub
|__usb3          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__3-4         258a:1006:3014 00  1.10   12MBit/s 300mA 2IFs (Gaming KB  Gaming KB  ffffff94ffffffd6ffffff9dffffffbd)
  |__3-7         046d:c52b:1209 00  2.00   12MBit/s 98mA 3IFs (Logitech USB Receiver ffffffc6ffffffa3ffffffd5ffffffbc)
  |__3-9         f400:f400:0100 00  2.10  480MBit/s 224mA 1IF  (SanDisk Cruzer Blade 4C531001520514109364)
|__usb4          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:14.0) hub
  |__4-4         0bda:8153:3100 00  3.00 5000MBit/s 288mA 2IFs (Realtek USB 10/100/1000 LAN 001000001)
````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 00:11:32:C9:C6:23  
          inet addr:169.254.89.153  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth1      Link encap:Ethernet  HWaddr 00:00:0C:73:7C:D4  
          inet addr:169.254.195.218  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth99     Link encap:Ethernet  HWaddr 
          inet addr:192.168.35.85  Bcast:192.168.35.255  Mask:255.255.255.0
          inet6 addr:  Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1799 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1952 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:815284 (796.1 KiB)  TX bytes:1919577 (1.8 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:188 errors:0 dropped:0 overruns:0 frame:0
          TX packets:188 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:26890 (26.2 KiB)  TX bytes:26890 (26.2 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

PeterSuh-Q3, 2022-04-11:
[dmesg.txt](https://github.com/bb-qq/r8152/files/8464337/dmesg.txt)

bb-qq, 2022-04-23:
I uploaded a driver for the broadwllnk platform. Try it please.
https://github.com/bb-qq/r8152/releases/download/untagged-4e3f9f9c212f7b83594a/r8152-broadwellnk-2.15.0-7.spk

PeterSuh-Q3, 2022-04-23:
Thank you so much.
But I think the download link is broken.
Check please.

PeterSuh-Q3, 2022-04-23:
I checked the exact link again in the release.
Thank you.
https://github.com/bb-qq/r8152/releases/download/2.15.0-7/r8152-broadwellnk-2.15.0-7.spk

closed 2022-04-23 by PeterSuh-Q3

PeterSuh-Q3, 2022-04-23:
It works very well.

I have one more USB NIC installed on the dual LAN.

You don't have to touch the grub.cfg, just reboot after installing the driver.