bb-qq/r8152 #210: D-Link DUB-E250: does not appear in DS218+ Network Interface list, but does appear in USB device list

bobk88, 2022-04-13:
### Description of the problem

I successfully installed the Apollo Lake driver per your instructions - version 2.15.0-5 is running, as verified in Package Center.

I connected a D-Link DUB-E250 to one of the rear USB 3.0 ports on a DS218+ via a Pearstone (B&H) USB-C to USB-A adapter.

The DUB-E250 does not appear in the device list in Control Panel > Network > Network interface, but it does appear in the list of USB devices - see lsusb output below.

Same results when connected to the front USB port.

The D-Link works fine when plugged into a USB-C port on my hackintosh running macOS 12.3.1, using the driver com.apple.driver.usb.cdc.ncm

Happy to provide more info, but as I'm a beginner (and not an engineer), I'll need specific steps.

Thanks!

### Description of your products

DS218+
Linux [drive name] 4.4.180+ #42218 SMP Mon Oct 18 19:17:56 CST 2021 x86_64 GNU/Linux synology_apollolake_218+
DSM 7.0.1-42218
D-Link DUB-E250 - a review at Amazon says it uses an 8156b chipset, for what that's worth.

### Description of your environment

NAS is connected to a Netgear GS105 hub
Adapter is connected to NAS with a Cat.5e cable, about 3 feet long; same results with a Cat 6 cable, 10-15 feet long

### Output of ``dmesg`` command

NOTE: The full output was over 2,000 lines long. When I pasted it here, the web site said my comment was too long. I don't know which part is relevant, so I guessed and pasted in the last ~100 lines. Let me know if you need more. Thanks!

[340862.271877] synoboot1 unmounted, process=synowedjat
[345317.589096] ata2: wake up from deepsleep, reset link now
[345317.839910] ata2: device plugged sstatus 0x133
[345324.588358] ata1: wake up from deepsleep, reset link now
[345324.874017] ata1: device plugged sstatus 0x133
[345326.372295] ata2: link is slow to respond, please be patient (ready=0)
[345330.768114] ata2: softreset failed (device not ready)
[345330.773894] ata2: SRST fail, set srst fail flag
[345331.659066] ata2: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[345331.666124] ata2: link reset sucessfully clear error flags
[345331.677110] ata2.00: disabling queued TRIM support
[345331.684638] ata2.00: disabling queued TRIM support
[345331.684653] ata2.00: configured for UDMA/133
[345331.689559] ata2: wake up successful, the reset fail can be ignored
[345333.365977] ata1: link is slow to respond, please be patient (ready=0)
[345337.760797] ata1: softreset failed (device not ready)
[345337.766560] ata1: SRST fail, set srst fail flag
[345338.652774] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[345338.659819] ata1: link reset sucessfully clear error flags
[345338.670786] ata1.00: disabling queued TRIM support
[345338.678476] ata1.00: disabling queued TRIM support
[345338.678497] ata1.00: configured for UDMA/133
[345338.683410] ata1: wake up successful, the reset fail can be ignored
[354125.438299] ata1: wake up from deepsleep, reset link now
[354125.725344] ata1: device plugged sstatus 0x133
[354132.447015] ata2: wake up from deepsleep, reset link now
[354132.699047] ata2: device plugged sstatus 0x133
[354134.169678] ata1: link is slow to respond, please be patient (ready=0)
[354138.615499] ata1: softreset failed (device not ready)
[354138.621265] ata1: SRST fail, set srst fail flag
[354139.505476] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[354139.512518] ata1: link reset sucessfully clear error flags
[354139.523498] ata1.00: disabling queued TRIM support
[354139.531104] ata1.00: disabling queued TRIM support
[354139.531118] ata1.00: configured for UDMA/133
[354139.536008] ata1: wake up successful, the reset fail can be ignored
[354141.227386] ata2: link is slow to respond, please be patient (ready=0)
[354145.624194] ata2: softreset failed (device not ready)
[... 99 lines cut ...]

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:312456 errors:0 dropped:0 overruns:0 frame:0
          TX packets:312456 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:19569824 (18.6 MiB)  TX bytes:19569824 (18.6 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tun1000   Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet addr:169.254.241.92  P-t-P:169.254.241.92  Mask:255.255.248.0
          inet6 addr: fec0:a2b2:16::15c/112 Scope:Site
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1400  Metric:1
          RX packets:4368 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2051 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:100 
          RX bytes:388880 (379.7 KiB)  TX bytes:501322 (489.5 KiB)

imsoso, 2022-04-15:
same situation on release: 2.15.0-5 DSM7.x
device ds920+，[UGREEN USB C to 2.5G Ethernet Adapter

bb-qq, 2022-04-23:
I could not find any output about driver loading from your log. Could you attach the full dmesg logs as a file?

bobk88, 2022-04-23:
See attached. Thanks!

[2022-04-22 dmesg.txt](https://github.com/bb-qq/r8152/files/8546195/2022-04-22.dmesg.txt)

bb-qq, 2022-04-23:
Sorry, the logs from the beginning are not included because it has been a long time since the startup. Can you please reboot the NAS once and get the logs again?

bobk88, 2022-04-23:
Here's the output after restart. Is this better?

[2022-04-22 dmesg.txt](https://github.com/bb-qq/r8152/files/8546215/2022-04-22.dmesg.txt)

bb-qq, 2022-04-23:
Thank you. I found the current version of driver does not support this device.
So I created new the release. Could you try it?
https://github.com/bb-qq/r8152/releases/download/2.15.0-7/r8152-apollolake-2.15.0-7.spk

bobk88, 2022-04-23:
Yes, it's working! Thanks very much!

At first it was set to "Manual Configuration" in the "Network Interface" settings, but I changed it to DHCP. Is that OK?

Are there any other settings I need to change to use it?

[image]

bb-qq, 2022-04-23:
> Is that OK?

It is OK. Many people use this driver with DHCP.

bb-qq, 2025-01-02:
I will close this issue as the problem seems to be resolved.

closed 2025-01-02 by bb-qq