bb-qq/r8152 #211: Issue using XIAOLO gigabit ethernet adapter RTL8156B

tissme971, 2022-04-22:
> Seems like i can't make it work with my adapter.
When installed and using the sudo command line, nothing happened.
I launched the manual installation of the package and this time the package installed but my adapter still not working

> Synology DS918+
>     - synology_apollolake_918+
>     - DSM 7.0.1-42218 Update 3
> - XIAOLO Type C-2.5G-DE

> - NAS directly connected to the 2,5G port of my router

> - cable category Cat.7

### Output of ``dmesg`` command

````
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.180+ (root@build15) (gcc version 7.5.0 (GCC) ) #42218 SMP Mon Oct 18 19:17:56 CST 2021
[    0.000000] Command line: root=/dev/md0 earlyprintk=apl console=ttyS2,115200n8 ihd_num=4 netif_num=2 HddHotplug=1 SataPortMap=23 sata_remap=0>2:1>3:2>0:3>1 syno_hw_version=DS918+ vender_format_version=2 syno_hdd_detect=18,179,176,175 syno_hdd_enable=21,20,19,9 syno_usb_vbus_gpio=13@0000:00:15.0@1,11@0000:00:15.0@2 sn=19C0PDN166600 macs=001132c1ba81,001132c1ba82
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000] x86/fpu: xstate_offset[3]:  960, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]: 1024, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Supporting XSAVE feature 0x01: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x02: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x08: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x10: 'MPX CSR'
[    0.000000] x86/fpu: Enabled xstate features 0x1b, context size is 1088 bytes, using 'standard' format.
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000086000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000000fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000012151000-0x0000000067c17fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067c18000-0x0000000067c18fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000067c19000-0x0000000067c42fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000067c43000-0x0000000067c7dfff] usable
[    0.000000] BIOS-e820: [mem 0x0000000067c7e000-0x0000000068c7dfff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000068c7e000-0x000000007822efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007822f000-0x000000007aebefff] reserved
[    0.000000] BIOS-e820: [mem 0x000000007aebf000-0x000000007afbefff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007afbf000-0x000000007affefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007afff000-0x000000007affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007b000000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000d3709000-0x00000000d370afff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000e3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff800000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000047fffffff] usable
[    0.000000] bootconsole [earlyser0] enabled
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x01dd9918-0x01dd9927] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x0000000000057fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000058000-0x0000000000058fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000059000-0x0000000000085fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000000086000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x0000000001dd9917] usable
[    0.000000] reserve setup_data: [mem 0x0000000001dd9918-0x0000000001dd9927] usable
[    0.000000] reserve setup_data: [mem 0x0000000001dd9928-0x000000000fffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000010000000-0x0000000012150fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000012151000-0x0000000067c17fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000067c18000-0x0000000067c18fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000067c19000-0x0000000067c42fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000067c43000-0x0000000067c7dfff] usable
[    0.000000] reserve setup_data: [mem 0x0000000067c7e000-0x0000000068c7dfff] reserved
[... 783 lines cut ...]
[  106.844670] e1000e: Copyright(c) 1999 - 2019 Intel Corporation.
[  106.850664] Intel(R) Gigabit Ethernet Linux Driver - version 5.3.5.22
[  106.850665] Copyright(c) 2007 - 2018 Intel Corporation.
[  106.887419] systemd[1]: Created slice ActiveInsight's slice.
[  106.898753] systemd[1]: Starting ActiveInsight's slice.
[  106.900040] igb 0000:04:00.0: added PHC on eth0
[  106.900043] igb 0000:04:00.0: Intel(R) Gigabit Ethernet Linux Driver
[  106.900045] igb 0000:04:00.0: eth0: (PCIe:2.5GT/s:Width x1) 
[  106.900049] igb 0000:04:00.0 eth0: MAC: 00:11:22:33:44:55
[  106.900055] igb 0000:04:00.0: eth0: PBA No: Unknown
[  106.900067] igb 0000:04:00.0: LRO is disabled
[  106.900069] igb 0000:04:00.0: Using MSI-X interrupts. 1 rx queue(s), 1 tx queue(s)
[  106.938857] zram: Added device: zram0
[  106.939152] zram: Added device: zram1
[  106.939437] zram: Added device: zram2
[  106.939708] zram: Added device: zram3
[  106.954995] igb 0000:05:00.0: added PHC on eth1
[  106.954998] igb 0000:05:00.0: Intel(R) Gigabit Ethernet Linux Driver
[  106.955000] igb 0000:05:00.0: eth1: (PCIe:2.5GT/s:Width x1) 
[  106.955003] igb 0000:05:00.0 eth1: MAC: 00:11:22:33:44:55
[  106.955008] igb 0000:05:00.0: eth1: PBA No: Unknown
[  106.955020] igb 0000:05:00.0: LRO is disabled
[  106.955022] igb 0000:05:00.0: Using MSI-X interrupts. 1 rx queue(s), 1 tx queue(s)

````

tissme971, 2022-04-22:
Part 2 of dmesg + lsusb + ifconfig -a
````
[  107.014062] systemd[1]: Mounted Debug File System.
[  107.023809] systemd[1]: Mounted Temporary Directory.
[  107.034507] systemd[1]: Started Set boot sequence config.
[  107.045916] systemd[1]: Listening on Journal Socket (/dev/log).
[  107.058655] systemd[1]: Started unbind USB device.
[  107.068920] systemd[1]: Started Synology Task Scheduler Init.
[  107.071766] usb-storage 1-4:1.0: USB Mass Storage device detected
[  107.071929] scsi host5: usb-storage 1-4:1.0
[  107.072070] usbcore: registered new interface driver usb-storage
[  107.082716] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[  107.088620] ehci-pci: EHCI PCI platform driver
[  107.094922] md: md1 stopped.
[  107.100448] uhci_hcd: USB Universal Host Controller Interface driver
[  107.122739] input: Power Button as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input0
[  107.130466] md: bind<sdc2>
[  107.130768] md: bind<sda2>
[  107.130991] md: bind<sdb2>
[  107.132617] md/raid1:md1: active with 3 out of 4 mirrors
[  107.136289] md1: detected capacity change from 0 to 2147418112
[  107.153525] ACPI: Power Button [PWR2]
[  107.157692] systemd[1]: Started check integrity of rd.gz decompression.
[  107.172075] systemd[1]: Started Link coredump path to volume if possible.
[  107.186513] systemd[1]: Started Set hostname.
[  107.195827] systemd[1]: Started Load ntb_brd module.
[  107.206524] systemd[1]: Started Set time zone information to kernel.
[  107.220279] systemd[1]: Started Disabling fan checking flag ....
[  107.233032] systemd[1]: Reached target Sockets.
[  107.242274] systemd[1]: Starting Sockets.
[  107.263084] systemd[1]: Starting Journal Service...
[  107.268722] systemd[1]: Reached target Local File Systems.
[  107.278576] systemd[1]: Starting Local File Systems.
[  107.290566] systemd[1]: Starting Set boot sequence config for temporary use...
[  107.299908] systemd[1]: Starting D-Bus Session Message Bus...
[  107.312648] systemd[1]: Starting D-Bus System Message Bus...
[  107.323905] systemd[1]: Starting udev Coldplug all Devices...
[  107.335109] systemd[1]: Starting Mount usb fs...
[  107.348198] systemd[1]: Started Set boot sequence config for temporary use.
[  107.369470] systemd[1]: Started Journal Service.
[  107.396855] systemd-journald[5550]: Received request to flush runtime journal from PID 1
[  107.534758] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[  107.542146] [drm] Driver supports precise vblank timestamp query.
[  107.549070] vgaarb: device changed decodes: PCI:0000:00:02.0,olddecodes=io+mem,decodes=io+mem:owns=io+mem
[  107.561931] [drm] Finished loading DMC firmware i915/bxt_dmc_ver1_07.bin (v1.7)
[  107.573172] [drm] Initialized i915 1.6.0 20171222 for 0000:00:02.0 on minor 0
[  107.582595] ACPI: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
[  107.597507] acpi device:4c: registered as cooling_device4
[  107.606247] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input1
[  107.622555] [drm] Cannot find any crtc or sizes
[  107.634916] i915 0000:00:02.0: fb0: inteldrmfb frame buffer device
[  107.694585] usbcore: registered new interface driver usblp
[  107.716818] usbcore: registered new interface driver usbhid
[  107.725086] usbhid: USB HID core driver
[  107.795099] synobios open /dev/ttyS1 success
[  107.806086] Adding 2097084k swap on /dev/md1.  Priority:-1 extents:1 across:2097084k 
[  107.815124] correction with 0x00
[  107.818903] zram0: detected capacity change from 0 to 2494562304
[  107.826616] 2022-4-21 23:33:33 UTC
[  107.835745] synobios: load, major number 201
[  107.836969] Adding 2436092k swap on /dev/zram0.  Priority:1 extents:1 across:2436092k SS
[  107.840004] zram1: detected capacity change from 0 to 2494562304
[  107.845662] Adding 2436092k swap on /dev/zram1.  Priority:1 extents:1 across:2436092k SS
[  107.848610] zram2: detected capacity change from 0 to 2494562304
[  107.854045] Adding 2436092k swap on /dev/zram2.  Priority:1 extents:1 across:2436092k SS
[  107.857277] zram3: detected capacity change from 0 to 2494562304
[  107.862866] Adding 2436092k swap on /dev/zram3.  Priority:1 extents:1 across:2436092k SS
[  107.897603] Brand: Synology
[  107.897609] Model: DS-918+
[  107.897613] This is default settings: set group disks wakeup number to 1, spinup time deno 1
[... 531 lines cut ...]
          inet6 addr: fe80::2e0:4cff:fe68:a97/64 Scope:Link
          inet6 addr: 2a01:cb08:994:e100:2e0:4cff:fe68:a97/64 Scope:Global
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:219 errors:0 dropped:0 overruns:0 frame:0
          TX packets:84 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:54272 (53.0 KiB)  TX bytes:17658 (17.2 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:482908 errors:0 dropped:0 overruns:0 frame:0
          TX packets:482908 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:572276690 (545.7 MiB)  TX bytes:572276690 (545.7 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````
Thank you in advance for your reply :)

tissme971, 2022-04-22:
[image]
[image]

bb-qq, 2022-04-23:
```
[62442.856190] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[62442.856391] r8152 2-2:1.0 eth2: carrier on
[62551.920832] r8152 2-2:1.0 eth2: carrier off
```
```
eth2 Link encap:Ethernet HWaddr 00:E0:4C:68:0A:97
inet6 addr: fe80::2e0:4cff:fe68:a97/64 Scope:Link
inet6 addr: 2a01:cb08:994:e100:2e0:4cff:fe68:a97/64 Scope:Global
UP BROADCAST MULTICAST MTU:1500 Metric:1
RX packets:219 errors:0 dropped:0 overruns:0 frame:0
TX packets:84 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:54272 (53.0 KiB) TX bytes:17658 (17.2 KiB)
```

The driver seems to be working fine. Have you rebooted the NAS, and are the symptoms the same?

tissme971, 2022-04-23:
Yes, i did reboot several times but still no ping when adapter connected to my Nas :(

tissme971, 2022-04-23:
Hello, i made it work by launching ifconfig eth2 up
It's now connected and all is working !
Thanks for all of your job on this driver and thanks for taking time to answer me :)

closed 2022-04-25 by tissme971