bb-qq/r8152 #214: Could not fetch the default gateway ip

danielsongchn, 2022-04-25:
Hi I used the driver and the adapter on my DS 1821+ at DSM 6.2.4. At first it worked flawlessly. Then I installed VMM and ran DSM7.1 in it. The problem came after I uninstalled the VMM. 
The adapter can be found. But there is no default gateway. Now I have to connect another 1G cable to make NAS connect to the internet.
I tried to uninstall the driver and reinstall it. It did not work. I used another network adapter, which worked well on my Mac, and still, there is the problem. It seems like it is related to the DSM or the driver, not a hardware issue.

bb-qq, 2022-04-29:
Are you saying that you now have a problem when using DHCP without a problem when setting the IP address statically?

danielsongchn, 2022-04-29:
No. The 2.5G adapter can not find the default gateway without the help of another 1G network. Once there is a default gateway in General in DSM, it works well even if I unplugged the 1G network. But it does not work anymore if I restart NAS.
I guess it is related to the DSM. Maybe something is not right during the installation and uninstallation of VMM. To have a stable connection with high speed, I purchased E10G18-T1. At first, it could not be recognized and the light was off. Eventually, I managed to make it work by reinstallation of VMM. Now it still works after I uninstalled VMM. But I just have to find another place to make full use of this 2.5G adapter.

bb-qq, 2022-05-01:
I assume it is a routing issue since the default gateway is not an interface-specific setting, but I can't imagine what is going on.

If you could write down the output of `ifconfig -a` and `netstat -nr` for the cases where you are communicating well and the cases where you are not, respectively, I might be able to provide some advice.

danielsongchn, 2022-05-01:
Thanks very much for the kind suggestion. I am not good at using these commands. I think I will just let it go. I really appreciate your work and help.

bb-qq, 2023-08-13:
I will close this issue, but the latest driver has a lot of improvements in it, so the symptoms may be resolved.

closed 2023-08-13 by bb-qq