bb-qq/r8152 #216: I am stuck on the "install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su"

coldheater1, 2022-04-26:
### Description of the problem

I am stuck on the "install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su"

### Description of your products

> - the product name of your NAS model - DS220+
>     - and output of ``uname -a`` command
uname -a
Linux Coldheater 4.4.180+ #42218 SMP Mon Oct 18 19:16:55 CST 2021 x86_64 GNU/Linux synology_geminilake_220+

>     - DSM version7.0.01-42218
> - the product name of the ethernet dongle - Pluggable 2.5 Gig USC adapter
### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name - Laptop with no eithernet port that is why I am using the dongle
>     - Ethernet switch: model name  - Zyxel 5-Port 2.5G Multi-Gigabit Unmanaged Switch for Home Entertainment or SOHO Network [MG-105]
> - cable category Cat 6

### Output of ``dmesg`` command

```` Its a ton of stuff
(paste output here, or attach as a file)
````

### Output of ``lsusb`` command
````
lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F00911771345A48)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
(paste output here)
````

### Output of ``ifconfig -a`` command

````
ifconfig -a
docker0   Link encap:Ethernet  HWaddr 02:42:8F:C3:49:C5
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth0      Link encap:Ethernet  HWaddr 00:11:32:CC:97:09
          inet addr:192.168.1.2  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fecc:9709/64 Scope:Link
          inet6 addr: 2605:a601:a61d:b100:211:32ff:fecc:9709/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:30789 errors:0 dropped:0 overruns:0 frame:0
          TX packets:13279 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:8743696 (8.3 MiB)  TX bytes:10223195 (9.7 MiB)
          Interrupt:84

eth1      Link encap:Ethernet  HWaddr 00:11:32:CC:97:0A
          inet addr:169.254.5.5  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:85 base 0x2000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:8541 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8541 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:1292233 (1.2 MiB)  TX bytes:1292233 (1.2 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

STARchDUST, 2022-04-27:
I suppose you gotta try installing it and let it fail for the first time, then try the SSH command?

coldheater1, 2022-04-27:
That is the first thing that I did.

STARchDUST, 2022-04-27:
You DSM user has all admin rights, correct?

STARchDUST, 2022-04-27:
When I try it now, it gives me the same thing as on yours. But it's probably because I've already done the SSH command and installation before?

coldheater1, 2022-04-27:
Yes I have the full admin rights.

coldheater1, 2022-04-28:
I figured it out.  When I tried to install it failed and I closed the window.  You have to wait for two failed attempts.  The first one will fail to install and the other one will fail to load Driver for Aquatia AQC11U.  Then you can close out the window. Run the sudo install -m 4755 -o root -D /var/packages/aqc111/target/aqc111/spk_su /opt/sbin/spk_su command and it worked. Thanks everyone.

coldheater1, 2022-04-28:
However, after installing it, I plugged in my adapter, stopped the driver and started it again, then I check my network interface and it does not show up. Any suggestions?

STARchDUST, 2022-04-28:
That's odd as well...leave it in and reboot, it should show up? Some article said if your NAS has a front socket then use that becoz the rear ones on some models may be under powered...

coldheater1, 2022-04-28:
I left it in last night and rebooted and it still does not show up. I read about using the front port and that is what I have been using. I tried the back just to see if it would recognize it and it doesn't either.

bb-qq, 2022-05-01:
Did you check using the same USB cable? If so, changing the USB port connected to the NAS and seeing if there is a change in the lsusb output might give you a clue.

bb-qq, 2024-06-24:
I would like to close this issue as I have not heard back from you in a long time.
If you have any questions, feel free to re-open.