bb-qq/r8152 #217: TX issues

STARchDUST, 2022-04-27:
### Description of the problem

Unable to download from NAS via the 2.5G dongle. Device hangs within 1 minute after download started. Cannot stream videos etc. via the dongle. **But uploading is perfectly fine and fast.**

Same issue on Windows Explorer, browsers and the DS File app on iOS. 

Dongle performs fine on PC.

Driver version: all 2.15 drivers tested, same issue. Current version 2.15.0-4

### Description of your products

> - Synology DS118
>     - DSM version 7.0.1-42218 Update 3
> - the product name of the ethernet dongle: Unique Flatworm AX88U with RTL8156B

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: TP Link TL-NG421
>     - Ethernet switch: Mikrotik 5009
> - cable category  Cat.7

### Output of ``dmesg`` command

````
(paste output here, or attach as a file)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

bb-qq, 2022-04-29:
Which port on the switch are you using? The 2.5Gbps port seems to have some issues.
https://www.reddit.com/r/mikrotik/comments/rq7ytu/rb5009_25gbe_slower_than_1gbe/

bb-qq, 2022-12-04:
I decided to merge topics on rtd1296 into issue #275.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq