bb-qq/r8152 #219: Could you help me a driver for DS 414j?Thank you!

smartsino, 2022-05-06:
### Description of the problem
Could you help me a driver for DS 414j? Thank you!
My email:smartsino@gmail.com

> Replace this line with your answer.
Ds 414j
DSM 7.1

### Description of your products

> - the product name of your NAS model
>     - and output of ``uname -a`` command
>     - DSM version x.x
> - the product name of the ethernet dongle

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

### Output of ``dmesg`` command

````
(paste output here, or attach as a file)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

bb-qq, 2022-08-20:
I added the driver for comcerto2k platforms such as DS414j.
https://github.com/bb-qq/r8152/releases/download/2.16.3-1/r8152-comcerto2k-2.16.3-1.spk

bb-qq, 2022-12-04:
I close this issue. If you have any problems, please create a new issue.

closed 2022-12-04 by bb-qq