bb-qq/r8152 #221: XTVTX Adaptater USB 3.0 2.5Gbps not recognised by USB port on DS415Play

bertrandtriplet, 2022-05-19:
### Description of the problem

I got a cheap (less than 20€) USB3.0 2.5Gbps adapter from Amazon, which bear a RTL8156B chipset.
The Nas is an old DS415Play with DSM 6.2. The driver (evansport 2.15.0-1) is recognized
But when i plug the dongle in the USB 3.0 port nothing goes up.
I see the lights from the led, but that all.
The dongle seems to be recognised on a DS920+ with no problem.
It also seems to be recognised on the usb 2.0 (using lsusb command)

Is it a faulty dongle or a problem with the driver ?

### Description of your products

Nas :
DS415Play
Uname :
Linux DataStation 3.2.40 #25556 SMP PREEMPT Thu Jul 1 14:28:54 CST 2021 i686 GNU/Linux synology_evansport_415play
DSM version 6.2
Dongle :
XTVTX Adaptateur USB 3.0 vers RJ45 Gigabit Ethernet,2.5Gbps RTL8156B chipset

### Description of your environment

Connection to  PC via 2.5 Gbps Ethernet Switch.

### Output of ``dmesg`` command

[dmesg result.txt](https://github.com/bb-qq/r8152/files/8732500/dmesg.result.txt)

### Output of ``lsusb`` command

Scenario 1 : Dongle connected to front USB port :

|__usb1          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.0) hub
  |__1-1         0bda:8156:3100 00  2.10  480MBit/s 200mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
|__usb2          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.1) hub
|__usb3          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.2) hub
|__usb4          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (Linux 3.2.40 etxhci_hcd-161118 Etron xHCI Host Controller 0000:03:00.0) hub
|__usb5          1d6b:0003:0302 09  3.00 5000MBit/s   0mA 1IF  (Linux 3.2.40 etxhci_hcd-161118 Etron xHCI Host Controller 0000:03:00.0) hub

Scenario 2 : random USB Key connected to back USB3 port :

|__usb1          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.0) hub
|__usb2          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.1) hub
|__usb3          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.2) hub
|__usb4          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (Linux 3.2.40 etxhci_hcd-161118 Etron xHCI Host Controller 0000:03:00.0) hub
  |__4-2         0951:1665:0000 00  2.00  480MBit/s 200mA 1IF  (Kingston DataTraveler 2.0 20CF302E25F1AF819C0A0573)
|__usb5          1d6b:0003:0302 09  3.00 5000MBit/s   0mA 1IF  (Linux 3.2.40 etxhci_hcd-161118 Etron xHCI Host Controller 0000:03:00.0) hub

Network adapter connected to USB3 port :

|__usb1          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.0) hub
|__usb2          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.1) hub
|__usb3          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.2) hub
|__usb4          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (Linux 3.2.40 etxhci_hcd-161118 Etron xHCI Host Controller 0000:03:00.0) hub
|__usb5          1d6b:0003:0302 09  3.00 5000MBit/s   0mA 1IF  (Linux 3.2.40 etxhci_hcd-161118 Etron xHCI Host Controller 0000:03:00.0) hub

enumeration from USB5 line takes a lot of time.

USB3 HDD connected to USB3 port :

|__usb1          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.0) hub
|__usb2          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.1) hub
|__usb3          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (ehci_hcd 0000:01:0d.2) hub
|__usb4          1d6b:0002:0302 09  2.00  480MBit/s   0mA 1IF  (Linux 3.2.40 etxhci_hcd-161118 Etron xHCI Host Controller 0000:03:00.0) hub
|__usb5          1d6b:0003:0302 09  3.00 5000MBit/s   0mA 1IF  (Linux 3.2.40 etxhci_hcd-161118 Etron xHCI Host Controller 0000:03:00.0) hub
  |__5-1         2109:0715:0000 00  3.10 5000MBit/s 224mA 1IF  (VLI Manufacture String VLI Product String 000000123DE9)

### Output of ``ifconfig -a`` command

eth0      Link encap:Ethernet  HWaddr 00:11:32:43:65:03
          inet addr:192.168.1.170  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2a01:e0a:992:4350:211:32ff:fe43:6503/64 Scope:Global
          inet6 addr: fe80::211:32ff:fe43:6503/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1705897857 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1020795324 errors:0 dropped:0 overruns:0 carrier:57
          collisions:0 txqueuelen:1000
          RX bytes:3854504891 (3.5 GiB)  TX bytes:680515359 (648.9 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:16436  Metric:1
          RX packets:69476 errors:0 dropped:0 overruns:0 frame:0
          TX packets:69476 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:9778609 (9.3 MiB)  TX bytes:9778609 (9.3 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

bb-qq, 2022-05-28:
My driver is not involved in the process until it shows up on the lsusb output, so I suspect a hardware(NAS/dongle) or cable problem. The same goes for the symptom of the device recognized as a USB 2.0 connection.

If the driver is not working correctly for the setup that includes the device in the output of lsusb, I can investigate further by looking at the output of dmesg.

bb-qq, 2022-08-11:
I close this issue due to no response. Please feel free to reopen if you have additional questions.

closed 2022-08-11 by bb-qq