bb-qq/r8152 #224: The DS218 has successfully installed the rtd1296 driver. However, it only links up to 1000Base-T.

kimiakiyamada, 2022-06-14:
### Description of the problem
The DS218 has successfully installed the rtd1296 driver([r8152-rtd1296-2.15.0-7.spk](https://github.com/bb-qq/r8152/releases/download/2.15.0-7/r8152-rtd1296-2.15.0-7.spk)). However, it only links up to 1000Base-T.

> Replace this line with your answer.

### Description of your products

> - the product name of your NAS model
DS218
>     - and output of ``uname -a`` command
Linux TAKOYAKI 4.4.180+ #42661 SMP Fri May 27 17:13:39 CST 2022 aarch64 GNU/Linux synology_rtd1296_ds218
>     - DSM version x.x
DSM 7.1-42661 Update 2

> - the product name of the ethernet dongle
Planex USB-LAN2500R
### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
using Ethernet switch: NEC Aterm bl1000hw, port4 : 10GBASE-T
> - cable category (Cat.5e or Cat.6 or Cat.7)
Cat.5e, Cat.6, Cat.6A

### Output of ``dmesg`` command

````
[dmesg_20220611.txt](https://github.com/bb-qq/r8152/files/8895326/dmesg_20220611.txt)

````

### Output of ``lsusb`` command

````
|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb3          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb4          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
  |__4-1         0bda:8156:3000 00  2.10  480MBit/s 500mA 1IF  (Planex USB 10/100/1G/2.5G LAN 000000001)
|__usb5          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
|__usb6          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller xhci-hcd.8.auto) hub
|__usb7          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller xhci-hcd.8.auto) hub````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 00:11:32:F5:E4:13
          inet addr:192.168.1.254  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fef5:e413/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:391878 errors:0 dropped:0 overruns:0 frame:0
          TX packets:107936 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:93746859 (89.4 MiB)  TX bytes:25975939 (24.7 MiB)
          Interrupt:11

eth1      Link encap:Ethernet  HWaddr 1C:C0:35:03:FC:E3
          inet addr:192.168.0.17  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::1ec0:35ff:fe03:fce3/64 Scope:Link
          inet6 addr: 240f:6d:2042:1:1ec0:35ff:fe03:fce3/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:130023 errors:0 dropped:0 overruns:0 frame:0
          TX packets:259873 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:27811313 (26.5 MiB)  TX bytes:303774344 (289.7 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:10567 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10567 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:2641382 (2.5 MiB)  TX bytes:2641382 (2.5 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

kimiakiyamada, 2022-06-14:
I'm sorry, I've just confirmed that it's a known issue below. Please close this matter.
https://github.com/bb-qq/r8152/issues/136

bb-qq, 2022-06-14:
```
  |__4-1         0bda:8156:3000 00  2.10  480MBit/s 500mA 1IF  (Planex USB 10/100/1G/2.5G LAN 000000001)
```

This log indicates that the device is recognized as a USB 2.0 device. So changing connecting port might resolve the issue.
But as you pointed out, this model seems to have stability issues.

bb-qq, 2022-12-04:
I decided to merge topics on rtd1296 into issue #275.
If you notice anything new, please comment there.

closed 2022-12-04 by bb-qq