bb-qq/r8152 #225: link down & link up

dozer47528, 2022-06-15:
### Description of the problem

I can find the log about link down and link up in synology logs.

[image]

After that, I checked the syslog.log from ternimal:

```
2022-06-15T04:28:11+08:00 HOME syslog-ng[6527]: Configuration reload request received, reloading configuration;
2022-06-15T04:28:11+08:00 HOME syslog-ng[6527]: Configuration reload finished;
2022-06-15T08:04:08+08:00 HOME syslog-ng[6527]: Configuration reload request received, reloading configuration;
2022-06-15T08:04:08+08:00 HOME syslog-ng[6527]: Configuration reload finished;
2022-06-15T10:30:53+08:00 HOME syslog-ng[6527]: Error reloading configuration; reason='Unable to trigger a reload while another reload attempt is in progress'
2022-06-15T10:30:53+08:00 HOME syslog-ng[6527]: Configuration reload request received, reloading configuration;
2022-06-15T10:30:53+08:00 HOME syslog-ng[6527]: Configuration reload finished;
2022-06-15T10:30:58+08:00 HOME syslog-ng[6527]: Configuration reload request received, reloading configuration;
2022-06-15T10:30:58+08:00 HOME syslog-ng[6527]: Configuration reload finished;
2022-06-15T10:30:58+08:00 HOME syslog-ng[6527]: Configuration reload request received, reloading configuration;
2022-06-15T10:30:58+08:00 HOME syslog-ng[6527]: Configuration reload finished;
2022-06-15T10:30:58+08:00 HOME syslog-ng[6527]: Configuration reload request received, reloading configuration;
2022-06-15T10:30:58+08:00 HOME syslog-ng[6527]: Configuration reload finished;
2022-06-15T10:30:58+08:00 HOME syslog-ng[6527]: Configuration reload request received, reloading configuration;
2022-06-15T10:30:58+08:00 HOME syslog-ng[6527]: Configuration reload finished;
```

### Description of your products

> - the product name of your NAS model
>     - Linux HOME 4.4.180+ #42661 SMP Fri Apr 1 15:33:03 CST 2022 x86_64 GNU/Linux synology_geminilake_920+
>     - DSM 7.1-42661 Update 1
> - the product name of the ethernet dongle
>     - [ASUSTOR AS-U2.5G](https://amzn.to/2ZRx1pi)

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - Ethernet switch: TL-SH1005
> - cable category: Cat.5e

### Output of ``dmesg`` command

[dmesg.log](https://github.com/bb-qq/r8152/files/8905782/dmesg.log)

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         1a40:0101:0111 09  2.00  480MBit/s 100mA 1IF  ( ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-2.4     0665:5161:0002 00  1.10  1.5MBit/s 100mA 1IF  (INNO TECH USB to Serial ffffffd8ffffffb9ffffffd0ffffffa2)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F000898AA13F221)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 0013000001)````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:74:6E:36:44
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:74ff:fe6e:3644/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:48178 errors:0 dropped:0 overruns:0 frame:0
          TX packets:52378 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:4033173 (3.8 MiB)  TX bytes:5821423 (5.5 MiB)

docker472 Link encap:Ethernet  HWaddr 2E:75:3D:4F:B2:C0
          inet6 addr: fe80::2c75:3dff:fe4f:b2c0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:404 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4755 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:72002 (70.3 KiB)  TX bytes:1185366 (1.1 MiB)

dockere0c Link encap:Ethernet  HWaddr E2:55:0F:6E:1E:8E
          inet6 addr: fe80::e055:fff:fe6e:1e8e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:47774 errors:0 dropped:0 overruns:0 frame:0
          TX packets:51979 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:4635663 (4.4 MiB)  TX bytes:5645887 (5.3 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:D5:C4:EB
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:97 base 0xa000

eth1      Link encap:Ethernet  HWaddr 00:11:32:D5:C4:EC
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:96 base 0xc000

eth2      Link encap:Ethernet  HWaddr 00:E0:4C:68:15:D0
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:754916974 errors:0 dropped:0 overruns:0 frame:0
          TX packets:257424670 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1079600220163 (1005.4 GiB)  TX bytes:26572830930 (24.7 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:318776 errors:0 dropped:0 overruns:0 frame:0
          TX packets:318776 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:62044116 (59.1 MiB)  TX bytes:62044116 (59.1 MiB)

ovs-syste Link encap:Ethernet  HWaddr D2:91:17:DC:EF:27
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth0  Link encap:Ethernet  HWaddr 00:11:32:D5:C4:EB
          inet addr:192.168.11.11  Bcast:192.168.11.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:D5:C4:EC
          inet addr:169.254.86.175  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:12 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:2738 (2.6 KiB)  TX bytes:0 (0.0 B)

ovs_eth2  Link encap:Ethernet  HWaddr 00:E0:4C:68:15:D0
          inet addr:192.168.11.11  Bcast:192.168.11.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:15d0/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:745615879 errors:0 dropped:0 overruns:0 frame:0
          TX packets:248873206 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:1077682515177 (1003.6 GiB)  TX bytes:24752576206 (23.0 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap021132 Link encap:Ethernet  HWaddr 62:53:D5:19:97:C8
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:13942667 errors:0 dropped:0 overruns:0 frame:0
          TX packets:14369303 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:10373851703 (9.6 GiB)  TX bytes:10506527909 (9.7 GiB)
````

bb-qq, 2022-07-30:
Can you fix the link speed by configuration of the switch?
If you can not, could you try direct connection to PC?

stepman0, 2022-10-06:
I have the same issue with a Club3D USB adapter. Sporadically the link goes down and after 1-4 seconds up again.

My setup:
DS918+
DSM 7.1.1-42962 Update 1
Driver 2.15.0-10 DSM7.x
Cable connection to my FritzBox 6660 Router's 2.5G port.

Any suggestion?

bb-qq, 2022-10-09:
You can isolate the cause of the problem by examining what happens when the r8152 dongle is connected to a PC, without changing the environment such as cables and switches.

markoweb2, 2022-12-10:
I have the same problem with the same setup.
DS918+, Club3D 2.5G adapter, DSM 7.1.1, direct connection with PC (Intel I226V).

For me, the probleem seems to be related to Jumbo Frames, tried MTU with 8000 and 9000.
When I put my PC to sleep, Synology side keeps getting link down/up events every few seconds on the 2.5G adapter, thus preventing Synology itself from sleeping.
Without Jumbo Frames, all seems to be good.

What is also strange, that after this issue occurs, then I can't use Jumbo Frames from Synology side.
The drop down list doesn't have anything above 1500 MTU and the previous manually entered value of 8000 is considered invalid in DSM.

The only solution is to restart the NAS, to get a higher MTU back again.

This seems to be some sort of bug in the synology Realtek driver.
Any chance of fixing it?

Also, when I unplug the Club3D adapter and put it back in again, the network interface for it does not appear in DSM.
Restarting the Realtek driver package does not help.
Have to restart DSM for it to start working.
Hoping this could also be fixed somehow...
(did not test if the interface not appearing after plugging back in, is because the MTU and link down/up issue had occured or does it not appear after a steady working state as well)

bb-qq, 2022-12-31:
> This seems to be some sort of bug in the synology Realtek driver.
> Any chance of fixing it?

I have no idea about driver fixes, but you might want to try a smaller MTU such as 4000.

> Also, when I unplug the Club3D adapter and put it back in again, the network interface for it does not appear in DSM.

This driver does not support hot-plugging by default. To enable hot-plugging, install udev rules with the following command:
```
sudo bash /var/packages/r8152/scripts/install-udev-rules
```

bb-qq, 2023-07-16:
The problem of unstable links may be avoided by setting the driver.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#fix-to-25gbps-r8152-specific

Also, can anyone confirm what happens when using the latest driver?

biblia19, 2023-08-26:
> The problem of unstable links may be avoided by setting the driver. https://github.com/bb-qq/r8152/wiki/Troubleshooting#fix-to-25gbps-r8152-specific
> 
> Also, can anyone confirm what happens when using the latest driver?

Hello I'm use MTU 9000 on DS920+ with [NEXI NX-U3025G](https://prod.danawa.com/info/?pcode=12204683)(Type-A, Korea only)
and [r8152-geminilake-2.17.1-1_7.2.spk](https://github.com/bb-qq/r8152/releases/download/2.17.1/r8152-geminilake-2.17.1-1_7.2.spk)

```
sudo ethtool -s eth2 autoneg on advertise 0x802f
```

This command works for me perfectly
No more useless logs **link down & up** and **DDNS updates**
Thank you for updates

hhessel, 2023-08-28:
I cannot really verify my claims but situation was even more dire for me. I'm using a Netgear XS508M 8 Port 10GbE Multi-Gigabit Switch for my home networking. Every link- down and up down message for the eth2 device (the Club3D usb network adapter) lead to a situation where the entire Synology froze up and wasn't reachable anymore. 

I replaced the CAT 5E cable between the USB device and the switch with a CAT6 cable and put `sudo ethtool -s eth2 autoneg on advertise 0x802f` Looks stable so far but fingers crossed because I can't really proof that it's directly linked to the external USB device.

bb-qq, 2023-09-07:
Could you paste the `dmesg` output here so that I would like to know exactly what your situation?

hhessel, 2023-09-07:
Hi @bb-qq 

Please find the dumped dmesg output attached here. 

Many thanks 
Henrik 

[dmesg.txt](https://github.com/bb-qq/r8152/files/12545870/dmesg.txt)

bb-qq, 2023-09-10:
```
[  824.646206] r8152 2-1:1.0 eth2: carrier off
[  828.999262] r8152 2-1:1.0 eth2: carrier on
```

You mean that the network is disconnected at the time these logs are displayed, right?

If so, then there is indeed something wrong with the Ethernet layer, and if it is not a problem with other 10gbps devices, I would like to try disabling auto-negotiation on the switch side, is that possible?

hhessel, 2023-09-10:
@bb-qq Thanks for the suggestion. Unfortunately it's not a managed switched. But thanks, will add to this issue if the problem persists.

PenelopeFudd, 2024-02-28:
I've been having this problem as well; carrier shutting down every minute or two for three seconds:
```
$ dmesg -T | tail
[Tue Feb 27 20:39:21 2024] r8152 2-1.4:1.0 eth1: carrier off
[Tue Feb 27 20:39:24 2024] r8152 2-1.4:1.0 eth1: carrier on
[Tue Feb 27 20:41:05 2024] r8152 2-1.4:1.0 eth1: carrier off
[Tue Feb 27 20:41:08 2024] r8152 2-1.4:1.0 eth1: carrier on
[Tue Feb 27 20:41:47 2024] r8152 2-1.4:1.0 eth1: carrier off
[Tue Feb 27 20:41:50 2024] r8152 2-1.4:1.0 eth1: carrier on
[Tue Feb 27 20:42:26 2024] r8152 2-1.4:1.0 eth1: carrier off
[Tue Feb 27 20:42:29 2024] r8152 2-1.4:1.0 eth1: carrier on
```
I've read elsewhere that turning off autonegotiation helps, and 15 minutes ago I tried it and the connection has been stable ever since:
```
ethtool -s eth1 speed 100 duplex full autoneg off
```
Wish I'd known this before I threw away my switch!  :-P

hhessel, 2024-02-28:
@PenelopeFudd It recently switched over to a RTL8152BG device and no problems so far. Will report here if there are similar issues.

bb-qq, 2024-06-15:
I have recently learned that this behavior may be related to Energy Efficient Ethernet. If this is the case, it can be disabled with the following command:
```
ethtool --set-eee eth2 eee off
```

bb-qq, 2024-06-15:
This issue has been created a long time ago and I would like to close.
If you still have problems, please create a new issue individually and fill out the template.

closed 2024-06-15 by bb-qq