bb-qq/r8152 #228: 2.15.0-6 armada38x upload speed low

zhucaidan, 2022-06-24:
### Description of the problem

I'm using UGREEN USB-A to 2.5G adapter, when the driver version is 2.15.0-6,
The upload speed will be locked to 100mb/s, download speed is 220mb/s (whitch is normal).

So, I tried to use old version driver 2.15.0-5, The upload speed is about 125mb/s (still low and unstable),
but the download speed is around 200mb/s (fast, but unstable between 180-210 mb/s).

### Description of your products

> DS216j
>     Linux SynologyNAS 3.10.108 #42661 SMP Fri Apr 8 02:09:08 CST 2022 armv7l GNU/Linux synology_armada38x_ds216j
>     DSM 7.1-42661 Update 2
> UGREEN USB-A to 2.5G Ethernet adapter

### Description of your environment

>TPLINK-2.5G Ethernet switch
>     Windows 11 ASUS motherboard 2.5G Port
>     Realtek PCIe 2.5GbE Family Controller DEV_8125
> Cat 6

### Output of ``dmesg`` command

````
[   71.274013] workqueue: max_active 1024 requested for vhost_scsi is out of range, clamping between 1 and 512
[   73.167253] fuse init (API version 7.22)
[   74.511662] 8021q: 802.1Q VLAN Support v1.8
[   75.557979] findhostd uses obsolete (PF_INET,SOCK_PACKET)
[   77.297832] r8152: module verification failed: signature and/or required key missing - tainting kernel
[   77.444073] usbcore: registered new interface driver r8152
[   77.613832] usb 4-1: reset SuperSpeed USB device number 2 using xhci-hcd
[   77.676731] usb 4-1: Parent hub missing LPM exit latency info.  Power management will be impacted.
[   77.757674] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9bf05680
[   77.766223] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9bf056ac
[   77.774756] xhci-hcd f10f8000.usb3: xHCI xhci_drop_endpoint called with disabled ep 9bf056d8
[   85.419936] eth0:
[   85.421788] link up
[   85.423913] , full duplex
[   85.425058] , speed
[   85.427260] 1 Gbps
[   94.363752] netif_napi_add() called with weight 256 on device eth%d
[   94.433574] r8152 4-1:1.0 eth1: v2.15.0 (2021/04/15)
[   94.438571] r8152 4-1:1.0 eth1: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[   94.454245] r8152 4-1:1.0 eth1: chip rev 14
[   95.900872] IPv6: ADDRCONF(NETDEV_UP): eth1: link is not ready
[  188.382732] eth0:
[  188.384595] link down
[  205.461561] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[  205.468303] r8152 4-1:1.0 eth1: carrier on
[  327.392406] Synotify use 16384 event queue size
````

### Output of ``lsusb`` command

````
|__usb4          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 xhci-hcd xHCI Host Controller f10f8000.usb3) hub
  |__4-1         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 401000001)
````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 00:11:32:6E:F2:55
          inet addr:192.168.1.10  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:2594 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1129 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1024
          RX bytes:918758 (897.2 KiB)  TX bytes:2354128 (2.2 MiB)
          Interrupt:193

eth1      Link encap:Ethernet  HWaddr 08:26:AE:35:31:20
          inet addr:192.168.1.20  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::a26:aeff:fe35:3120/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2667003 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1482441 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3880732872 (3.6 GiB)  TX bytes:2042349542 (1.9 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:6188 errors:0 dropped:0 overruns:0 frame:0
          TX packets:6188 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:855283 (835.2 KiB)  TX bytes:855283 (835.2 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2022-07-30:
What method did you use to measure throughput. Did you use iperf?

zhucaidan, 2022-07-31:
> What method did you use to measure throughput. Did you use iperf?
I'm use SMB 3.0 and AFP to test the speed, they are the same result.

I did try to find a way install iperf on armada38x CPU, but synology does not provide docker on this CPU architecture,

So is kind hard to run a iperf test, do you think there is a possible way to launch the test?

bb-qq, 2022-08-02:
This package seems to work for you.
https://synocommunity.com/package/synocli-monitor#

zhucaidan, 2022-08-02:
> This package seems to work for you. https://synocommunity.com/package/synocli-monitor#

Finally I can test it, Thank you!

**When NAS as server:**

```
Synology -s result:
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec   270 MBytes  2.27 Gbits/sec
[  5]   1.00-2.00   sec   271 MBytes  2.27 Gbits/sec
[  5]   2.00-3.00   sec   260 MBytes  2.18 Gbits/sec
[  5]   3.00-4.00   sec   271 MBytes  2.27 Gbits/sec
[  5]   4.00-5.00   sec   270 MBytes  2.27 Gbits/sec
[  5]   5.00-6.00   sec   261 MBytes  2.19 Gbits/sec
[  5]   6.00-7.00   sec   251 MBytes  2.10 Gbits/sec
[  5]   7.00-8.00   sec   231 MBytes  1.94 Gbits/sec
[  5]   8.00-9.00   sec   269 MBytes  2.26 Gbits/sec
[  5]   9.00-10.00  sec   255 MBytes  2.14 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate     receiver
[  5]   0.00-10.00  sec  2.55 GBytes  2.19 Gbits/sec

PC -c result:
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   270 MBytes  2.27 Gbits/sec
[  4]   1.00-2.00   sec   271 MBytes  2.27 Gbits/sec
[  4]   2.00-3.00   sec   260 MBytes  2.18 Gbits/sec
[  4]   3.00-4.00   sec   271 MBytes  2.27 Gbits/sec
[  4]   4.00-5.00   sec   270 MBytes  2.27 Gbits/sec
[  4]   5.00-6.00   sec   262 MBytes  2.19 Gbits/sec
[  4]   6.00-7.00   sec   251 MBytes  2.10 Gbits/sec
[  4]   7.00-8.00   sec   231 MBytes  1.94 Gbits/sec
[  4]   8.00-9.00   sec   269 MBytes  2.26 Gbits/sec
[  4]   9.00-10.00  sec   255 MBytes  2.14 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  2.55 GBytes  2.19 Gbits/sec                  sender
[  4]   0.00-10.00  sec  2.55 GBytes  2.19 Gbits/sec                  receiver
```

**When PC as server:**

```
Synology -c result:
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   280 MBytes  2.34 Gbits/sec    0    226 KBytes
[  5]   1.00-2.00   sec   276 MBytes  2.32 Gbits/sec    0    226 KBytes
[  5]   2.00-3.00   sec   278 MBytes  2.33 Gbits/sec    0    226 KBytes
[  5]   3.00-4.00   sec   278 MBytes  2.34 Gbits/sec    0    226 KBytes
[  5]   4.00-5.00   sec   241 MBytes  2.02 Gbits/sec    0    226 KBytes
[  5]   5.00-6.00   sec   279 MBytes  2.34 Gbits/sec    0    226 KBytes
[  5]   6.00-7.00   sec   280 MBytes  2.35 Gbits/sec    0    226 KBytes
[  5]   7.00-8.00   sec   279 MBytes  2.34 Gbits/sec    0    226 KBytes
[  5]   8.00-9.00   sec   279 MBytes  2.35 Gbits/sec    0    226 KBytes
[  5]   9.00-10.00  sec   269 MBytes  2.25 Gbits/sec    0    226 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  2.68 GBytes  2.30 Gbits/sec    0             sender
[  5]   0.00-10.00  sec  2.67 GBytes  2.30 Gbits/sec                  receiver

PC -s result:
[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-1.00   sec   266 MBytes  2.23 Gbits/sec
[  5]   1.00-2.00   sec   276 MBytes  2.32 Gbits/sec
[  5]   2.00-3.00   sec   278 MBytes  2.33 Gbits/sec
[  5]   3.00-4.00   sec   279 MBytes  2.34 Gbits/sec
[  5]   4.00-5.00   sec   241 MBytes  2.02 Gbits/sec
[  5]   5.00-6.00   sec   279 MBytes  2.34 Gbits/sec
[  5]   6.00-7.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   7.00-8.00   sec   279 MBytes  2.34 Gbits/sec
[  5]   8.00-9.00   sec   280 MBytes  2.35 Gbits/sec
[  5]   9.00-10.00  sec   268 MBytes  2.25 Gbits/sec
[  5]  10.00-10.05  sec  13.0 MBytes  2.35 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  5]   0.00-10.05  sec  0.00 Bytes  0.00 bits/sec                  sender
[  5]   0.00-10.05  sec  2.67 GBytes  2.29 Gbits/sec                  receiver
```

The iperf3 result shows that is no problem with the speeed, just wired I dont't know why maybe is the DSM wrong.

My internal storage is 2 samsung SSD with RAID0, I check they're in good health, so it's not the disk probelm.

I also tried to plug in an external SSD on NAS, the write speed is the same as 100mb/s,

So I think it might be the CPU is too old and slow, or a DSM software bug causing this problem?

bb-qq, 2022-08-02:
Controlling USB devices consume more CPU resources than PCIe, so performance is likely to be insufficient.

In that case, the following are possible remedies:

* Increase the MTU as much as possible (7000 if 9000 is not possible) to reduce USB transactions
* Tune Samba by changing its settings directly

zhucaidan, 2022-08-03:
> Controlling USB devices consume more CPU resources than PCIe, so performance is likely to be insufficient.
>
> In that case, the following are possible remedies:
>
> * Increase the MTU as much as possible [...]

MTU setting is not helping much, I will do some search on SMB settings, Thank you!

bb-qq, 2022-08-03:
I would appreciate it if you could share any good practices when you find them.
I close this issue as it does not seem to be something I should be doing.

closed 2022-08-03 by bb-qq