bb-qq/r8152 #229: stop and start bb-qq each time after reboot/startup the ds918+ for rtl8152/8153 driver

JackyWay, 2022-06-28:
### Description of the problem

> I have to stop and start bb-qq each time after reboot/startup the ds918+ for rtl8152/8153 driver. Else, the network interface won't display the USB 2.5 adaptor. Could you assist or someone who knows how to fix it? Thanks. 

### Description of your products

> - AsRock J3455, DS918+
>     # uname -a
Linux DS-J3455 4.4.59+ #25426 SMP PREEMPT Tue May 12 04:54:55 CST 2020 x86_64 GNU/Linux synology_apollolake_918+
>     - DSM 6.2.3-25426
> - 8156b USB 2.5GbE of the ethernet dongle

### Description of your environment

> - connect directly to a PC
>     - PC: 8156b USB 2.5GbE 

> - cable category (Cat.6)

### Output of ``dmesg`` command

````
(paste output here, or attach as a file)
````
[Uploading dmesg.log…]()

### Output of ``lsusb`` command

````
~# lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.59+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-4         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)

````

### Output of ``ifconfig -a`` command

````
~# ifconfig -a
docker0   Link encap:Ethernet  HWaddr 02:42:B8: 
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80:: /64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:31885 errors:0 dropped:0 overruns:0 frame:0
          TX packets:26067 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:512092437 (488.3 MiB)  TX bytes:3623316 (3.4 MiB)

dockereb7 Link encap:Ethernet  HWaddr 7A:4C:B5 
          inet6 addr: fe80:: /64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:31885 errors:0 dropped:0 overruns:0 frame:0
          TX packets:26075 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:512538827 (488.7 MiB)  TX bytes:3623964 (3.4 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32: 
          inet addr:192.168.50.20  Bcast:192.168.50.255  Mask:255.255.255.0
          inet6 addr: 2409 /64 Scope:Global
          inet6 addr: fe80 /64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:787941 errors:0 dropped:0 overruns:0 frame:0
          TX packets:641746 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:224137402 (213.7 MiB)  TX bytes:675207515 (643.9 MiB)
          Interrupt:120 base 0xc000

eth1      Link encap:Ethernet  HWaddr 00: 
          inet addr:10.0.0.2  Bcast:10.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80 /64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:4877 errors:0 dropped:0 overruns:0 frame:0
          TX packets:346 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1216878 (1.1 MiB)  TX bytes:84094 (82.1 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:11074 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11074 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:1678143 (1.6 MiB)  TX bytes:1678143 (1.6 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

shu0802, 2022-07-21:
我是dsm7.1 918+，得重新插拨网线才变成2500m/s。经常由2500m/s变成1000m/s或者100m/s；MTU值设置超过1500后web就不能访问。

bb-qq, 2022-07-31:
@JackyWay, what driver version are you using? Past drivers have the issue regarding startup.

JackyWay, 2022-09-18:
> @JackyWay, what driver version are you using? Past drivers have the issue regarding startup.

It's 2.14.0-2.

I have appied the workaround with the task schedule when the Synology boots up to execute below commands.

synoservice --disable pkgctl-r8152
synoservice --enable pkgctl-r8152

JackyWay, 2022-09-18:
> 我是dsm7.1 918+，得重新插拨网线才变成2500m/s。经常由2500m/s变成1000m/s或者100m/s；MTU值设置超过1500后web就不能访问。 )

检查网线是不是超五类或者以上。

rsporsche, 2022-09-30:
I have the same issue with a Sabrent adapter and version 2.15.0-1 on my DS418play.

I'll try the scheduled task workaround since it looks like this isn't being developed for DSM6 anymore. I don't want to upgrade to DSM7 until I can move my Home Assistant stuff which requires additional USB devices off the NAS.

bb-qq, 2022-12-04:
Since the contents of the script have changed significantly since this issue was created, I would like to close this issue.

If you experience problems using the latest release, please create another issue or reopen this issue.
https://github.com/bb-qq/r8152/releases/tag/2.16.3-3

closed 2022-12-04 by bb-qq