bb-qq/r8152 #231: TrendNet TUC-ET2G not showing up in interfaces

Auberon3, 2022-07-15:
### Description of the problem
Hi, driver installed, it's green in packages, followed instructions it seems ok but the adapter not showing up in network interfaces
I bought  TUC-ET2G as mentioned compatible it is plugged with USB-C --> USB-A adapter provided in the kit

### Description of your products
> - DS1515 Alpine version not DS1515+
> - Linux XXXXXX 3.10.108 #42661 SMP Mon Jun 27 15:03:29 CST 2022 armv7l GNU/Linux synology_alpine_ds1515
> - DSM 7.1-42661 update 3
> - TRENDNET TUC-ET2G kit with USB-A adapter

### Description of your environment

> - connected directly to a PC or using a Ethernet TRENDNET TUC-ET2G adapter and crossed cable
> - TRENDNET TUC-ET2G
> - N/A
> - cable category (Cat.6 shielded) tested and working, the nas works when connected through onboard nic with the same config
     just the trendnet adapter seems not detected in DSM interfaces section

### Output of ``dmesg`` command
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 3.10.108 (root@build2) (gcc version 8.5.0 (GCC) ) #42661 SMP Mon Jun 27 15:03:29 CST 2022
[    0.000000] CPU: ARMv7 Processor [412fc0f4] revision 4 (ARMv7), cr=30c73c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] Machine: AnnapurnaLabs Alpine (Device Tree), model: Annapurna Labs Alpine Dev Board
[    0.000000] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] On node 0 totalpages: 65536
[    0.000000] free_area_init_node: node 0, pgdat 80649fc0, node_mem_map 806d8000
[    0.000000]   Normal zone: 60 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 48640 pages, LIFO batch:3
[    0.000000]   HighMem zone: 21 pages used for memmap
[    0.000000]   HighMem zone: 16896 pages, LIFO batch:3
[    0.000000] PERCPU: Embedded 2 pages/cpu @80978000 s40384 r8192 d16960 u65536
[    0.000000] pcpu-alloc: s40384 r8192 d16960 u65536 alloc=2*32768
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 65476
[    0.000000] Kernel command line: console=ttyS0,115200 ip=off pci=pcie_bus_perf initrd=0x07500000 root=/dev/md0 rw syno_hw_version=DS1515 ihd_num=0 netif_num=4 vmalloc=512M syno_hdd_act_led=10,11,22,23,24
[    0.000000] Synology Hardware Version: DS1515-j
[    0.000000] Internal HD num: 0
[    0.000000] Internal netif num: 4
[    0.000000] SYNO GPIO hdd active led pin: 10 11 22 23 24
[    0.000000] PID hash table entries: 4096 (order: -1, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 262144 (order: 5, 1048576 bytes)
[    0.000000] Inode-cache hash table entries: 131072 (order: 4, 524288 bytes)
[    0.000000] allocated 524288 bytes of page_cgroup
[    0.000000] please try 'cgroup_disable=memory' option if you don't want memory cgroups
[    0.000000] Memory: 2048MB = 2048MB total
[    0.000000] Memory: 2081088k/2081088k available, 16064k reserved, 540672K highmem
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff8000   (  32 kB)
                   fixmap  : 0xff400000 - 0xff800000   (4096 kB)
                   vmalloc : 0xdf800000 - 0xff000000   ( 504 MB)
                   lowmem  : 0x80000000 - 0xdf000000   (1520 MB)
                   pkmap   : 0x7fe00000 - 0x80000000   (   2 MB)
                   modules : 0x7ec00000 - 0x7fe00000   (  18 MB)
                     .text : 0x80008000 - 0x805a7680   (5758 kB)
                     .init : 0x805a8000 - 0x805f9dc0   ( 328 kB)
                     .data : 0x80600000 - 0x8064ddd8   ( 312 kB)
                      .bss : 0x8064ddd8 - 0x806d7d38   ( 552 kB)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] sched_clock: 32 bits at 375MHz, resolution 2ns, wraps every 11453ms
[    0.000162] Calibrating delay loop... 2793.47 BogoMIPS (lpj=13967360)
[    0.060025] pid_max: default: 32768 minimum: 301
[    0.060081] Security Framework initialized
[    0.060095] AppArmor: AppArmor initialized
[    0.060118] Mount-cache hash table entries: 4096
[... 863 lines cut ...]
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth3      Link encap:Ethernet  HWaddr xxxxxxxxxxxx
          inet addr:169.254.40.88  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:11960 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11960 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1801404 (1.7 MiB)  TX bytes:1801404 (1.7 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

bb-qq, 2022-07-31:
Added the support for the new variant of TUC-ET2G(=v2.0R) in the latest release.
https://github.com/bb-qq/r8152/releases/tag/2.15.0-9

Auberon3, 2022-07-31:
thanks it works both on my DS1515 and DS920+. on DS920+ I get stable 280mb/s on big files. but on DS1515 it is less stable speed fluctuates, on little files speed is really poor like 40mb/s and on big file it reach 175mb/s

bb-qq, 2022-08-01:
One user reports that the driver with DS1515+ is unstable in his environment. #71
This model seems to have some problems with xHCI.

i-am-at0m, 2023-03-09:
> Added the support for the new variant of TUC-ET2G(=v2.0R) in the latest release. https://github.com/bb-qq/r8152/releases/tag/2.15.0-9

Is there any plan to build this into a DSM6-compatible release?

bb-qq, 2023-03-19:
I added DSM6 builds for the latest release.
https://github.com/bb-qq/r8152/releases/tag/2.16.3-4

bb-qq, 2023-07-16:
I would like to close this issue as there does not seem to be any work to continue on this issue.

closed 2023-07-16 by bb-qq