bb-qq/r8152 #232: Slow Speeds using Cable Creations 2.5G ethernet to USB adapter - DS920+

abracadabra1111, 2022-07-22:
Recently purchased and installed a cable creations 2.5G ethernet to USB adapter on my DS920+ using the front USB port. Installed the gemini lake drivers listed on the [bb-qq release page](https://github.com/bb-qq/r8152/releases) and everything went smoothly. Synology network page and TP Link switch all show 2.5Gbps connection. But when I run speed tests using iperf or copy files to the NAS, I'm seeing ~1Gbps speeds. Any tips for troubleshooting?

[image]
[image]

Sjekke, 2022-07-22:
What type is your switch? (See problem here https://github.com/bb-qq/r8152/issues/180#issuecomment-1011055172)

abracadabra1111, 2022-07-22:
TL-SG3210XHP-M2

abracadabra1111, 2022-07-22:
> What type is your switch? (See problem with TP Link TL-SG108-M2 here [#180 (comment)](https://github.com/bb-qq/r8152/issues/180#issuecomment-1011055172))

TL-SG3210XHP-M2

abracadabra1111, 2022-07-22:
Device is working properly. The two devices 2.5Gbps devices were on different VLANs and the router is limited to 1Gbps ports. Placing both devices on same VLAN resulted in proper speed.
[image]

closed 2022-07-22 by abracadabra1111