bb-qq/r8152 #233: To enable the hot plug feature, After successful installation of the driver, execute the following command

s128300901, 2022-07-30:
:~$ sudo bash /var/packages/aqc111/scripts/install-udev-rules
bash: /var/packages/aqc111/scripts/install-udev-rules: No such file or directory

s128300901, 2022-07-30:
920+

bb-qq, 2022-07-30:
I had written the wrong command. The correct one is this.
`sudo bash /var/packages/r1852/scripts/install-udev-rules`

s128300901, 2022-07-30:
sergey@synology:~$ sudo bash /var/packages/r1852/scripts/install-udev-rules
Password:
bash: /var/packages/r1852/scripts/install-udev-rules: No such file or directory

bb-qq, 2022-07-31:
Could you try the new release?
https://github.com/bb-qq/r8152/releases/tag/2.15.0-9

And my previous comment was wrong. (Already fixed)
wrong: `sudo bash /var/packages/r1852/scripts/install-udev-rules`
correct: `sudo bash /var/packages/r8152/scripts/install-udev-rules`

s128300901, 2022-07-31:
sergey@synology:~$ sudo bash /var/packages/r8152/scripts/install-udev-rules
Password:
Updating Hardware Database Index...
UDEV rules have been installed to /usr/lib/udev/rules.d
lrwxrwxrwx 1 root root 50 Jul 31 08:09 /usr/lib/udev/rules.d/51-usb-r8152-net.rules -> /var/packages/r8152/scripts/51-usb-r8152-net.rules

bb-qq, 2022-07-31:
The script was successfully finished. If the Hot Plug does not seem to work, please stick the output of `dmesg`.

s128300901, 2022-07-31:
[image]
the adapter is disconnected from the nas, but it shows as if in the system

s128300901, 2022-07-31:
[image]

s128300901, 2022-07-31:
[image]

s128300901, 2022-08-10:
[2.15.0-10 DSM7.x] 7.1.1-42951 does not work

Sjekke, 2022-08-10:
I installed the latest version of the driver and it works perfect BUT I always have to stop and restart my Synology DHCP server to get it working again. Is there some kind of command that we can put in the script to stop and start the DHCP server after the driver has activated the USB adapter?

bb-qq, 2023-07-16:
Since the Hot Plug seems to be working in the versions released since then, I would like to close this issue if there are no new reports of problems.

As for DHCP, it has been improved in later releases as well. So please try it and if you find any problems, please create a new issue and write the details.

bb-qq, 2024-06-24:
I Since there have been no additional comments for a long time, this issue is closed. 
Feel free to re-open it if you have any questions.

closed 2024-06-24 by bb-qq