bb-qq/r8152 #234: Installation of r8152-apollolake-2.15.0-9.spk Fails on DS718+

TheS1R, 2022-07-31:
Installation of r8152-apollolake-2.15.0-9.spk (and earlier r8152-apollolake-2.15.0-8.spk as well) on DS718+ (DSM 7.1-42661 Update 3) fails. Reverted to r8152-apollolake-2.15.0-7.spk.

guda0700, 2022-08-02:
me too,my  NAS is 918+

TheS1R, 2022-08-02:
No, this fails in the same way. Uninstalled this driver, and reinstalled r8152-apollolake-2.15.0-7.spk to resolve.

bb-qq, 2022-08-03:
@TheS1R , could you paste the output of these commands?
```
sudo /var/packages/aqc111/scripts/start-stop-status stop
sudo /var/packages/aqc111/scripts/start-stop-status start

TheS1R, 2022-08-03:
Do you want me to do this before or after attempting to install new version?

bb-qq, 2022-08-03:
I would like you to run the command **after** installing the new version.

TheS1R, 2022-08-03:
$ sudo /var/packages/r8152/scripts/start-stop-status stop
Password: 
++ dirname /var/packages/r8152/scripts/start-stop-status
+ script_root=/var/packages/r8152/scripts
++ realpath -m /var/packages/r8152/scripts/..
+ package_root=/var/packages/r8152
++ basename /var/packages/r8152
+ driver_name=r8152
+ driver_root=/var/packages/r8152/target/r8152
+ case $1 in
+ set_interfaces down
+ action=down
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='bond0
bonding_masters
eth0
eth1
eth2
lo
sit0
tun0'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ bond0 = \l\o ]]
+ [[ bond0 == \d\o\c\k\e\r* ]]
+ [[ bond0 == \o\v\s* ]]
+ [[ bond0 == \s\y\n\o* ]]
+ [[ bond0 == \s\i\t* ]]
+ [[ bond0 == \t\a\p* ]]
+ [[ bond0 == \t\u\n* ]]
+ [[ -n '' ]]
++ ls -ld /sys/class/net/bond0/device/driver
ls: cannot access '/sys/class/net/bond0/device/driver': No such file or directory
+ driver_location=

$ sudo /var/packages/r8152/scripts/start-stop-status start
++ dirname /var/packages/r8152/scripts/start-stop-status
+ script_root=/var/packages/r8152/scripts
++ realpath -m /var/packages/r8152/scripts/..
+ package_root=/var/packages/r8152
++ basename /var/packages/r8152
+ driver_name=r8152
+ driver_root=/var/packages/r8152/target/r8152
+ case $1 in
+ '[' -e /var/packages/r8152/scripts/apply-memory-setting ']'
++ count_usb_interfaces
++ ls -l /sys/class/net/
++ grep /usb
++ wc -l
+ initial_count=1
+ '[' -e /var/packages/r8152/target/r8152/mii.ko ']'
+ '[' -e /var/packages/r8152/target/r8152/usbnet.ko ']'
+ /sbin/insmod /var/packages/r8152/target/r8152/r8152.ko
insmod: ERROR: could not insert module /var/packages/r8152/target/r8152/r8152.ko: File exists

[image]

bb-qq, 2022-08-03:
Could try this package?
https://github.com/bb-qq/r8152/releases/download/2.15.0-9/r8152-apollolake-2.15.0-10pre2.spk

If it does not work for you, please run previous debug commands again.

TheS1R, 2022-08-03:
That appears to work! Do you need any additional debug info?

bb-qq, 2022-08-03:
Thank you. I don't need additional logs anymore. I will create a new release later.

TheS1R, 2022-08-08:
Resolved with 2.15.0-10.

closed 2022-08-08 by TheS1R