bb-qq/r8152 #235: Fail to run the package service

Hopemannz, 2022-08-07:
### Description of the problem
After installation, when I run it,  error message as "Fail to run the package service". 
In package Center, RTL8152/RTL8153 driver show stooped. Tried Repair but still the same.
USB dongle plug into back and front port on DS420+,  no difference.

### Description of your products

> - the product name of your NAS model
DS420+

>     - and output of ``uname -a`` command
Linux myhome 4.4.180+ #42661 SMP Mon Jun 27 15:07:20 CST 2022 x86_64 GNU/Linux synology_geminilake_420+

>     - DSM version x.x
DSM 7.1-42661 Update 3

> - the product name of the ethernet dongle
No name from Aliexpress , info center show as below:
USB 10/100/1G/2.5G LAN - Realtek Semiconductor Corp.
VID : 0bda PID:8156 REV:31.04 

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name

>     - Ethernet switch: model name
Connect to Vodafone ( HUAWEI OEM) Boardband Fiber router ( with 5 port)

> - cable category (Cat.5e or Cat.6 or Cat.7)
Cat 6

### Output of ``dmesg`` command
````
[dmesg.txt](https://github.com/bb-qq/r8152/files/9276005/dmesg.txt)
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F000C4880BE4082)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3104 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 4013000001)
````

### Output of ``ifconfig -a`` command

````
bond0     Link encap:Ethernet  HWaddr 00:11:32:E2:82:42
          inet addr:192.168.1.4  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2407:7000:a14d:2700:211:32ff:fee2:8242/64 Scope:Global
          inet6 addr: fe80::211:32ff:fee2:8242/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:8872 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7982 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3616238 (3.4 MiB)  TX bytes:8997600 (8.5 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:E2:82:42
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:8872 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7982 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3616238 (3.4 MiB)  TX bytes:8997600 (8.5 MiB)
          Interrupt:96 base 0xc000

eth1      Link encap:Ethernet  HWaddr 00:11:32:E2:82:43
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:97 base 0xe000

eth2      Link encap:Ethernet  HWaddr 1C:BF:CE:FB:E5:8B
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:3513 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3513 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:545230 (532.4 KiB)  TX bytes:545230 (532.4 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

````

TheS1R, 2022-08-07:
This may be similar to Issue #234.

Hopemannz, 2022-08-08:
> This may be similar to Issue #234.

Thanks. I tried r8152-geminilake-2.15.0-7, it works!  
But 2.15.0-8 also fail.
Not sure OP (bb-qq) still need the debug (status stop/start) info or not?

bb-qq, 2022-08-08:
If the new driver does not work for you, I would like to debug it.
https://github.com/bb-qq/r8152/releases/tag/2.15.0-10

Hopemannz, 2022-08-08:
> If the new driver does not work for you, I would like to debug it. https://github.com/bb-qq/r8152/releases/tag/2.15.0-10

Thanks, the  -10 release works fine.  Really appreciate your hard work!

bb-qq, 2022-08-08:
Thanks for confirming.

closed 2022-08-08 by bb-qq