bb-qq/r8152 #236: Asking for a Club 3D CAC-1420 USB 3.2 Gen1 Type A to RJ45 2.5Gbps Ethernet Adapter driver for DSM 7.1.1 RC on Synology DS418play

YVES59, 2022-08-13:
Hello everyone !!!

I have been using the GitHub drivers for my USB Realtek 2.5GbE adapter since the beginning without any problem, many thanks for this great work of the developers !!!

Unfortunately, after the upgrade of my DS418play to DSM  7.1.1 - 42951 RC from DSM 7.1 - 42661 Update 4, the driver is no longer functionnal (even with a full reinstallation).

May I ask the opinion of the developers ?

Many thanks for your help, as this 2.5Gbs wired ethernet connection of my DS418play is very important for me. !!!!!!

Best Regards 

````

mrjohnpoz, 2022-08-14:
Yeah seeing the same thing - see my comment on the other thread about dsm 7.1

I just had to shutdown my nas because of a planned power outage from the elec company, and was scheduled to be longer than my ups could run.

When I brought it back up, yeah had to cycle it to dhcp, let it not get an IP.. (169.254 address) then switch it back to static and it came back up.

TheS1R, 2022-08-15:
I have two DS718+ (apollolake) also with DSM 7.1.1-42951 RC. I experienced the same failure with static IP addresses. My workaround was to manually assign static IP addresses in my router (Asus GT-AX6000 with Merlin firmware) and have the NASs obtain the static addresses via DHCP.

mrjohnpoz, 2022-08-16:
That is an option if you have dhcp server running on the network your connected to - but in my setup its just direct connection between my pc and the nas for 2.5ge transfer of data between the two.

Guess I could run dhcp server on my PC for the nas connection - but use to work before.  Its not the end of the world since only time I really reboot nas is update or some extended power outage which are rare.

It never use to be an issue, have been running this for quite some time with multiple updates to the DSM software and the bb-qq driver version.

TheS1R, 2022-08-16:
Used to work in my case as well.

mrjohnpoz, 2022-08-16:
I think something changed in 7.1.1, hope that bb-qq figures it out.  

Any info I can provide just ask.. But doesn't look like he has seen the thread yet.

bb-qq, 2022-08-20:
Since [2.15.0-8](https://github.com/bb-qq/r8152/releases/tag/2.15.0-8) I have changed the storage location for persisting network setting. I guess this is the cause of this issue. 

I expected that when the system goes to shut down, the current network setting would be stored new location. But it seems to be failed.

If the same problem occurs every time you reboot, try to Stop&Start the driver from the Package Center. This will ensure that your settings are saved.
If the problem occurs only once and the setting seems to persist now, the settings have been saved correctly and you will not have the same problem in the future.

hhessel, 2022-08-20:
@bb-qq I'm not facing the same problem as above that my static ip settings are for the Club3D interface are visible but seemly not functional after a reboot. DS Assistant reports the following after the reboot:

[image]

So it shows that there is manual IP address assigned but the IP address is not reachable even though if I use another interface to connect it shows in DSM. I have to switch the 2,5GB interface to another static IP and then back to the original IP. 

I tried your proposed solution of stopping and starting driver (over another interface) and even reinstalling it. Unfortunately always with the same problem afterwards. Happy to support or help in case necessary. Thanks

bb-qq, 2022-08-20:
I will check the behavior once my NAS receives the 7.1.1 rollout release.

hhessel, 2022-08-20:
@bb-qq Thanks, probably the better approach than manually updating to 7.1.1. like it ;)

mrjohnpoz, 2022-08-20:
> Club3D interface are visible but seemly not functional after a reboot

Mine were visible in the dsm setting, but if you actually looked on the interfaces with ifconfig from ssh to nas then they were not there.  I will check next time I reboot the nas to see what happens I have stopped and started the driver.

bb-qq, 2022-10-09:
(Link to #246)

Could you try this test version?
https://github.com/bb-qq/r8152/releases/download/2.16.3-1/r8152-apollolake-2.16.3-2test.spk

blasiusx, 2022-10-09:
Seee my comment at

https://github.com/bb-qq/r8152/issues/246#issuecomment-1272484723

It works fine for me. Thank you very much!!!

bb-qq, 2022-10-11:
Thank you. I close this issue since officially versioned drivers have been released.
https://github.com/bb-qq/r8152/releases/tag/2.16.3-2

closed 2022-10-11 by bb-qq

mrjohnpoz, 2022-10-11:
@bb-qq is there any reason to change to this version vs the test one?

bb-qq, 2022-10-11:
The functionality of the driver is exactly the same to the test version, with only some improvements to the logging.