bb-qq/r8152 #237: Compability report for 2.15.0-10 DSM7.x-v1000 (DS1821+)

tonigrzb, 2022-08-16:
### Description of the problem

Compability report for 2.15.0-10 DSM7.x-v1000

### Description of your products

DS1821+ running on DSM 7.1-42661 Update 4
Linux synology 4.4.180+ #42661 SMP Mon Jun 27 15:11:30 CST 2022 x86_64 GNU/Linux synology_v1000_1821+

using a [ULT-WIIQ USB 2,5G Ethernet Adapter (USB-A)](https://www.amazon.de/gp/product/B09GBFZ9DZ/) (RTL8156B)

### Description of your environment

connected via a 10GbE Switch (multigig-capable)

### Output of ``lsusb`` command
at Front USB 

|__usb4          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:09:00.4) hub
  |__4-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001300E04C680293)

### Remarks

Needed a Reboot after Installation, because it had issues showing up in DSM and retrieving an IP
(Not shure if helpful, but Adapter has a static IP in the router.)

Currently Working 

Speed (using smb): 

NAS -> PC ~295 MB/s
PC -> NAS ~230-240 MB/s

TheS1R, 2022-08-16:
Take a look at Issue #236 — it appears that you describe the same issue.

FlorianHeigl, 2022-12-08:
I have the same adapter, it didn't attach a driver (DS920+, Geminilake).
Otherwise, as described - first install failed, second worked. (both using `synopkg install`)
I will try the reboot, leaving the info here for analysis

Package name: `r8152-geminilake-2.16.3-3.spk'

`dmesg` of plugging in
```
[ 2061.429409] xhci_hcd 0000:00:15.0: device is plugged out, empty URBs
[ 2064.309525] usb 2-1: USB disconnect, device number 6
[ 2064.542597] usb 2-1: new SuperSpeed USB device number 7 using xhci_hcd
```

The driver itself seems to have loaded

`dmesg` of boot
```
[  493.476604] r8168 Gigabit Ethernet driver 8.045.12-NAPI loaded
[  493.489355]  Reinit 1 8001 
[  493.492497]  Reinit 2 1 
[  493.495357] systemd[1]: Started Journal Service.
[  493.505312]  Reinit 1 8001 
[  493.508445]  Reinit 2 1 
[  493.511912] r8168: This product is covered by one or more of the following patents: US6,570,884, US6,115,776, and US6,327,625.
[  493.524662] r8168  Copyright (C) 2017  Realtek NIC software team <nicfae@realtek.com> 
                This program comes with ABSOLUTELY NO WARRANTY; for details, please see <http://www.gnu.org/licenses/>. 
                This is free software, and you are welcome to redistribute it under certain conditions; see <http://www.gnu.org/licenses/>. 
[  493.559174] r8168 Gigabit Ethernet driver 8.045.12-NAPI loaded
```

`lsusb` looks like this:

```
root@nastmp01:/tmp# lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F000CCB9713F231)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 2IFs (Realtek USB 10/100/1G/2.5G LAN 001300E04C680239)
```