bb-qq/r8152 #238: DS412+ (Cable Creation 8156) [Unable to achieve 2.5GBe transfer speeds]

SpinCykle, 2022-08-17:
### Description of the problem

Unable to achieve 2.5GBe speeds.  2.5t base recognized by switch and clients, however, I can not achieve the rated transfer speeds. 

### Description of your products

> DS412+
> 
> DSM 6.2.4-25556 Update 6
> CableCreation 2.5G Ethernet to USB Adapter

### Description of your environment

> Ethernet switch
> 
> QNAP QSW-1105-5T
> Cat.5e

### Output of ``dmesg`` command

````
(paste output here, or attach as a file)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````
I appear to be stuck at gigabit speeds. Why is this happening? I have tried using SMB v3 and MTU 9000 but it does not help.  I am using driver version 2.15.0 release. Any help would be greatly appreciated. 

[image]

TheS1R, 2022-08-19:
I believe that you need at least Cat 6 cables for speeds above 1 Gbps.

Sjekke, 2022-08-19:
If you have a good cable, you can run 10gbit on cat5e over less then 20m

TheS1R, 2022-08-19:
Also, 2.15.0-x is **NOT** recommended for DSM 6.x.

SpinCykle, 2022-08-19:
> I believe that you need at least Cat 6 cables for speeds above 1 Gbps.

I've run iperf3 from other clients PC->PC on the network with the same calbe (5e) and there are no limitations. See the results below. I don't think it's my cable. 
[image]

SpinCykle, 2022-08-19:
> Also, 2.15.0-x is **NOT** recommended for DSM 6.x.

What version do you reccomend? The description for 2.15.0 says it's 6.2x compatible. The other versions 15.1, 15.2 ect are not.

TheS1R, 2022-08-19:
Interesting as the oldest 2.15.0-2 is the oldest release in the 2.15.0 series that I can see in github. 2.15.0-2 "_Added support for DSM7_",  and it states that "_Users of DSM 6.2 should use past releases_". Have you tried 2.14.0-3?

bb-qq, 2022-08-20:
Thank you all, but based on [reports from other users](https://github.com/bb-qq/r8152/wiki/Compatibility), I am wondering if the Cedarview platform is having some hardware issues.

bb-qq, 2022-08-20:
What will the throughput be if I set the MTU to about 4000? Some hardware may have different optimal values, so throughput may improve a little.

transformcube, 2022-12-22:
> ### Description of the problem
> 
> Unable to achieve 2.5GBe speeds. 2.5t base recognized by switch and clients, however, I can not achieve the rated transfer speeds.
> ### Description of your products
> 
> > DS412+ w/ 4GB RAM
> > DSM 6.2.4-25556 Update 6
> > CableCreation 2.5G Ethernet to USB Adapter
> 
> ### Description of your environment
> 
> > Ethernet switch
> > QNAP QSW-1105-5T
> > Cat.5e
> 
> ### Output of `dmesg` command
> 
> ```
> (paste output here, or attach as a file)
> ```
> 
> ### Output of `lsusb` command
> 
> ```
> (paste output here)
> ```
> 
> ### Output of `ifconfig -a` command
> 
> ```
> (paste output here)
> ```
> 
> I appear to be stuck at gigabit speeds. Why is this happening? I have tried using SMB v3 and MTU 9000 but it does not help. I am using driver version 2.15.0-1 release. Any help would be greatly appreciated.
> 
> [image]

It is hardware limitation (processor) of DS412+, their max throughput about 150MB/s.

[https://www.smallnetbuilder.com/nas/nas-reviews/synology-ds412-diskstation-reviewed/](url)