bb-qq/r8152 #239: 1515+ asustor as-u2.5g2 RX status -71

tjjh89017, 2022-08-23:
### Description of the problem

1515+ with as-u2.5g2 will have problem.
after a night.
dmesg will show a lot of ` r8152 3-2:1.0 eth4: get_registers -71` or `RX status -71`
and reduce speed to 100Mbps only (in DSM7 netwokr )

Thanks a lot

### Description of your products

`Linux nas 3.10.108 #42661 SMP Mon Jun 27 15:00:55 CST 2022 x86_64 GNU/Linux synology_avoton_1515+`
`DSM 7.1-42661 Update 4 `
`ASUSTOR AS-U2.5G2`
`driver 2.16.3-1 DSM7.x`

### Description of your environment

> - connect directly to a PC or using a Ethernet switch
>     - PC: NIC model name or chipset name
>     - Ethernet switch: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)

Connect to CRS309 with Mikrotik S+RJ10 gbic.
PC is windows with Intel X520 + Finisar 10G-LR fiber

### Output of ``dmesg`` command
[dmesg.txt](https://github.com/bb-qq/r8152/files/9400550/dmesg.txt)

````
[  +0.017431] r8152 3-2:1.0 eth4: chip rev 14
[  +0.004684] r8152 3-2:1.0 eth4: rx_buf_sz will be overridden to 16384
[  +0.780810] IPv6: ADDRCONF(NETDEV_UP): eth4: link is not ready
[  +4.895321] IPv6: ADDRCONF(NETDEV_CHANGE): eth4: link becomes ready
[  +0.008719] r8152 3-2:1.0 eth4: carrier on
[Aug23 15:38] r8152 3-2:1.0 eth4: get_registers -71
[ +11.334771] r8152 3-2:1.0 eth4: get_registers -71
````
Other part need another night to gen.

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:16.0) hub
  |__1-1         8087:07db:0002 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.1     f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 6500584050CA6086)
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
|__usb3          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:04:00.0) hub
  |__3-2         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 0013000001)````

### Output of ``ifconfig -a`` command

````
eth0      Link encap:Ethernet  HWaddr 00:11:32:XX:XX:XX
          inet addr:XX.XX.0.16  Bcast:XX.XX.0.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:51682904 errors:0 dropped:0 overruns:0 frame:0
          TX packets:39970899 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:55219939722 (51.4 GiB)  TX bytes:55754336695 (51.9 GiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:XX:XX:XX
          inet addr:169.254.196.175  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 00:11:32:XX:XX:XX
          inet addr:169.254.74.101  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth3      Link encap:Ethernet  HWaddr 00:11:32:XX:XX:XX
          inet addr:169.254.122.180  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth4      Link encap:Ethernet  HWaddr 00:E0:4C:XX:XX:XX
          inet addr:XX.XX.0.129  Bcast:XX.XX.0.255  Mask:255.255.255.0
          inet6 addr: fe80::2e0:4cff:fe68:df4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2475549 errors:0 dropped:0 overruns:0 frame:0
          TX packets:266163 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:3709110090 (3.4 GiB)  TX bytes:27996916 (26.6 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:14 errors:0 dropped:0 overruns:0 frame:0
          TX packets:14 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1276 (1.2 KiB)  TX bytes:1276 (1.2 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

war59312, 2022-08-26:
I'm seeing them too on my DS1815+ over at https://github.com/bb-qq/r8152/issues/43 but doesn't seem to be causing a big issue.

Good TB of data sent/received at this point.

bb-qq, 2022-12-04:
The 2.16.3-x releases addressed this kind of issue.
https://github.com/bb-qq/r8152/releases/tag/2.16.3-1

Could you try the latest version?
https://github.com/bb-qq/r8152/releases/tag/2.16.3-3

tjjh89017, 2022-12-04:
> The 2.16.3-x releases addressed this kind of issue. https://github.com/bb-qq/r8152/releases/tag/2.16.3-1
> 
> Could you try the latest version? https://github.com/bb-qq/r8152/releases/tag/2.16.3-3

Ok, I will try it
Thanks a lot

war59312, 2022-12-08:
Nice!

Any change could be backported for DSM 6.2.,4 ?

Happy to pay you a fair $$ to get full DSM 6 support. :)

LogicDX342, 2022-12-12:
I encountered the same error message on my DS718+ (DSM7.0.1) after updating to the latest version.

dmesg

```
[39571.235988] Module [r8152] is removed. 
[39571.240315] usbcore: deregistering interface driver r8152
[39571.842200] iSCSI:iscsi_target.c:620:iscsit_del_np CORE[0] - Removed Network Portal: 192.168.31.132:3260 on iSCSI/TCP
[39571.883819] iSCSI:iscsi_target.c:620:iscsit_del_np CORE[0] - Removed Network Portal: [fe80::2e0:4cff:fe68:ae0]:3260 on iSCSI/TCP
[39575.212110] audit: type=1400 audit(1670868744.052:830): apparmor="DENIED" operation="capable" profile="/volume*/@appstore/Virtualization/bin/synoccctool" pid=21856 comm="synoccctool" capability=16  capname="sys_module"
[39576.752266] audit: type=1325 audit(1670868745.592:831): table=filter family=2 entries=47
[39576.769312] audit: type=1325 audit(1670868745.609:832): table=filter family=2 entries=47
[39576.830189] audit: type=1325 audit(1670868745.670:833): table=filter family=2 entries=47
[39576.863797] audit: type=1325 audit(1670868745.704:834): table=filter family=10 entries=43
[39576.910396] audit: type=1325 audit(1670868745.750:835): table=filter family=10 entries=43
[39576.937020] audit: type=1325 audit(1670868745.777:836): table=filter family=10 entries=43
[39576.992584] audit: type=1325 audit(1670868745.832:837): table=filter family=2 entries=51
[39577.026215] audit: type=1325 audit(1670868745.866:838): table=filter family=10 entries=47
[39577.068478] audit: type=1325 audit(1670868745.908:839): table=filter family=2 entries=44
[39577.123474] audit: type=1325 audit(1670868745.963:840): table=filter family=10 entries=31
[39577.895473] etxhci_hcd-170202 0000:02:00.0: WARN Event TRB for slot 2 ep 2 with no TDs queued?
[39577.905646] etxhci_hcd-170202 0000:02:00.0: WARN Event TRB for slot 2 ep 3 with no TDs queued?
[39577.915300] etxhci_hcd-170202 0000:02:00.0: WARN Event TRB for slot 2 ep 6 with no TDs queued?
[39579.027931] usb 4-1: reset SuperSpeed USB device number 3 using etxhci_hcd-170202
[39579.180510] r8152 4-1:1.0 eth2: v2.16.3 (2022/07/06)
[39579.186082] r8152 4-1:1.0 eth2: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[39579.203476] r8152 4-1:1.0 eth2: chip rev 14
[39579.208147] r8152 4-1:1.0 eth2: rx_buf_sz will be overridden to 16384
[39579.215429] usbcore: registered new interface driver r8152
[39579.331731] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[39579.396770] device ovs_eth2 entered promiscuous mode
[39579.426292] device eth2 entered promiscuous mode
[39579.456982] IPv6: ADDRCONF(NETDEV_UP): ovs_eth2: link is not ready
[39583.298948] IPv6: ADDRCONF(NETDEV_CHANGE): ovs_eth2: link becomes ready
[39583.309054] r8152 4-1:1.0 eth2: Promiscuous mode enabled
[39583.316694] r8152 4-1:1.0 eth2: carrier on
[39586.086388] iSCSI:iscsi_target.c:528:iscsit_add_np CORE[0] - Added Network Portal: 192.168.31.132:3260 on iSCSI/TCP
[39586.101303] iSCSI:iscsi_target.c:528:iscsit_add_np CORE[0] - Added Network Portal: [fe80::2e0:4cff:fe68:ae0]:3260 on iSCSI/TCP
[39588.670057] audit_printk_skb: 162 callbacks suppressed
[39588.675810] audit: type=1325 audit(1670868757.510:895): table=filter family=2 entries=47
[39588.692682] audit: type=1325 audit(1670868757.533:896): table=filter family=2 entries=47
[39588.711375] audit: type=1325 audit(1670868757.552:897): table=filter family=2 entries=47
[39588.729878] audit: type=1325 audit(1670868757.570:898): table=filter family=10 entries=43
[39588.746870] audit: type=1325 audit(1670868757.587:899): table=filter family=10 entries=43
[39588.764441] audit: type=1325 audit(1670868757.605:900): table=filter family=10 entries=43
[39588.781699] audit: type=1325 audit(1670868757.622:901): table=filter family=2 entries=51
[39588.809906] audit: type=1325 audit(1670868757.650:902): table=filter family=10 entries=47
[39588.839991] audit: type=1325 audit(1670868757.680:903): table=filter family=2 entries=44
[39588.867627] audit: type=1325 audit(1670868757.708:904): table=filter family=10 entries=31
[39661.930101] sd 5:0:1:0: [isda] Synchronizing SCSI cache
[39662.127334] device tap021132267b9d left promiscuous mode
[39662.133308] audit_printk_skb: 156 callbacks suppressed
[39662.139054] audit: type=1700 audit(1670868830.977:957): dev=tap021132267b9d prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
[39662.246928] device tap02113228f556 left promiscuous mode
[39662.252932] audit: type=1700 audit(1670868831.097:958): dev=tap02113228f556 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
[39666.398078] IPv6: ADDRCONF(NETDEV_UP): tap021132267b9d: link is not ready
[39666.417660] device tap021132267b9d entered promiscuous mode
[39666.423919] audit: type=1700 audit(1670868835.268:959): dev=tap021132267b9d prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[39666.469311] IPv6: ADDRCONF(NETDEV_UP): tap02113228f556: link is not ready
[39666.488935] device tap02113228f556 entered promiscuous mode
[39666.495186] audit: type=1700 audit(1670868835.340:960): dev=tap02113228f556 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[39666.543478] IPv6: ADDRCONF(NETDEV_UP): tap0211322eab5b: link is not ready
[39666.567657] device tap0211322eab5b entered promiscuous mode
[39666.573925] audit: type=1700 audit(1670868835.418:961): dev=tap0211322eab5b prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[39666.659354] scsi host6: TCM_Loopback
[... cut ...]
[40897.706709] r8152 4-1:1.0 eth2: get_registers -71
[40897.724461] r8152 4-1:1.0 eth2: get_registers -71
[40897.742212] r8152 4-1:1.0 eth2: get_registers -71
[40897.759954] r8152 4-1:1.0 eth2: get_registers -71
[40897.777698] r8152 4-1:1.0 eth2: get_registers -71
[40898.288835] usb 4-1: reset SuperSpeed USB device number 3 using etxhci_hcd-170202
[40898.329951] r8152 4-1:1.0 eth2: Promiscuous mode enabled
[40898.654474] r8152 4-1:1.0 eth2: get_registers -71
[41216.860110] r8152 4-1:1.0 eth2: get_registers -71
```

bb-qq, 2022-12-31:
The latest drivers include a process to recover the driver without stopping it when get_registers -71 occurs. So, in environments where this error occurs, the error itself will continue to occur. 

If you are also experiencing stability issues, please let us know what symptoms you are experiencing.

tjjh89017, 2023-01-16:
I still get lots of  `r8152 3-1:1.0 eth4: get_registers -71` in dmesg
but network is stable.

tjjh89017, 2023-01-17:
I still found that NIC is disconnected without reasons.

```
OMIT
[2426295.819435] r8152 3-1:1.0 eth4: get_registers -71
[2426295.844084] r8152 3-1:1.0 eth4: get_registers -71
[2426295.868735] r8152 3-1:1.0 eth4: get_registers -71
[2426295.905740] r8152 3-1:1.0 eth4: get_registers -71
[2426295.930393] r8152 3-1:1.0 eth4: get_registers -71
[2426296.707934] usb 3-1: USB disconnect, device number 3
[2426299.360518] usb 3-1: new SuperSpeed USB device number 4 using etxhci_hcd-170202
[2426299.389548] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 2 with no TDs queued?
[2426299.399406] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[2426299.409384] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 6 with no TDs queued?
[2426300.522467] usb 3-1: reset SuperSpeed USB device number 4 using etxhci_hcd-170202
[2426300.748415] r8152 3-1:1.0 eth4: v2.16.3 (2022/07/06)
[2426300.754167] r8152 3-1:1.0 eth4: This product is covered by one or more of the following patents:
                                US6,570,884, US6,115,776, and US6,327,625.

[2426300.772157] r8152 3-1:1.0 eth4: chip rev 14
[2426300.777032] r8152 3-1:1.0 eth4: rx_buf_sz will be overridden to 16384
[2426355.148775] igb 0000:00:14.0 eth0: igb: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: None
[2426479.184505] Module [r8152] is removed.
[2426479.189058] usbcore: deregistering interface driver r8152
[2426482.180266] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 2 with no TDs queued?
[2426482.190219] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[2426482.200100] etxhci_hcd-170202 0000:04:00.0: WARN Event TRB for slot 1 ep 6 with no TDs queued?
[2426483.314054] usb 3-1: reset SuperSpeed USB device number 4 using etxhci_hcd-170202
[2426483.484938] r8152 3-1:1.0 eth4: v2.16.3 (2022/07/06)
[2426483.490693] r8152 3-1:1.0 eth4: This product is covered by one or more of the following patents:
                                US6,570,884, US6,115,776, and US6,327,625.

[2426483.508693] r8152 3-1:1.0 eth4: chip rev 14
[2426483.513568] r8152 3-1:1.0 eth4: rx_buf_sz will be overridden to 16384
[2426483.521012] usbcore: registered new interface driver r8152
[2426483.640329] IPv6: ADDRCONF(NETDEV_UP): eth4: link is not ready
[2426504.904988] igb 0000:00:14.0 eth0: igb: eth0 NIC Link is Down
[2426510.828428] IPv6: ADDRCONF(NETDEV_CHANGE): eth4: link becomes ready
[2426510.838020] r8152 3-1:1.0 eth4: carrier on
[2426555.180721] r8152 3-1:1.0 eth4: get_registers -71
[2426565.196001] r8152 3-1:1.0 eth4: get_registers -71
[2426577.216685] r8152 3-1:1.0 eth4: get_registers -71
[2426589.240173] r8152 3-1:1.0 eth4: get_registers -71
[2426603.265215] r8152 3-1:1.0 eth4: get_registers -71
[2426617.292250] r8152 3-1:1.0 eth4: get_registers -71
[2426630.308894] r8152 3-1:1.0 eth4: get_registers -71
[2426644.331561] r8152 3-1:1.0 eth4: get_registers -71
```

bb-qq, 2023-01-28:
How often does disconnection occur?

tjjh89017, 2023-01-28:
I first found this issue after 3 days (installing the 2.5g NIC)
and reboot it.
then next day (in 1day) it disconnect again.

bb-qq, 2023-03-18:
Well.., overall, [the avaton platform has a poor track record](https://github.com/bb-qq/r8152/wiki/Compatibility) of stable operation, so it is possible that the hardware itself is having problems.

tjjh89017, 2023-03-18:
> Well.., overall, [the avaton platform has a poor track record](https://github.com/bb-qq/r8152/wiki/Compatibility) of stable operation, so it is possible that the hardware itself is having problems.

Thanks for all your work
I will give up this platform
Thanks again

closed 2023-03-18 by tjjh89017