bb-qq/r8152 #240: Couldn't install driver on DSM 7.1-42661 Update 4

manapasms, 2022-08-28:
### Description of the problem
I installed the latest gemini version of the driver (2.15.0-10 DSM7.x) and failed.
Then I tried to install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su after this, it changed the status to Repair, and it told me "Failed to run the package service".
I also tried the Pre-release version, and it has the same error message.
I think about trying to clear the old package and run it again, but I'm not sure how to do it.

### Description of your products
My NAS is Synology DS920+.
Linux Family 4.4.180+ #42661 SMP Mon Jun 27 15:07:20 CST 2022 x86_64 GNU/Linux synology_geminilake_920+
DSM 7.1-42661 Update 4
My Ethernet Dongle is Plugable Ethernet Adapter 2.5Gbps 
https://www.amazon.com/Plugable-Ethernet-Adapter-Compatible-Thunderbolt/dp/B084L4JL9K/

### Description of your environment
Connect the Plugable USB to NAS, and the Ethernet Cable connects between the unmanaged switch and the NAS.
The Ethernet Switch is Zyxel MG-105.
The cable is Cat5e.

### Output of ``dmesg`` command
[logs.txt](https://github.com/bb-qq/r8152/files/9439063/logs.txt)

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0013D8A913F212)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:E1:36:35:55
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:e1ff:fe36:3555/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:8118 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5365 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:844236 (824.4 KiB)  TX bytes:1188650 (1.1 MiB)

docker-d2 Link encap:Ethernet  HWaddr 02:42:12:54:A4:3B
          inet addr:172.19.0.1  Bcast:172.19.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker070 Link encap:Ethernet  HWaddr 5E:D4:D6:3D:0F:8E
          inet6 addr: fe80::5cd4:d6ff:fe3d:f8e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6 errors:0 dropped:0 overruns:0 frame:0
          TX packets:175 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:612 (612.0 B)  TX bytes:7964 (7.7 KiB)

docker0c9 Link encap:Ethernet  HWaddr 36:09:31:B1:D7:62
          inet6 addr: fe80::3409:31ff:feb1:d762/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:18 errors:0 dropped:0 overruns:0 frame:0
          TX packets:181 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1372 (1.3 KiB)  TX bytes:8394 (8.1 KiB)

docker189 Link encap:Ethernet  HWaddr 3E:3A:3D:CC:B4:79
          inet6 addr: fe80::3c3a:3dff:fecc:b479/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:121 errors:0 dropped:0 overruns:0 frame:0
          TX packets:235 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:8898 (8.6 KiB)  TX bytes:13606 (13.2 KiB)

docker191 Link encap:Ethernet  HWaddr C6:DA:10:26:39:C4
          inet6 addr: fe80::c4da:10ff:fe26:39c4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:10 errors:0 dropped:0 overruns:0 frame:0
          TX packets:181 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:4042 (3.9 KiB)  TX bytes:9622 (9.3 KiB)

docker254 Link encap:Ethernet  HWaddr 4E:77:05:B3:1B:9A
          inet6 addr: fe80::4c77:5ff:feb3:1b9a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3954 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2755 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:581245 (567.6 KiB)  TX bytes:687745 (671.6 KiB)

docker292 Link encap:Ethernet  HWaddr 7A:66:D7:52:A3:4C
          inet6 addr: fe80::7866:d7ff:fe52:a34c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:169 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:7358 (7.1 KiB)

docker574 Link encap:Ethernet  HWaddr 12:23:E7:36:85:85
          inet6 addr: fe80::1023:e7ff:fe36:8585/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:121 errors:0 dropped:0 overruns:0 frame:0
          TX packets:248 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:8898 (8.6 KiB)  TX bytes:22188 (21.6 KiB)

docker825 Link encap:Ethernet  HWaddr 8E:0B:C1:38:63:05
          inet6 addr: fe80::8c0b:c1ff:fe38:6305/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:319 errors:0 dropped:0 overruns:0 frame:0
          TX packets:304 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:23030 (22.4 KiB)  TX bytes:19470 (19.0 KiB)

dockerb82 Link encap:Ethernet  HWaddr D6:F2:9C:A6:9E:9D
          inet6 addr: fe80::d4f2:9cff:fea6:9e9d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:169 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:0 (0.0 B)  TX bytes:7358 (7.1 KiB)

dockerc33 Link encap:Ethernet  HWaddr 76:D5:CC:11:40:12
          inet6 addr: fe80::74d5:ccff:fe11:4012/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:189 errors:0 dropped:0 overruns:0 frame:0
          TX packets:222 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:14146 (13.8 KiB)  TX bytes:12108 (11.8 KiB)

dockere5f Link encap:Ethernet  HWaddr EE:54:47:F5:4F:C4
          inet6 addr: fe80::ec54:47ff:fef5:4fc4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:26 errors:0 dropped:0 overruns:0 frame:0
          TX packets:180 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:1972 (1.9 KiB)  TX bytes:8273 (8.0 KiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:F3:27:47
          inet addr:169.254.33.101  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:93 base 0x4000

eth1      Link encap:Ethernet  HWaddr 00:11:32:F3:27:48
          inet addr:23.124.167.168  Bcast:23.124.167.255  Mask:255.255.254.0
          inet6 addr: fe80::211:32ff:fef3:2748/64 Scope:Link
          inet6 addr: 2600:1700:158:2e30::48/128 Scope:Global
          inet6 addr: 2600:1700:158:2e30:211:32ff:fef3:2748/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:50711 errors:0 dropped:0 overruns:0 frame:0
          TX packets:69777 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:17513297 (16.7 MiB)  TX bytes:31215677 (29.7 MiB)
          Interrupt:92 base 0x8000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:46998 errors:0 dropped:0 overruns:0 frame:0
          TX packets:46998 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:7930653 (7.5 MiB)  TX bytes:7930653 (7.5 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

bb-qq, 2022-09-06:
Could you run these commands and paste the output here?
```
sudo /var/packages/r8152/scripts/start-stop-status stop
sudo /var/packages/r8152/scripts/start-stop-status start

cfthnm, 2022-09-12:
I met a same situation.
Here is output of "sudo /var/packages/r8152/scripts/start-stop-status stop"

> ++ dirname /var/packages/r8152/scripts/start-stop-status
>  + script_root=/var/packages/r8152/scripts
> ++ realpath -m /var/packages/r8152/scripts/..
> + package_root=/var/packages/r8152
> ++ basename /var/packages/r8152
> + driver_name=r8152
> + driver_root=/var/packages/r8152/target/r8152
> + case $1 in
> + set_interfaces down
> + action=down
> + target_interface_name=
> ++ ls /sys/class/net
> + all_interface_names='docker0
> docker89bac60
> eth0
> eth1
> lo
> ovs_eth0
> ovs-system
> sit0
> syno_ovs_bonds'
> + '[' -n ']'
> + '[' '!' -e /sys/class/net/ ']'
> + for interface_name in ${all_interface_names}
> + [[ -n '' ]]
> + '[' '!' -e /sys/class/net/docker0/device/driver ']'
> + continue
> + for interface_name in ${all_interface_names}
> + [[ -n '' ]]
> + '[' '!' -e /sys/class/net/docker89bac60/device/driver ']'
> + continue
> + for interface_name in ${all_interface_names}
> + [[ -n '' ]]
> + '[' '!' -e /sys/class/net/eth0/device/driver ']'
> ++ ls -ld /sys/class/net/eth0/device/driver
> + driver_location='lrwxrwxrwx 1 root root 0 Sep 13 01:32 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/r8168'
> ++ grep r8152
> ++ echo 'lrwxrwxrwx 1 root root 0 Sep 13 01:32 /sys/class/net/eth0/device/driver -> ../../../../bus/pci/drivers/r8168'
> + '[' '!' -z '' ']'
> + for interface_name in ${all_interface_names}
> + [[ -n '' ]]
> + '[' '!' -e /sys/class/net/eth1/device/driver ']'
> ++ ls -ld /sys/class/net/eth1/device/driver
> + driver_location='lrwxrwxrwx 1 root root 0 Sep 13 01:32 /sys/class/net/eth1/device/driver -> ../../../../bus/pci/drivers/r8168'
> ++ echo 'lrwxrwxrwx 1 root root 0 Sep 13 01:32 /sys/class/net/eth1/device/driver -> ../../../../bus/pci/drivers/r8168'
> ++ grep r8152
> + '[' '!' -z '' ']'
> + for interface_name in ${all_interface_names}
> + [[ -n '' ]]
> + '[' '!' -e /sys/class/net/lo/device/driver ']'
> + continue
> + for interface_name in ${all_interface_names}
> + [[ -n '' ]]
> + '[' '!' -e /sys/class/net/ovs_eth0/device/driver ']'
> + continue
> + for interface_name in ${all_interface_names}
> + [[ -n '' ]]
> + '[' '!' -e /sys/class/net/ovs-system/device/driver ']'
> + continue
> + for interface_name in ${all_interface_names}
> + [[ -n '' ]]
> + '[' '!' -e /sys/class/net/sit0/device/driver ']'
> + continue
> + for interface_name in ${all_interface_names}
> + [[ -n '' ]]
> + '[' '!' -e /sys/class/net/syno_ovs_bonds/device/driver ']'
> + continue
[... 90 lines cut ...]
> + cp /var/packages/r8152/etc/ifcfg-eth2 /etc/sysconfig/network-scripts/ifcfg-eth2
> + ifconfig eth2 up
> + set_ovs_interface eth2 up
> + interface_name=eth2
> + action=up
> + config_file_ovs=/etc/sysconfig/network-scripts/ifcfg-ovs_eth2
> + config_storage_location_ovs=/var/packages/r8152/etc/ifcfg-ovs_eth2
> + is_ovs_enable
> + '[' '!' -f /usr/syno/etc/synoovs/ovs_reg.conf ']'
> ++ cat /usr/syno/etc/synoovs/ovs_reg.conf
> ++ wc -l
> + use=1
> + '[' 0 -eq 1 ']'
> + return 0
> + '[' -f /etc/sysconfig/network-scripts/ifcfg-ovs_eth2 ']'
> + '[' up = down ']'
> + '[' up = up ']'
> + '[' -f /var/packages/r8152/etc/ifcfg-ovs_eth2 ']'
> + cp /var/packages/r8152/etc/ifcfg-ovs_eth2 /etc/sysconfig/network-scripts/ifcfg-ovs_eth2
> + '[' up = up ']'
> + ovs-vsctl add-br ovs_eth2
> + ovs-vsctl add-port ovs_eth2 eth2
> ovs-vsctl: cannot create a port named eth2 because a port named eth2 already exists on bridge ovs_eth0

I do add eth2 to bridge ovs_eth0 manually.

bb-qq, 2022-09-17:
Interfaces added by this driver cannot use ovs-derived features such as making bond members.

cfthnm, 2022-09-17:
> Interfaces added by this driver cannot use ovs-derived features such as making bond members.

I fixed this problem by binding eth0&eth1 to ovs_eth2，and it works nice.

bb-qq, 2022-10-09:
In the next release, I change it so that it does nothing if the interface is already configured as a member of ovs. If you would like to try it out and let me know the results.

https://github.com/bb-qq/r8152/releases/download/2.16.3-1/r8152-geminilake-2.16.3-2test.spk

evanrich, 2023-03-25:
FWIW with same DSM version, and 2.16.3-4 version, I had the same problem on a DS620SLIM.   instlled driver, rebooted, ran the sudo -u etc command listed, package shows error and no interface.  only after I ran the stop/start command above did it finally show the nic.

bb-qq, 2025-01-02:
This Issue is closed because the driver already supports Open vSwitch. Please feel free to re-open it if you have any questions.

closed 2025-01-02 by bb-qq