bb-qq/r8152 #241: Package in error state with DSM 7.1.1

codex125, 2022-09-05:
### Description of the problem

It isn't possible to get the package running after updating to DSM 7.1.1
I used version 2.15.0-7 before updating DSM and tried 2.15.0-10 and 2.16.3-1 after the update.
Both versions gave an error at the installation and the NIC isn't shown in the system settings.

### Description of your products

DS218+
DSM 7.1.1-42962 
Club 3D CAC-1420
[image]
[image]

mrjohnpoz, 2022-09-05:
Did you run the sudo command, I just updated to 7.1.1 and had the same thing just run the listed sudo command again and all working as before.

codex125, 2022-09-05:
That worked, thank you very much!
Stupid that I have not tried that.

closed 2022-09-05 by codex125