bb-qq/r8152 #244: Only USB0 at Xpenology DS918 / DSM7.0.1

macace2000, 2022-09-15:
Hello

I have tried your driver package at my xpenology baremetal system DSM7.0.1 apollolake.
The mainboard has only one nic. (The original DS918 has two)
After installation,  the CLUB 3D CAC-1420 ist listed after Ifconfig -a command in ssh only as usb0.
The internal nic is eth0

When i build a second PCI-E nic in the system this nic become eth1.
When i Install now the USB-Nic, it works correct at eth2. (Three Lan Ports)

After a shutdown and removing the PCI-E Nic, the USB-Nic will become only usb0 again.
I use the automated repill loader 0.4 alpha 7. Config mac1, netif num etc. makes no difference.

bb-qq, 2022-09-17:
Could try this command to enable udev rules?
```
sudo bash /var/packages/r8152/scripts/install-udev-rules
```

bb-qq, 2023-08-13:
I would like to close this issue since there has been no response. If you have any questions, feel free to re-open it.

closed 2023-08-13 by bb-qq