bb-qq/r8152 #245: Synology DS416j driver - MARVELL Armada 388 architecture

JohnnySKK, 2022-09-16:
Dear bb-qq,

could you also include the MARVELL Armada 388 ARCH in the pre-built drivers please? I have one DS416j NAS for specific backups and would like to test it with my 2.5Gbit USB adapters. 

I tried the driver on a DS414j (comcerto2k arch)  - also with 512MB RAM - and it works - I have to check the actual speeds once the RAID array is rebuilt (~2 days left..), the CPU is hammered right now :)

Thanks a lot!

bb-qq, 2022-09-18:
I added the driver for armada38x models.
https://github.com/bb-qq/r8152/releases/download/2.16.3-1/r8152-armada38x-2.16.3-1.spk

bb-qq, 2022-10-09:
I would appreciate it if you report the working status.

bb-qq, 2023-08-13:
I would like to close this issue since there has been no response. If you have any questions, please create a new issue.

closed 2023-08-13 by bb-qq