bb-qq/r8152 #246: Unable to Configure Static IP after Upgrade to DSM 7.1.1-42962 Update 1

TheS1R, 2022-09-22:
After overnight update of DS718+ (apollolake) to DSM 7.1.1-42962 Update 1, I lost access to NAS via 2.5 Gbps adaptor — second DS718+ was still accessible via 2.5 Gbps adaptor. After troubleshooting, I discovered that the only difference between two NASs was one configured using static IP (_not working_) and other configured using DHCP (_working_). The same situation occurs for both 2.16.3-1 DSM7.x as well as 2.15.0-10 DSM7.x. I have seen this earlier, but it had since resolved itself.

Static IPs via DHCP reservations configured in Asus router for both. Changing static IP to DHCP in NAS resolved issue.

onkelblaubaer, 2022-09-23:
I have the same problem with my DS920+ (apollolake) after updating to DSM 7.1.1-42962 Update 1. The connection via the USB adapter only worked again after I switched to DHCP. Thanks to TheS1R for the hint with the switch to DHCP. The IP address remained the same after switching to DHCP. I have the driver 2.15.0-10 DSM7.x in use.

TheS1R, 2022-09-23:
@onkelblaubaer Be sure to make a DHCP reservation in your router to ensure that the IP address does not change over time.

Jasmine3352, 2022-09-23:
I have the following inputs that might help out:  DSM 7.1.1-42962 Update 1 ( DS 1815, DS1515 and DS918+)
I have the driver 2.15.0-10 DSM7.x in use.

After any restarting the NAS, the IP are NOT seen by the Synology Assistant. 
I use static IPs. I change to another IP  xxx.xxx.xxx.x20 to xxx.xxx.xxx.21 , save, then change back, it works.
removing / replacing the LAN cable doesn't help.

I have verified that reserving a IP in the DHCP pool does indeed work. 
After any restarting the NAS, the IP are now seen properly by the Synology Assistant.

gcgheinen, 2022-09-24:
The same for me. After I switched to DHCP and back again, the network card works again. (DSM218+ , network adapter RTL8156B)

onkelblaubaer, 2022-09-24:
> @onkelblaubaer Be sure to make a DHCP reservation in your router to ensure that the IP address does not change over time.

i have reserved the ip address and assigned it to the MAC address. However, the IP address is not assigned by the DHCP server. This address is not in the range of the DHCP server.

onkelblaubaer, 2022-09-24:
> I have the following inputs that might help out: DSM 7.1.1-42962 Update 1 ( DS 1815, DS1515 and DS918+) I have the driver 2.15.0-10 DSM7.x in use.
>
> After any restarting the NAS, the IP are NOT seen [...]

I try this solution, but when i restart the NAS, i have no connection again. I switched back to DHCP. This workaround also survives a reboot.

onkelblaubaer, 2022-09-24:
> The same for me. After I switched to DHCP and back again, the network card works again. (DSM218+ , network adapter RTL8156B)

I tried this method as well, but after rebooting again no connection to the NAS was possible. Only switching to DHCP "survives" a reboot.

blasiusx, 2022-09-24:
DS916+ and DS918+ with DSM 7.1.1 Update 1. Club 3D 2.5 GBe Adapter. Driver 2.15.0-10 and 2.16.3-1.

> I try this solution, but when i restart the NAS, i have no connection again. I switched back to DHCP. This workaround also survives a reboot.

I have to confirm that. Entering a static IP works....But unfortunately only until the next restart of the disk station because the connection failed after the reboot again. It looks like the static IP is not stored permanently in the Diskstation. DHCP works for me as the only solution, and then assign it as a fixed static IP in the router.

With both driver 2.15.0-10 and 2.16.3-1.

Jasmine3352, 2022-09-24:
I hope I haven't confused folks,
I am running DSM 7.1.1-42962 Update 1 (**DS620slim, DS1815+, DS1515+ and DS918+**) driver 2.15.0-10.
The 2.5GBE adapters IP/MAC are specified in the DCHP Server Address Reservation on my router. and this is working fine. 
The IPs are _within_ the IP Address Pool (192.168.0.100 - 192.168.0.249)

mrjohnpoz, 2022-09-25:
Have run into this as well.. ds918+ whenever you reboot the static set IP does not work.  It shows up in the dsm that static is set.  But if you look with ifconfig the IP is not on the interface.

Changing to dhcp (which fails since no dhcp on this network gives you 169.254.x.x address) then switching it back to static and cycle of the interface now works.

It is something that can live with, because its rare that the nas reboots - normally only on a dsm update.  But it is annoying.

520hacker, 2022-09-29:
I has been update to DSM 7.1.1-42962 Update 1 , let me show how I resovle the issue . 
- my network setting is nas(10.0.0.1) directly connected to my windows pc(10.0.0.2),its used for internal file transfers only.
- I found the net disk from smb icon was gray, can not open on PC 's explorer. 
- ping nas result is time out .
- can not open any service from nas by the 10.0.0.1 .
- I found the ip setting both on nas and my local pc seems correctly but actually its not works .

I did the changes and the at last the network working now 
- I tried disable the network card and enable it again on nas.
- I tried run the command via SSH  sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su
- I tried enbaled DHCP service on nas , and set 10.0.0.1 as net route .
- I tried use DHCP mode on PC ,and reset it to 10.0.0.2 as manually ip .
- I tried disable the network card and enable it again on PC.

the magic is , network back !!!
not sure which step is the key, any way , its working again.
haha.

bb-qq, 2022-10-09:
Could you try this test version?
https://github.com/bb-qq/r8152/releases/download/2.16.3-1/r8152-apollolake-2.16.3-2test.spk

hhessel, 2022-10-09:
> Could you try this test version? https://github.com/bb-qq/r8152/releases/download/2.16.3-1/r8152-apollolake-2.16.3-2test.spk

Unfortunately, in this version the same bug appears. Even though the static IP is visible in the DSM UI it's not active after the reboot and you need to switch over to DHCP and back to a static config to reenable the  interface.

bb-qq, 2022-10-09:
Hmm, could you paste the log saved as `/var/packages/r8152/var/log/start-stop-status.log`?
(Logging feature was added from this version.)

blasiusx, 2022-10-09:
DS918+

The new test driver 2.16.3-2 test works fine for me. Survived several reboots, from Static IP to Dynamic IP and vice versa. After a restart, the disk station is recognized, even with a static IP. (Synology Assissant shows static IP after Reboot).

The test driver could not be installed as an update for me, I had to completely uninstall the old one (2.15.0-10), reboot and then reinstall it. Of course, via the usual installation instructions. (sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su).

I've done a few copy tests, for me at least everything looks very, very good.

Now one more driver for the Braswell chipset for the Diskstation and I'm happy. :-)

Many thanks for the great work, "bb-qq".

hhessel, 2022-10-09:
> Hmm, could you paste the log saved as `/var/packages/r8152/var/log/start-stop-status.log`? (Logging feature was added from this version.)

Here you go. Also removed the old version but didn't reboot before installing  2.16.3-2 so this might be the difference to @blasiusx 

[start-stop-status.log](https://github.com/bb-qq/r8152/files/9741060/start-stop-status.log)

hhessel, 2022-10-09:
Okay, removed the old version, rebooted and installed the test version. Seems to work now. Many thanks!

TheS1R, 2022-10-09:
Same here -- uninstalled current version, rebooted and installed this test version. Static IP appears to be working correctly again. THANKS!

Jasmine3352, 2022-10-09:
Static IP after Upgrade to DSM 7.1.1-42962 Update 1 – verified ok on DS918+ , bounced between DCHP and static IP a few times, discovered properly.

Rich

https://github.com/bb-qq/r8152/releases/download/2.16.3-1/r8152-apollolake-2.16.3-2test.spk<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fbb-qq%2Fr8152%2Freleases%2Fdownload%2F2.16.3-1%2Fr8152-apollolake-2.16.3-2test.spk&data=05%7C01%7C%7Cd13e3dcf0b8b4d63dd0108daa9bc36fb%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638008923095522670%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=ISbG0CqGddEAIcG9HkMVGkymMgfLbjHahMLeicyA4A4%3D&reserved=0>

Jasmine3352, 2022-10-10:
Upgrade to DSM 7.1.1-42962 Update 1 on DS620Slim apollolake NOT installing properly.
Coming up saying it needs to be repaired.

https://github.com/bb-qq/r8152/releases/download/2.16.3-1/r8152-apollolake-2.16.3-2test.spk<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fbb-qq%2Fr8152%2Freleases%2Fdownload%2F2.16.3-1%2Fr8152-apollolake-2.16.3-2test.spk&data=05%7C01%7C%7Cd13e3dcf0b8b4d63dd0108daa9bc36fb%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638008923095522670%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=ISbG0CqGddEAIcG9HkMVGkymMgfLbjHahMLeicyA4A4%3D&reserved=0>

-

TheS1R, 2022-10-10:
If installation needs to be repaired (_but that doesn't work_), uninstall, reboot, and install update from scratch. This will require running the "_sudo install ..._" command as outlined in the **Releases** page instructions.

onkelblaubaer, 2022-10-10:
> Could you try this test version? https://github.com/bb-qq/r8152/releases/download/2.16.3-1/r8152-apollolake-2.16.3-2test.spk

Hello, 
the update to the version r8152-geminilake-2.16.3-2test.spk on my DS920+ also gave an error. I typed "sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su" again via the console for safety. Nevertheless, the new driver could not be installed. However, version 2.16.3-2 was already displayed in the package center. I changed the config from DHCP to static and rebooted - The diskstation was reachable after that and the IP address remained static...

Good Work :-)

Conclusion: works also with error message. The driver does not need to be removed before installing the new one. (in my case ;-))

mrjohnpoz, 2022-10-10:
So uninstalled my 2.16.3-1 that I had installed.  Rebooted and installed the 2test version of apollolake on my ds918+ didn't even have to do the cli command.. It just installed just fine.. no errors.  Saw that it even remembered my static IP..  Iperf test to the ip looked good.

I then rebooted again for good measure to see if came up all on its own with the IP.. All good, iperf good test.

Whatever you did seems to have fixed the issue - all good here, thanks so much!

Jasmine3352, 2022-10-10:
I can now verify that the 'test' is operational on my DS620slim device.

Maybe not the place to post but had a question:

Has anyone been successful using 2 adapters, or bonding of 2 adapters?

Just asking.

Rich

bb-qq, 2022-10-11:
Thank you all. I close this issue since officially versioned drivers have been released.
https://github.com/bb-qq/r8152/releases/tag/2.16.3-2