bb-qq/r8152 #247: How to configure new driver?

aeon3, 2022-09-24:
Hi! I'm very new to NAS interface. I just followed the steps to install the driver but I can't find a way to configure it.

Also it says first time running I should do special steps, but I can't find the readme to follow those required special steps.

THANKS!

Sjekke, 2022-09-24:
This?

Note:

The installation will fail the first time. After that, run the following command from the SSH terminal:
sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su

aeon3, 2022-09-24:
> This?
> 
> Then make a script in TaskScheduler (root) with this commands
> 
> ifconfig eth2 up synonet –dhcp eth2

Thank you!

2 concerns though:

1- How do I make this script? do I SSH from my pc? I have no clue how to arrive at the part

2- I dont think a new NIC was added to my NAS after the driver installation. I only see LAN 1, LAN 2 and PPPOE. I have a ds920+ which has 2 rj45 ports. So shouldnt I be seeing a 3rd LAN?

Sjekke, 2022-09-24:
Did you already execute : 

sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su via SSH??

You can find the Taskscheduler under Control Panel > Task Scheduler > User defined script. Give a name for your script and run it is as Root. Add 

ifconfig eth2 up
synonet –dhcp eth2

Save it and reboot.

TheS1R, 2022-09-24:
What is the purpose of script? What schedule did you set up?

Sjekke, 2022-09-24:
If your NAS already has 2 network ports (eth0 and eth1), the new 2.5 Gbit/s adapter will be called eth2.  If your NAS has only one network port by default, the new adapter will be named eth1.  If your NAS has 4 network ports, the adapter will be named eth4... The first line (above) will mount/upload the new network card.  The second applies the default DHCP setting to this card. Scheduled > boot

aeon3, 2022-09-24:
Followed everything you mentioned but still doesnt work :(

> If your NAS already has 2 network ports (eth0 and eth1), the new 2.5 Gbit/s adapter will be called eth2. If your NAS has only one network port by default, the new adapter will be named eth1. If your [...]

aeon3, 2022-09-24:
[image]

aeon3, 2022-09-24:
lan 1 works fine, its the one in the back of the NAS. When I connect the ethernet cable to the adapter I lose connection.

Sjekke, 2022-09-25:
Can you edit LAN 3?

aeon3, 2022-09-25:
> Can you edit LAN 3?

Yes, I think it might have worked now. I edited it and gave it a static IP similar do the DHCP one.

[image]

Is there something else I need to doublecheck?

Sjekke, 2022-09-25:
Then it should work. What is the connection speed?

aeon3, 2022-09-25:
I havent received my 2.5G switch yet to test the full speed. Right now it says 1000 Mbps and transfer speeds max out at around 110 MB/s

I couldn't find a way to change MTU to 1900 and SMB protocol to v3. DO you know how I can do those changes?

TheS1R, 2022-09-25:
DSM --> Control Panel --> Network --> Network Interface --> LAN 3 --> Edit --> Set MTU value manually.
DSM --> Control Panel --> File Services --> SMB --> Advanced Settings --> Minimum/Maximum SMB protocol.

Sjekke, 2022-09-25:
> I havent received my 2.5G switch yet to test the full speed. Right now it says 1000 Mbps and transfer speeds max out at around 110 MB/s.
> 
> 
> 
> Should arrive in the next few days :)
> 
> 
> 
> However, I couldn't find a way to change MTU to 1900 and SMB protocol to v3. DO you know how I can do those changes?

Ok 👍. Glad I could help.

aeon3, 2022-09-25:
> DSM --> Control Panel --> Network --> Network Interface --> LAN 3 --> Edit --> Set MTU value manually. DSM --> Control Panel --> File Services --> SMB --> Advanced Settings --> Minimum/Maximum SMB protocol.

thanks! should I put SMB3 in both min and max?

aeon3, 2022-09-25:
> Ok 👍. Glad I could help.

thank you so much, couldnt have done it without you.

aeon3, 2022-10-07:
Guys one last question that came to my mind, if anyone can chime in. Since I changed the MTU to 1900 in the NAS connection, should I also find a way to change to in each of my computers that are connecting to the NAS? Those are still with the default 1500 MTU.

aeon3, 2022-10-07:
> MTU of 1900? Do you mean 9000?

Yes, my bad. 9000. Is it enough to change just that parameter just in the NAS network settings?

aeon3, 2022-10-07:
> You must configure it on every device that will be accessing the NAS ...

So I have my PC and NAS connected to a switch which is connected to a router which is connected to a modem.

The switch is auto managed so I cant do anything with it. 

In this scenario which devices would I need to tweak the MTU? Only my Pc?

aeon3, 2022-10-14:
guys its me again.

so I just updated my NAS and it wiped out the 2.5G LAN settings...will that happen with every update? Is there a way to bypass that? Or a way to make it reconfigure those upon finishing the update?

i dont like how unreliable it turns out to be if im away from the NAS and I lose connection for good. I had to take out the cable from the 2.5G adapter and replug it back into one of the regular 1G ports to get access again.

bb-qq, 2022-12-10:
> In this scenario which devices would I need to tweak the MTU? Only my Pc?

You may need to change the router's setting, but no need for the switch typically.

> so I just updated my NAS and it wiped out the 2.5G LAN settings...will that happen with every update?

Usually, the settings are retained after the DSM is updated.
Please create a new issue and let me know if the problem recurs.

bb-qq, 2023-08-13:
I would like to close this issue since there has been no response. If you have any questions, feel free to re-open it.

closed 2023-08-13 by bb-qq