bb-qq/r8152 #249: DS218+ Report

gaoo90, 2022-10-02:
DS 218+
DSM 7.1-42661 Update 4
2500Mbps USB C 2.5G External Wired Network Card Type-C To RJ45 Converter Ethernet Lan Adapter Hub For MacBook Ethernet
https://aliexpress.ru/item/1005003392518774.html?businessType=ProductDetail&item_id=1005003392518774&shareFrom=screenShot&sku_id=12000025571959698&spreadType=socialShare&srcSns=sns_More&tt=MG
Specifications:
Chipset:Realtek RTL8156 or Realtek RTL 8153
Interface:
Input Interface: 1*USB 3.0 Port
Output Interface: 1*10/100/1000Mbps/2500Mbps RJ45 Ethernet Port or 1*10/100/1000Mbps RJ45 Ethernet Port.

Driver 2.16.3-1

Asus RT-AX86U 2.5g Ethernet Port

1. If the link adapter is used in front of the DS then from time to time it disappears from the system interface. It helps to stop and start the driver package.

2. If the adapter is used in the back of the DS? the operation is stable and quite reliable.
Speeds are achieved at 160MB per second when using HDD 4-6TB disks

DervilRus, 2023-01-02:
DS920+
2.16.3-3 DSM7.x (reuploaded)
lan RTL8156 (ks-is ks-714)
https://ks-is.com/usb-3-1-ethernet-adapter-ks-is-ks-714?tag=2.5G

There are no problems with data transfer. Especially for a couple of hours I drove chia 100gb plots at a speed of 2.5.
But there is another problem! When you pull out and put back the adapter, the driver turns off. The same goes for rebooting. Must be manually enabled in the web interface.

bb-qq, 2023-01-29:
Hot-plugging the device is disabled by default. If you wish to enable it, please run the following command.

```
sudo bash /var/packages/r8152/scripts/install-udev-rules

DervilRus, 2023-01-29:
> Hot-plugging the device is disabled by default. If you wish to enable it, please run the following command.
> 
> ```
> sudo bash /var/packages/r8152/scripts/install-udev-rules
> ```

Thank you! I'll check! If this fixes it, then the driver is perfect. For 3 weeks I did not have a single failure, although I constantly copy large files at speed 2.5

VTimofeenko, 2025-01-25:
DS218+, DSM 7.1.1-42962 Update 6. Realtek RTL8156BG based adapter.

2.15.0-4 did not work upon first install (no adapter seen by the system but I didn't try rebooting). Installing [2.19.2-1](https://github.com/bb-qq/r8152/releases/tag/2.19.2-1) immediately showed adapter with appropriate speed.

Works fine after reboot with no extra configuration.

I also switched the main NAS IP to 2.5G interface and I _think_ DSM tried to re-request old IP from the (standalone) DHCP server when I rebooted the NAS. I had to clear the cache lease, restart DHCP and restart Synology twice before it got the old IP on the new interface.