bb-qq/r8152 #250: DS920+ new version now working

havrancek, 2022-10-11:
### Description of the problem

r8152-geminilake-2.16.3-2.spk is not working after the update, is stopped and need a repair

### Description of your products

DS920+

r8152-geminilake-2.16.3-1.spk is working as intended, when tried to update to 
r8152-geminilake-2.16.3-2.spk - it stops and need a repair
files have different size although in the past every new version was certain size

Sjekke, 2022-10-11:
Same here on 916+ Braswell platform...

BerndFPV, 2022-10-11:
Same here DS920+ 7.1.1
When I install r8152-geminilake-2.16.3-2.spk - is not working need to repair ...
Version r8152-geminilake-2.16.3-1.spk work but has after a reboot the static IP bug :-(

havrancek, 2022-10-11:
That doesn't work, it won't become active. It's stopped and needs repair.

dreimer1986, 2022-10-12:
When I went from 2.15.0-8 to 2.16.3-1 I had the same behavior. I fixed it by repair and then
sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su

It's quite standard for me coming from oooold builds of this driver to have a 2nd LAN adapter with the same IP for these updates as this happens if you toy around on a kernel module when it's running.

bb-qq, 2022-10-12:
Try the setup procedure again from the beginning.

Uninstall package -> Reboot -> Install package -> Run above command -> Install package

If this does not work for you. please paste `/var/log/synopkg.log` and `/var/packages/r8152/var/log/start-stop-status.log` here.

Githwx, 2022-10-19:
I encountered the same issue when i upgraded the driver to 2.16.3.2 from 2.15.0.10 version. It worked but failed to activate and leaving the message of "Repair".  i already tried a few times but the issue remain.  Not sure how to proceed.                                Mine is DS415+ with DSM 7.1.1 42962 update 1.

bb-qq, 2022-11-23:
Could you try the latest release?
https://github.com/bb-qq/r8152/releases/tag/2.16.3-3

bb-qq, 2022-11-23:
Sorry, I uploaded the wrong packages, so I re-uploaded the correct ones. Please re-download the package.

Githwx, 2022-11-25:
Thank you so much for the reply. 

However, the issue remain the same even I tried the new version of driver.

What should I do then ? Is there any info I need to share to help on diagnosis ? Pls advise. 

 

Thanks.  

答复: bb-qq/r8152 ***@***.***>
日期: 2022年11月23日 星期三 21:27
收件人: bb-qq/r8152 ***@***.***>
抄送: Githwx ***@***.***>, Comment ***@***.***>

bb-qq, 2022-11-25:
I need `/var/packages/r8152/var/log/start-stop-status.log` to investigate your case.

bb-qq, 2022-12-10:
Since I have not heard back from you in a while, I would like to close this issue.
If version 2.16.3-3 does not work, I think it is a different problem than the first one.
So if you wish to continue to the conversation, please create another new issue.

closed 2022-12-10 by bb-qq