bb-qq/r8152 #251: 920+ 7.1.1

s128300901, 2022-10-12:
sergey@synology:~$ sudo install -m 4755 -o root -D /var/packages/r8152/target/r8152/spk_su /opt/sbin/spk_su
install: cannot stat '/var/packages/r8152/target/r8152/spk_su': No such file or directory

bb-qq, 2022-10-12:
Try the procedure again from the beginning.

Uninstall package -> Install package -> Run above command -> Install package

If this does not work for you. please paste `/var/log/synopkg.log` here.

BlackBird11, 2022-10-13:
for me the driver also didn't work anymore
uninstalled -> rebooted -> installed newest -> failed -> copied command from above -> driver sucessfully running

unfortunately it still is not working with the adapter for me. i just can't access my nas

NAS: 918+ (apollolake)

bb-qq, 2022-10-13:
In the case of the driver successfully running, it is another issue. 
Please create a new issue with `/var/packages/r8152/var/log/start-stop-status.log` (need update to [2.16.3-2](https://github.com/bb-qq/r8152/releases/tag/2.16.3-2))

saquibmian, 2023-01-12:
This is happening for me, too.

Output of `/var/log/snypkg.log`:
```
2023/01/12 14:39:39 Use given volume to install, volume=[]
2023/01/12 14:39:39 Use any available volume to install, volume=[/volume1]
```
I am running DSM 7.1.1.-42962 Update 2.

bb-qq, 2023-01-28:
Are you using Safari?
There have been reports of problems with Safari regarding package installation.

saquibmian, 2023-02-11:
I was able to install the package using `synopkg`.

bb-qq, 2023-03-18:
It appears that not everyone has been able to resolve the issue, but since there seems to be no response, I will close it.

If you encounter similar problems, please create a new issue with information such as the browser you are using.

closed 2023-07-16 by bb-qq