bb-qq/r8152 #252: DS2413+ nas does not show up on the network

Manni1000, 2022-10-12:
i have a DS2413+ nas and installed the newest driver version. After the shh command it looked like the installation worked on the second try like deskibed in the installation tutorial. but the nas does not show up in the network when its conected via the usb adapter. Did I do something wrong? how can i fix that? Sadly its not possible to install a older version because sinology says i cant downgrade it.

Manni1000, 2022-10-12:
[start-stop-status.log](https://github.com/bb-qq/r8152/files/9768016/start-stop-status.log)

Manni1000, 2022-10-12:
[image]
the software says that its installed. but it does not show up.

bb-qq, 2022-10-13:
Your device seems to not be recognized correctly. Please paste the outputs of `lsusb` and `dmesg`.

Manni1000, 2022-10-13:
where are these files stored?

Manni1000, 2022-10-13:
lusb:
|__usb1          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1a.7) hub
|__usb2          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1d.7) hub
  |__2-1         f400:f400:0100 00  2.00  480MBit/s 480mA 1IF  (Synology Diskstation 0000000000000AE6)
|__usb3          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1a.0) hub
|__usb4          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.0) hub
|__usb5          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.1) hub
|__usb6          1d6b:0001:0310 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:00:1d.2) hub
|__usb7          1d6b:0002:0310 09  2.00  480MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:06:00.0) hub
|__usb8          1d6b:0003:0310 09  3.00 5000MBit/s 0mA 1IF  (Linux 3.10.108 etxhci_hcd-170202 Etron xHCI Host Controller 0000:06:00.0) hub
  |__8-2         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001300E04C68030E)

Manni1000, 2022-10-13:
dsmg:
[    0.000000] Mac4: 00c3679736b3
[    0.000000] Serial Number: DAKSN00020
[    0.000000] PID hash table entries: 4096 (order: 3, 32768 bytes)
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes)
[    0.000000] Memory: 4020816k/4718592k available (4642k kernel code, 536668k absent, 161108k reserved, 4199k data, 824k init)
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  CONFIG_RCU_FANOUT set to non-default value of 32
[    0.000000]  RCU dyntick-idle grace-period acceleration is enabled.
[    0.000000] NR_IRQS:4352 nr_irqs:712 16
[    0.000000] Console: colour dummy device 80x25
[    0.000000] console [ttyS0] enabled
[    0.000000] allocated 16777216 bytes of page_cgroup
[    0.000000] please try 'cgroup_disable=memory' option if you don't want memory cgroups
[    0.000000] hpet clockevent registered
[    0.000000] tsc: Fast TSC calibration using PIT
[    0.001000] tsc: Detected 2128.195 MHz processor
[    0.000005] Calibrating delay loop (skipped), value calculated using timer frequency.. 4256.39 BogoMIPS (lpj=2128195)
[    0.010647] pid_max: default: 32768 minimum: 301
[    0.015308] init_memory_mapping: [mem 0xde76c000-0xde76dfff]
[    0.020976]  [mem 0xde76c000-0xde76dfff] page 4k
[    0.021012] init_memory_mapping: [mem 0xdf345000-0xdf346fff]
[    0.026673]  [mem 0xdf345000-0xdf346fff] page 4k
[    0.026700] init_memory_mapping: [mem 0xdf34a000-0xdf34bfff]
[    0.032364]  [mem 0xdf34a000-0xdf34bfff] page 4k
[    0.032389] init_memory_mapping: [mem 0xdf360000-0xdf360fff]
[    0.038056]  [mem 0xdf360000-0xdf360fff] page 4k
[    0.038081] init_memory_mapping: [mem 0xdf361000-0xdf361fff]
[    0.043747]  [mem 0xdf361000-0xdf361fff] page 4k
[    0.043772] init_memory_mapping: [mem 0xdf363000-0xdf363fff]
[    0.049439]  [mem 0xdf363000-0xdf363fff] page 4k
[    0.049463] init_memory_mapping: [mem 0xdf364000-0xdf376fff]
[    0.055130]  [mem 0xdf364000-0xdf376fff] page 4k
[    0.055154] init_memory_mapping: [mem 0xdf377000-0xdf37afff]
[    0.060822]  [mem 0xdf377000-0xdf37afff] page 4k
[    0.060850] init_memory_mapping: [mem 0xdf380000-0xdf380fff]
[    0.066515]  [mem 0xdf380000-0xdf380fff] page 4k
[    0.066539] init_memory_mapping: [mem 0xdf392000-0xdf392fff]
[    0.072206]  [mem 0xdf392000-0xdf392fff] page 4k
[    0.072230] init_memory_mapping: [mem 0xdf3a2000-0xdf3a3fff]
[    0.077897]  [mem 0xdf3a2000-0xdf3a3fff] page 4k
[    0.077919] init_memory_mapping: [mem 0xdf3b1000-0xdf3b1fff]
[    0.083580]  [mem 0xdf3b1000-0xdf3b1fff] page 4k
[    0.083601] init_memory_mapping: [mem 0xdf3b4000-0xdf3b5fff]
[    0.089263]  [mem 0xdf3b4000-0xdf3b5fff] page 4k
[    0.089286] init_memory_mapping: [mem 0xdf3dc000-0xdf3ddfff]
[    0.094955]  [mem 0xdf3dc000-0xdf3ddfff] page 4k
[    0.094976] init_memory_mapping: [mem 0xdf3e8000-0xdf3e8fff]
[    0.100637]  [mem 0xdf3e8000-0xdf3e8fff] page 4k
[    0.100658] init_memory_mapping: [mem 0xdf3f2000-0xdf3f2fff]
[    0.106320]  [mem 0xdf3f2000-0xdf3f2fff] page 4k
[    0.106347] init_memory_mapping: [mem 0xdf469000-0xdf469fff]
[    0.112012]  [mem 0xdf469000-0xdf469fff] page 4k
[    0.112034] init_memory_mapping: [mem 0xdf46a000-0xdf46afff]
[    0.117694]  [mem 0xdf46a000-0xdf46afff] page 4k
[    0.117716] init_memory_mapping: [mem 0xdf483000-0xdf483fff]
[    0.123377]  [mem 0xdf483000-0xdf483fff] page 4k
[    0.123398] init_memory_mapping: [mem 0xdf6e8000-0xdf6e8fff]
[    0.129061]  [mem 0xdf6e8000-0xdf6e8fff] page 4k
[    0.129081] init_memory_mapping: [mem 0xdf6ee000-0xdf6eefff]
[    0.134743]  [mem 0xdf6ee000-0xdf6eefff] page 4k
[    0.146621] Security Framework initialized
[    0.150737] AppArmor: AppArmor initialized
[    0.154860] Mount-cache hash table entries: 256
[    0.159706] Initializing cgroup subsys memory
[    0.164104] Initializing cgroup subsys devices
[    0.168558] Initializing cgroup subsys freezer
[    0.173009] Initializing cgroup subsys blkio
[    0.177325] Disabled fast string operations
[... 951 lines cut ...]
[ 4864.431483] r8152 2-2:1.0 eth2: v2.16.3 (2022/07/06)
[ 4864.436729] r8152 2-2:1.0 eth2: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[ 4864.452659] r8152 2-2:1.0 eth2: chip rev 14
[ 4864.457059] r8152 2-2:1.0 eth2: rx_buf_sz will be overridden to 16384
[ 4873.435584] usb 2-2: USB disconnect, device number 3
[ 4877.679496] usb 8-2: new SuperSpeed USB device number 4 using etxhci_hcd-170202
[ 4877.708116] etxhci_hcd-170202 0000:06:00.0: WARN Event TRB for slot 1 ep 2 with no TDs queued?
[ 4877.717057] etxhci_hcd-170202 0000:06:00.0: WARN Event TRB for slot 1 ep 3 with no TDs queued?
[ 4877.725940] etxhci_hcd-170202 0000:06:00.0: WARN Event TRB for slot 1 ep 6 with no TDs queued?
[ 4878.837078] usb 8-2: reset SuperSpeed USB device number 4 using etxhci_hcd-170202
[ 4879.063923] r8152 8-2:1.0 eth2: v2.16.3 (2022/07/06)
[ 4879.069162] r8152 8-2:1.0 eth2: This product is covered by one or more of the following patents:
                        US6,570,884, US6,115,776, and US6,327,625.

[ 4879.085101] r8152 8-2:1.0 eth2: chip rev 14
[ 4879.089529] r8152 8-2:1.0 eth2: rx_buf_sz will be overridden to 16384
[ 5144.593707] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[15556.045872] e1000e: eth0 NIC Link is Down
[39718.701066] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[39722.168786] e1000e: eth0 NIC Link is Down
[39724.547882] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx
[39903.398922] e1000e: eth0 NIC Link is Down
[55769.289267] e1000e: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: Rx/Tx

Manni1000, 2022-10-13:
but only while i shh into the nas ofcorese.

Manni1000, 2022-10-15:
can someone help me please?

bb-qq, 2022-12-10:
Please try the latest drivers.
If that does not work, please let us know if the new NIC appears in the network configuration screen.

bb-qq, 2023-07-16:
I would like to close this issue as I have not heard back from you in a long time. If you have any questions, feel free to re-open it.

closed 2023-07-16 by bb-qq