bb-qq/r8152 #253: USB 2.5G dongle cannot be used as gateway

lymatthew, 2022-10-14:
Had this issue since DSM 7.1.1-42962 Update 1.  In fact, the 2.5G LAN does not show up in Network Interface under "Edit Service Order"

Using a DS 718+ and a UGREEN USB A to 2.5G adaptor (RTL8152). Had to connect the onboard 1G ethernet, otherwise cannot connect to QuickConnect or DDNS.

Nuufe, 2022-10-17:
Hi, same here. Since last update the network adapter is showing up but no connection is possible. Using the latest driver.
Now i am using the internal 1Gbit port too... :-(

agreatdayeveryday, 2022-10-25:
I submitted an issue but not sure if it's related to the 2.16.3-1 version, using latest DSM. I am using the Plugable adapter

bao3, 2022-10-31:
I have the similar issue :

DS 718+, RTL8152 , the driver from this repo works flawlessly. But the app DownloadStation show the message" could not connect to the tracker". Then I find I could not assign the default gateway to the USB network interface , and at the same time I could not edit the gateway list order ( it shows nothing).

So I need to add the default gateway by command line ( OR use the task )

`/sbin/route  add -net 0.0.0.0 gw 192.168.50.1`
 ( 192.168.50.1 is my ASUS router IP)

That command saves me and my DSM 7.1

bb-qq, 2022-11-01:
@bao3, Instead of manually setting the default gateway, could you try the following command to see if your symptoms improve? This command should apply the default gateway settings configured by the GUI.

```
. /etc/iproute2/script/gateway-mgt-function
set_default_gateway_interface
```

bao3, 2022-11-01:
> 

I tried but it  didn't work. 
[image]

bao3, 2022-11-01:
update that :

```

sh-4.4# set_default_gateway_interface
+ set_default_gateway_interface
++ /bin/get_key_value /etc/iproute2/config/default-gateway DEVICE
+ local DEVICE=eth2
++ /usr/syno/sbin/synonetdtool --get-interface-by-class -4 wan
++ sed -e 's/,/\n/g'
++ /bin/get_key_value /etc/iproute2/config/default-gateway-v6 DEVICE
+ local DEVICE_V6=
++ /usr/syno/sbin/synonetdtool --get-interface-by-class -6 wan
++ sed -e 's/,/\n/g'

```

bb-qq, 2022-11-01:
Have you already set the default gateway from the GUI? The logs show that the script is trying to use the eth2 gateway setting, but the setting seems to be empty.

bao3, 2022-11-01:
I did but not so lucky , it is blank indeed .

[image]

bb-qq, 2022-11-01:
Hmm, could you please paste the results of these commands?
```
sudo cat /etc/iproute2/config/default-gateway
sudo cat /etc/iproute2/config/eth2-table-rule
sudo cat /etc/iproute2/config/gateway_database
```

bao3, 2022-11-01:
```
ash-4.4# sudo cat /etc/iproute2/config/default-gateway
+ sudo cat /etc/iproute2/config/default-gateway
DEVICE=eth2
ash-4.4# sudo cat /etc/iproute2/config/eth2-table-rule
+ sudo cat /etc/iproute2/config/eth2-table-rule
[eth2_policy]
	mask=255.255.255.0
	ip=192.168.50.80
	ifname=eth2
	gateway=192.168.50.80
[multi-gateway]
	mask=
	ip=
	ifname=eth2
	gateway=192.168.50.1
ash-4.4#
ash-4.4# sudo cat /etc/iproute2/config/gateway_database
+ sudo cat /etc/iproute2/config/gateway_database
[bond0]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f30:8f0::1
[eth0]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f30:8f0::1
[eth1]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f30:2d80::1
[eth2]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f30:8f0::1
[ovs_eth0]
	dns=192.168.50.1
	gateway=192.168.50.1
[ovs_eth1]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f30:2d80::1
[ovs_eth2]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f31:a50::1
ash-4.4#

```

bao3, 2022-11-01:
Well ..... there is no `apple_if_config()` function in my `start-stop-status` script . So that is the problem ? The bellow code is my `start-stop-status` script

```
#!/bin/sh
# Copyright (C) 2000-2017 Synology Inc. All rights reserved.

set -eux

script_root=`dirname $0`
package_root=`realpath -m ${script_root}/..`
driver_name=`basename ${package_root}`
driver_root="${SYNOPKG_PKGDEST:-${package_root}/target}/${driver_name}"

is_ovs_enable ()
{
    if [ ! -f  '/usr/syno/etc/synoovs/ovs_reg.conf' ]; then
        return 1
    fi

    use=`cat /usr/syno/etc/synoovs/ovs_reg.conf | wc -l`
    if [ 0 -eq $use ]; then
        return 1
    else
        return 0
    fi
}

set_ovs_interface()
{
  interface_name=$1
  action=$2

  config_file_ovs=/etc/sysconfig/network-scripts/ifcfg-ovs_$interface_name
  config_storage_location_ovs=$package_root/etc/ifcfg-ovs_$interface_name

  if is_ovs_enable
  then
    if [ -f "$config_file_ovs" ] && [ "$action" = "down" ]
    then
      cp $config_file_ovs $config_storage_location_ovs
    elif [ "$action" = "up" ] && [ -f "$config_storage_location_ovs" ]
    then
      cp $config_storage_location_ovs $config_file_ovs
    fi
    if [ "$action" = "up" ]
    then
      ovs-vsctl add-br ovs_$interface_name
      ovs-vsctl add-port ovs_$interface_name $interface_name
      ifconfig ovs_$interface_name $action
    fi
    if ifconfig ovs_$interface_name 2> /dev/null && [ "$action" = "down" ]
    then
      ifconfig ovs_$interface_name $action
      ovs-vsctl del-br ovs_$interface_name
    fi

    if [ "$action" = "up" ] && [ -r "$config_file_ovs" ]
    then
      source $config_file_ovs
      if [ "$BOOTPROTO" = "dhcp" ]
      then
        synonet --dhcp ovs_$interface_name || true
      fi
    fi
  fi
}

set_interfaces()
{
  action=$1
  target_interface_name=${2:-}
  all_interface_names=$(ls /sys/class/net)

  if [ -n ${target_interface_name} ] && [ ! -e "/sys/class/net/${target_interface_name}" ] && [ "$action" = "down" ]
  then
    echo "${target_interface_name} is already detached."
    set_ovs_interface ${target_interface_name} down
    return 0
  fi

  for interface_name in ${all_interface_names}
  do
    # Skip other interfaces if an interface was explicitly specified
    if [[ -n "$target_interface_name" && "$interface_name" != "$target_interface_name" ]]
    then
      continue
    fi

    # Skip loopback/docker interface, since it doesn have a device/driver
    # folder.
    if [ ! -e /sys/class/net/$interface_name/device/driver ]
    then
      continue
    fi

    # For all other interfaces list the device driver folder and
    # bring interface up/down if the device driver folder links to
    # something with the target device driver in its name.
    driver_location=$(ls -ld /sys/class/net/$interface_name/device/driver)
    if [ ! -z "$(echo "$driver_location" | grep $driver_name)" ]
    then
      config_file=/etc/sysconfig/network-scripts/ifcfg-$interface_name
      config_storage_location=$package_root/etc/ifcfg-$interface_name
      if [ -f "$config_file" ] && [ "$action" = "down" ]
      then
        cp $config_file $config_storage_location
      elif [ "$action" = "up" ] && [ -f "$config_storage_location" ]
      then
        cp $config_storage_location $config_file
      fi
      ifconfig $interface_name $action
      set_ovs_interface $interface_name $action
      if [ "$action" = "up" ] && [ -r "$config_file" ] && ! is_ovs_enable
      then
        source $config_file
        if [ "$BOOTPROTO" = "dhcp" ]
        then
          synonet --dhcp $interface_name || true
        fi
      fi

      if [ "$action" = "up" ] && [ -e ${script_root}/apply-private-flags ]
      then
        sh ${script_root}/apply-private-flags $interface_name
      fi
    fi
  done
}

count_usb_interfaces()
{
  ls -l /sys/class/net/ | grep /usb | wc -l
}

case $1 in
  start)
    if [ -e ${script_root}/apply-memory-setting ]
    then
      sh ${script_root}/apply-memory-setting
    fi

    initial_count=`count_usb_interfaces`

    if [ -e ${driver_root}/mii.ko ]
    then
      /sbin/insmod ${driver_root}/mii.ko
    fi
    if [ -e ${driver_root}/usbnet.ko ]
    then
      /sbin/insmod ${driver_root}/usbnet.ko
    fi
    /sbin/insmod ${driver_root}/${driver_name}.ko

    if [ -r /usr/lib/udev/rules.d/51-usb-${driver_name}-net.rules ]
    then
      exit 0
    fi

    for i in `seq 1 60`
    do
      if [ `count_usb_interfaces` -ne $initial_count ]
      then
        break
      fi
      sleep 1
    done

    set_interfaces up
    exit 0
  ;;
  stop)
    set_interfaces down
    /sbin/rmmod ${driver_root}/${driver_name}.ko || true
    if [ -e ${driver_root}/usbnet.ko ]
    then
      /sbin/rmmod ${driver_root}/usbnet.ko || true
    fi
    if [ -e ${driver_root}/mii.ko ]
    then
      /sbin/rmmod ${driver_root}/mii.ko    || true
    fi
    exit 0
  ;;
  status)
    /sbin/lsmod | grep ${driver_name} && exit 0 || exit 3
  ;;
  killall)
    exit 0
  ;;
  log)
    exit 0
  ;;
  reload)
    set_interfaces down $2
    set_interfaces up $2
    exit 0
  ;;
  remove)
    set_interfaces down $2
    exit 0
  ;;
esac
```

bb-qq, 2022-11-01:
Oh, sorry, please update the driver to [2.16.3-2](https://github.com/bb-qq/r8152/releases/tag/2.16.3-2) before modifying the script.

bao3, 2022-11-02:
Ok, I updated to   `2.16.3-2 `, and reboot DSM . Then I changed `apply_if_config()` .

But it looks like nothing changed ,the result  as bellow:

```
ash-4.4# vi /var/packages/r8152/scripts/start-stop-status
ash-4.4#
ash-4.4#
ash-4.4# sudo cat /etc/iproute2/config/default-gateway
DEVICE=eth2
ash-4.4# sudo cat /etc/iproute2/config/eth2-table-rule
[eth2_policy]
	mask=255.255.255.0
	ip=192.168.50.80
	ifname=eth2
	gateway=192.168.50.80
[multi-gateway]
	mask=
	ip=
	ifname=eth2
	gateway=192.168.50.1
ash-4.4# sudo cat /etc/iproute2/config/gateway_database
[bond0]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f30:8f0::1
[eth0]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f30:8f0::1
[eth1]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f30:2d80::1
[eth2]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f30:8f0::1
[ovs_eth0]
	dns=192.168.50.1
	gateway=192.168.50.1
[ovs_eth1]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f30:2d80::1
[ovs_eth2]
	dns=192.168.50.1
	gateway_v6=fe80::fe34:97ff:fe38:ec80
	gateway=192.168.50.1
	dns_v6=2408:8215:f31:a50::1
ash-4.4# . /etc/iproute2/script/gateway-mgt-function
ash-4.4# set_default_gateway_interface
+ set_default_gateway_interface
++ /bin/get_key_value /etc/iproute2/config/default-gateway DEVICE
+ local DEVICE=eth2
++ /usr/syno/sbin/synonetdtool --get-interface-by-class -4 wan
++ sed -e 's/,/\n/g'
++ /bin/get_key_value /etc/iproute2/config/default-gateway-v6 DEVICE
+ local DEVICE_V6=eth2
++ /usr/syno/sbin/synonetdtool --get-interface-by-class -6 wan
++ sed -e 's/,/\n/g'
ash-4.4#
ash-4.4#
ash-4.4#
ash-4.4#
ash-4.4# route
+ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
169.254.0.0     0.0.0.0         255.255.0.0     U     0      0        0 eth1
169.254.0.0     0.0.0.0         255.255.0.0     U     0      0        0 eth0
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
192.168.50.0    0.0.0.0         255.255.255.0   U     0      0        0 eth2
ash-4.4#
```

bb-qq, 2022-11-05:
Did you restart the driver or the system?
If you already did that, please try the alternative modification,
https://github.com/bb-qq/r8152/releases/download/2.16.3-2/r8152-apollolake-2.16.3-3test.spk

bb-qq, 2022-12-04:
I am closing this issue for now as I have not heard back from you for a long time. If you try the latest release and it doesn't work, please re-open the issue.

https://github.com/bb-qq/r8152/releases/tag/2.16.3-3

closed 2022-12-04 by bb-qq